esp8266/Arduino #5880: Server Freezes if we disconnect the wifi network during stream of the file

laercionit, 2019-03-14:
- Hardware:			      [ESP-12]
- Core Version:			[2.4.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]
- Module:  			     [Nodemcu]
- Flash Size: 		             [1MB]

I have a situation that I could not solve.

Using the FSBrowser example, if I turn off the wifi router while the server is sending the file stream, esp8266 freezes without generating any debug information, simply stop processing the code.

Is there any way to avoid this?

devyte, 2019-03-15:
Is this reproducible with 2.5.0?

laercionit, 2019-03-15:
I did the same test using core 2.5.0.
Apparently it worked after disconnecting the wifi router during the stream of the file, I believe it has a treatment for this in core 2.5.0 that does not crash the esp8266.

Unfortunately my project is quite large, putting it in 2.5.0 has some adjustments I need to make. I saw that post is not accepting any more special characters and other situations that I need to test.

I would like to know if there are any changes I can make in core 2.4.2 to handle this situation if I lock the esp8266 if the wifi is turned off during the stream file.

devyte, 2019-03-16:
Closing due to already fixed.

devyte, 2019-03-16:
We can't modify 2.4.2 because we don't work with a branched release model, so you're right: you could locally add patches to your 2.4.2.
You would have to figure out what is the difference that you need. I suggest diffing Stream.h|cpp and Print.h|cpp and ESP8266WiFiClient.h|cpp and related files.