esp8266/Arduino #5884: feedback form issues https://github.com/esp8266/Arduino/issues/5872#issue-420296439

kaihangchen, 2019-03-16:
I tried to change the hardware, but found a new problem, there is always a node can not find the mesh AP, for example, The first and second can occasionally communicate but the third will not find the meshAP。
Can you tell me your IDE settings? 
thank you very much.

aerlon, 2019-03-16:
### Settings in IDE

- Module:  			NodeMCU 1.0
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

The way the mesh library works at the moment is that one node usually becomes the AP and the other nodes connect to it. So it is completely normal that the node with the lowest nodeID cannot find any AP to connect to (you can change this behaviour by modifying the `networkFilter` method). For me it looks like this:

[image]

Was that the issue you were having? Otherwise perhaps it is related to [this](https://github.com/esp8266/Arduino/issues/5032).

earlephilhower, 2019-07-25:
This bug doesn't have a MCVE or even a clear description of the problem.  The author didn't respond at all to @aerlon , either.  Closing.  If there is a real issue please follow the form and give a meaningful bug title so we can understand the issue.

closed 2019-07-25 by earlephilhower