esp8266/Arduino #5885: Soft WDT reset on interrupts

CRCinAU, 2019-03-16:
### Basic Infos
- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				Wemos D1 Mini clone
- Core Version:			2.5.0
- Development Env:		Platformio
- Operating System:		Fedora 29

### Problem Description
When using interrupts to count transitions (CHANGE), I get a stack trace and the program crashes. Have also tried with FALLING, but still same issue. I believe I've marked everything I need.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>
#define PIN_UP_LED D6
#define PIN_DOWN_LED D7

volatile uint8_t downChanges = 0;
volatile long lastDownChange = 0;
volatile uint8_t upChanges = 0;
volatile long lastUpChange = 0;

void ICACHE_RAM_ATTR handleUpInterrupt() {
    if ( millis() > (lastUpChange + 50 ) ) {
        upChanges++;
    }
    lastUpChange = millis();
}
void ICACHE_RAM_ATTR handleDownInterrupt() {
    if ( millis() > (lastDownChange + 50 ) ) {
        downChanges++;
    }
    lastDownChange = millis();
}

void setup() {
    pinMode(PIN_UP_LED, INPUT);
    pinMode(PIN_DOWN_LED, INPUT);
    attachInterrupt(digitalPinToInterrupt(PIN_UP_LED), handleUpInterrupt, CHANGE);
    attachInterrupt(digitalPinToInterrupt(PIN_DOWN_LED), handleDownInterrupt, CHANGE);
}

void loop() {
    Serial.print("upChanges = "); Serial.println(upChanges);
    Serial.print("downChanges = "); Serial.println(downChanges);
    Serial.print("PIN_UP_LED = "); Serial.println(digitalRead(PIN_UP_LED));
    Serial.print("PIN_DOWN_LED = "); Serial.println(digitalRead(PIN_DOWN_LED));
    bool UpLED = digitalRead(PIN_UP_LED);
    bool DownLED = digitalRead(PIN_DOWN_LED);
    delay(1000);
}

```

### Debug Messages
```
upChanges = 0
downChanges = 1
PIN_UP_LED = 0
PIN_DOWN_LED = 1
upChanges = 0
downChanges = 0
[... 338 lines cut ...]
3ffffeb0:  00000000 3fffdad0 3ffef480 00000030
3ffffec0:  4020d3fa 00000000 00000001 00000001
3ffffed0:  3fffdab0 00000000 3fffd9d0 3ffef480
3ffffee0:  00000000 00000000 00000001 3ffef450
3ffffef0:  00000000 3fffdad0 3ffef480 00000030
3fffff00:  3ffef2c8 3ffef2d0 3ffef1c8 4020d85e
3fffff10:  00000000 4bc6a7f0 00000baa 3ffef2fc
3fffff20:  40105504 000e9a51 3ffef4dc 00000000
3fffff30:  3ffee9b0 3ffef4dc 3ffe8514 3ffef4dc
3fffff40:  3ffef0b8 3ffef2fc 4020d348 3fffefa0
3fffff50:  3ffef4dc 3ffef0b8 00000064 4020160f
3fffff60:  3ffef0b8 00000000 3ffef450 402030b4
3fffff70:  4020e954 4020e968 3ffef0b8 40202fda
3fffff80:  402028b8 00000000 4020e994 4020e97c
3fffff90:  00000000 00000000 00000001 3ffef480
3fffffa0:  3fffdad0 00000000 3ffef450 4020d3f4
<<<stack<<<
ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld
```

CRCinAU, 2019-03-16:
I wonder if this is somewhat related to #5788 - even though I'm not using SPIFFS at all?

CRCinAU, 2019-03-16:
Debugging this further, the MCVE code actually works.... (if you add the missing Serial.begin(115200).

It looks like this dies when I do a check for OTA updates - which takes a couple of seconds to run.

If I wrap this in `noInterrupts()` and then `Interrupts()` when I call the function that does the OTA http check, everything seems to be fine.

Is this expected?

CRCinAU, 2019-03-16:
Pinging @d-a-v  - he seems the master of interrupts on ESP8266 from everything I've read. This may be something simple that I just haven't been able to get my head around... Interrupts seem like unstable voodoo at the moment :(

d-a-v, 2019-03-16:
I am far from being the interrupt master :)
Please summarize your findings, update your MCVE, add `Serial.begin()`
> actually works
> still crashed

Right after boot, or after some time ?
If you suspect millis(), have you tried to replace/remove it just to verify whether it's this call ?
Have you had a look to Functional Interrupts ?

I suggest to use git version master branch instead of v2.5.0.

CRCinAU, 2019-03-16:
I started adding code to the MCVE - but couldn't make it crash again until I hit the entire program I have :(

When I'm looking at the serial output, I can see a number of lines print, then there's a pause, sometimes there's a flood of loops run, then there's the reset again. It seems to do 2 x loops, then the pause, then sometimes a rush, then reset.

I'm going to try putting in a whole lot more Serial.print debug into my code and see what I can figure out before the crash.

This is my first experiment in interrupts - so I might be missing lots - but I've been finding what I can from your replies to other interrupt bug reports to hopefully get it right.

Functional Interrupts - do you have any links to information about this? As mentioned, this is my first attempt here.

CRCinAU, 2019-03-16:
Interesting news - when running with Debug port serial and Core+WiFi debug,  see the following output:

```
SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1/BearSSL:6778687
sleep disable

Running at millis: 2169
upChanges = 0
downChanges = 240
PIN_UP_LED = 0
PIN_DOWN_LED = 1
wifi evt: 2

Running at millis: 3171
upChanges = 0
downChanges = 150
PIN_UP_LED = 0
PIN_DOWN_LED = 1

scandone
```

At this point, the crash happens.

This code has millis() commented out in the interrupt - so I'm guessing we can rule that bit out... However I do note that I seem to get many more interrupts than I was expecting... I didn't see that when filtering for +50ms - however I wonder if the frequency of interrupts may be part of the problem?

devyte, 2019-03-16:
This looks like either an issue in the user's code or another instance of #5784 .
I can't reproduce with latest git and the provided MCVE. Closing for now.
If something actionable is found, please report it here and this will be reopened.

closed 2019-03-16 by devyte

laercionit, 2019-03-21:
I have the same problem.

d-a-v, 2019-03-21:
Are you able to reproduce the bug with core-release-2.4.2,

or better with core-git-master ?
https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version (#5882 #5816)

laercionit, 2019-03-21:
O problema não ocorre no 2.4.2, apenas no 2.5.0
Estou preparando ambiente para testar com a master agora.

CRCinAU, 2019-03-21:
I have a completely non-verified hunch on the problem... As the crash *always* happened at the 'scandone' print, I think that something to do with the wifi library is blocking or causing interrupts to stack up. I did actually find out that the signal I was monitoring was dirty as hell causing ~40-60k interrupts per second (using CHANGE). That may well be related.

As I can't easily change the circuit, I ended up using a 'best out of 10' polling method instead.

laercionit, 2019-03-21:
Unfortunately I can not put the master to run, it seems that incorporated SDFat in it, this is generating several conflicts in my project.

d-a-v, 2019-03-21:
It is supposed to be compatible and this version will be in core-release-2.5.1 soon,
so please elaborate.

Frogmore42, 2019-03-26:
I have been battling this issue trying to get 9-bit soft serial to work reliably.  I finally got some output from the SDK logging and discovered this:
https://github.com/espressif/ESP8266_NONOS_SDK/issues/90

It looks like the sdk doesn't work well when ISRs take longer than 50us, so high speed software serial is pretty much out.