esp8266/Arduino #5886: Use of String in libraries

adrionics, 2019-03-16:
As someone who has made great headway to remove all instances of String objects from my code, I was wondering why some libraries use them a lot, for instance the HTTPClient and WebServer libraries. If String use is not recommended, why do they still use them so much?

Are there any plans to update libraries to remove String use in the future?

d-a-v, 2019-03-16:
> why do they still use them so much?

Because this is history. Anyone is welcome to improve them. There is an ongoing work for WebServer #5636, but if you have any plan to improve current code, you are welcome.

Strings are not forbidden. They make holes (heap fragmentation) so special care has to be taken with them. Use `.reserve()` when possible, or `char*` when wise. A nice improvement recently happened #5690.

devyte, 2019-03-16:
> String use is not recommended

That sentence is incomplete. The missing part says "by inexperienced programmers".
String manages its own dynamic allocs internally, and the use *must* be aware of how it behaves in a tiny system like the ESP. Usage such as pass-by-value or long sequences of concats are bad. Using when receiving chars and not knowing when the sequence ends is preferable to C-style manual handling of the dynamic allocs, so it's good.
As a user, it can ease a whole lot of use cases, but you *must* be a responsible adult and use them correctly.
Closing due to non-issue.

closed 2019-03-16 by devyte

adrionics, 2019-03-17:
Thank you for the feedback.