esp8266/Arduino #5888: Are there several versions of ArduinoOTA?

alkhimey, 2019-03-16:
#### Platform

- Hardware:				ESP-12E (nodemcu v3 Lolin)
- Core Version:			?
- Development Env:		PlatformIO
- Operating System:		Ubuntu

### Problem Description

I located the [ArdinoOTA](https://platformio.org/lib/show/6178/ArduinoOTA) library in PlatformIO and installed it. Then I used the example sketch found [here](https://github.com/esp8266/Arduino/blob/e9d052c621953b7932cffb91c58c14207a5e526d/libraries/ArduinoOTA/examples/BasicOTA/BasicOTA.ino).

I got few compilation errors. I Investigated a bit and it seems that PlatformIO version of the library looks official, but different from the library in this repository. 

[image]

[image]

Now I understand that installing it is not required since ArduinoOTA is part of the framework.

So why the stand alone library exists?

devyte, 2019-03-16:
You seem to be confusing the Arduino ArduinoOTA with our ESP8266 ArduinoOTA.
Also, we support pio, but we don't maintain it, that's a different repo.
Closing due to non issue.

closed 2019-03-16 by devyte

jandrassy, 2019-03-17:
he ArduinoOTA library for esp8266 is bundled in boards package. the standalone library is for SAMD, nRF5 and ATmega with 64kB or more flash for sketch upload with Ethernet and WiFi libraries