esp8266/Arduino #5892: ESP.deepSleep not able to wake up device

mikegron, 2019-03-18:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12E (NodeMCU v3)
- Core Version:		        stable release (2.5.0)
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU 1.0 (ESP-12E)
- Flash Mode:			?
- Flash Size: 		        4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		?
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL (usb)
- Upload Speed:   115200

### Problem Description

I'm trying to make deep sleep work as intended. I made a very simple sketch that just sleeps for 30 seconds. I've attached a wire from GPIO_0 to the RST pin so that the deep sleep timer could reset the board. I am connected/powered via USB.

After the specified time, a LOW signal is correctly sent to the GPIO_0 pin as the device write the boot information but nothing else is happening afterward, I have to unplug the power or manually reset the board with the reset button. I tested with 115200 or 74880 bauds if that makes a difference, to see the boot info which is in 74880.

### Sketch

```cpp

void setup() {
  Serial.begin(74880);
  Serial.setTimeout(2000);
  while (!Serial) { }
  Serial.println("Started.");
  delay(5000);
  Serial.println("Going to sleep.");
  ESP.deepSleep(30e6);
}

void loop() { }

```

### Debug Messages

[image]

5chufti, 2019-03-18:
try putting a delay(10) after the ESP.deepSleep()

jandrassy, 2019-03-18:
do you have io 0 not pulled-up?

liebman, 2019-03-18:
Looking at the log messages this is working correctly. Deep sleep does not continue from the sleep, it starts with setup again. It’s basically a reset. If you have data you want to persist then you need to use RTC memory or flash.

mikegron, 2019-03-18:
Hmm if I remove the cable from D0 to RST I still have the "ets Jan..." message so I guess this is printed on wake and not when receiving a reset signal. I tried resetting "manually" by hooking RST to ground and that works, I still have the same boot message but the board starts properly. So maybe D0 simply does not send the wake (LOW) signal??

fsommer1968, 2019-03-18:
I use the following pin configuration for DO:
` pinMode(D0, WAKEUP_PULLUP);`
In addition instead of a direct connection from DO to RST I use a Schottky Diode (preferred) or a 680Ohm resistor.

fsommer1968, 2019-03-25:
I recommend closing this issue

mikegron, 2019-03-25:
Setting the pin to WAKEUP_PULLUP did not change the behaviour unfortunately. Could you tell me why would adding a resistor between D0 and RST fix that?
Also as a test, setting D0 to INPUT made the board go in a reset loop (so it looks like it _can_ work).

fsommer1968, 2019-03-25:
Wake up pin is Nodemcu pin D0 which is 8266 pin GPIO16. You are referring to GPIO_0 , this is not the Wakeup pin! Check your wiring and pin assignment: https://randomnerdtutorials.com/esp8266-deep-sleep-with-arduino-ide/

mikegron, 2019-03-26:
My bad for referring to GPIO_0, indeed I meant D0 (GPIO_16), I used the same exact link you posted (I also linked it in the problem description)
Pictures just to be sure:
[image]
[image]

mikegron, 2019-03-26:
It looks like the board is waking up in flash mode somehow, same as if I hold the flash button at the same time as the reset button. Although I did try setting D3 (GPIO 0) and D4 (GPIO 2) as INPUT_PULLUP, or attaching a cable from 3.3V to D3 (to make sure it's high). I have no ideas now..

devyte, 2019-03-26:
Are you physically resetting the board after flashing over serial?

mikegron, 2019-03-26:
Yeah, I also tried using a power supply, same result :/

fsommer1968, 2019-03-27:
I tried with a Nodemcu 0.9 (ESP12) module and the connection from DO to RST, and you sketch works:
`17:00:32.789 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
17:00:32.789 -> 
17:00:32.789 -> load 0x4010f000, len 1384, room 16 
17:00:32.789 -> tail 8
17:00:32.789 -> chksum 0x2d
17:00:32.789 -> csum 0x2d
17:00:32.789 -> v951aeffa
17:00:32.789 -> ~ld
17:00:32.857 -> Started.
17:00:37.853 -> Going to sleep.
17:01:06.604 -> 
17:01:06.604 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
17:01:06.604 -> 
17:01:06.604 -> load 0x4010f000, len 1384, room 16 
17:01:06.639 -> tail 8
17:01:06.639 -> chksum 0x2d
17:01:06.639 -> csum 0x2d
17:01:06.639 -> v951aeffa
17:01:06.639 -> ~ld
17:01:06.706 -> Started.
17:01:11.687 -> Going to sleep.
`
But when i forget the conenctionfrom D0 to RST this happens:
`17:01:45.529 -> Going to sleep.
17:02:14.316 -> 
17:02:14.316 ->  ets Jan  8 2013,rst cause:5, boot mode:(3,6)
17:02:14.316 -> 
17:02:14.316 -> ets_main.c `

So from my point of view there might be something with your board or the board description in Arduino for the ESP12E module is wrong.

mikegron, 2019-03-27:
Thank you for testing! There is probably something wrong with the board you're right..
I checked the pins (with nothing else connected) with a multimeter and I find the measures odd:
The RST pin always stays at 3.3V.
When running, the D0 pin is at 0V. When sleeping, the D0 pin is at 3.3V. When waking up, the D0 pin is set to 0V _continuously_. However, to wake the board with the RST pin, I need to set it to 3.3V -> 0V -> 3.3V. For instance, connecting RST to ground does _not_ reset the board, but removing the connection will reset it. But, disconnecting the D0 pin while it's at 0V does not have the same effect.
Do you see anything wrong with that?

fsommer1968, 2019-03-28:
Hm measuring dynamic signals with a multimeter where typically a logic analyser a scope or at least a logic tester is required is not more than a bad guess. 
I think you should look around for other using ESP12E with deep sleep. I have described what I was able to test. And so far I cannot see that there is something wrong with this release. 
As explained earlier: maybe the bord description needs an update for this specific board. I would look for the schematics of this board, set the board in the Arduino GUI to to generic 8266 and set each option manually.

mikegron, 2019-03-30:
Ok so I tested the exact same sketch with a [Wemos D1 R1](https://alselectro.files.wordpress.com/2018/04/d1_pinout.jpg?w=640) and it worked perfectly... I'm gonna close this issue and try to figure out what is wrong with the NodeMcu v3 !
Thanks a lot for your help.

closed 2019-03-30 by mikegron

5chufti, 2019-03-31:
To me that is an old version where the reset circuit does prevent the pulse form gpio16 to be recognized correctly. The new nodemcus have a different layout with one free (not assembled) resistor location to bridge when using deep sleep

petrilloa, 2019-04-22:
I have a similar issue with a bunch of ESP12F from Espressif.
We build more than 100 sensors with ESP12F from AI, and doesn´t have any problem to wake up from deep sleep, GPIO16 direct connection to RESET.
With the new batch, the factory send us ESP8266MOD (phisically looks the same as the old ones, and justo like an ESP12F), but trying a simple sketch to wake up, the devices hangs until we press reset.

We have a 3.3v from MCP1700, 1000uF and 0.1nF capacitor from VCC to GND. (input 3xAA Batteries)
GPIO15 pulldown with 10k
GPIO0 pullup with 10k
GPIO2 pullup with 10k

To connect the reset we try...

1-Connect Reset to GPIO16 with wire
2-Connect Reset to GPIO with 330 ohms resistor
3-Connect Reset to GPIO with Diode (normal and schottky).

Debbug:

**AI ESP8266 12-F  (WORKING)**

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld
Chip ID: 276350

ResetInfo.reason = 5: Deep-Sleep Wake
.
I'm awake.
Going into deep sleep for 3 seconds

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld
Chip ID: 276350

ResetInfo.reason = 5: Deep-Sleep Wake
.
I'm awake.
Going into deep sleep for 3 seconds

**Espressif ESP8266 12-F (NOT WORKING)**

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v951aeffa
~ld
Chip ID: 6198051

ResetInfo.reason = 5: Deep-Sleep Wake
.
I'm awake.
Going into deep sleep for 3 seconds

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

Please any help!?

fsommer1968, 2019-04-22:
Hi petrilloa, 
that is sad! Try to get schematics for both boards and compare them. Beside the GPIO16 and reset I would check the SPI flash connection.

petrilloa, 2019-04-22:
There is a custom PCB, with a bare ESP12F. Si there is not diff schematics.

Skipper-is, 2019-06-01:
Same board, Lolin NodeMCU v3. Exactly the same issue. I've got 3 of the same board, and they all exhibit the same behaviour described here.

ArsineMan, 2019-06-10:
Same Problem. I purchased 10 of these guys. So far 3 are functional and 4 are not. I literally just plug them into the same breadboard location and load the exact same sketch.  I stuck an oscilloscope on the D0 pin and noted the working boards do drop all the way to 0 whereas the broken ones do not.

devimc, 2019-07-28:
I updated my platformio framework and its packages (framework-arduinoespressif8266) and now I'm facing this issues :cry: , but if I revert everything to 2018-12-03, `ESP.deepSleep` works again  :smile: , this means that something changed in the library or toolchain :thinking:

ArsineMan, 2019-07-28:
@devimc, that's interesting. I assumed it was my hardware so I bought from a different vendor and its started working.

ptutt1, 2019-08-25:
I have a Lolin Nodemcu v3 and have same issue as described.  Will enter deep sleep but will not come out.  Interestingly, it is drawing around 72ma when running, then around 250ua when in deep sleep.  After it attempts to wake up, I can see the on board led flash, it uses around 36ma.  My guess is as other has mentioned it wakes into flash state.  I also find that I have to press the reset button twice to wake it from deep sleep.  I can also need to ground RST twice to exit deepsleep.

I also captured D0 output on oscilloscope and can see it seems to be correctly grounding to trigger the reset, but as mentioned, if reset needs to happen twice, then it's not going to work.

[image]

Easiest fix seems to be getting new hardware that supports deepsleep properly.

5chufti, 2019-08-30:
have a look at #6007

magc2806, 2020-03-30:
Hello ! I am having this exact problem with my Amica NodeMCU Esp8266. Has anyone found a solution for this or I must change my board? How do I know my board's version ? And which ones dont have this problem? thank you in advance

ptutt1, 2020-03-30:
@magc2806 in my case I purchased a WeMos ESP8266 D1 Mini and there was no problem.  My problem was with the Lolin Nodemcu v3 but I can't comment on other boards.

magc2806, 2020-03-30:
Thanks a lot for your reply @ptutt1 . Have you tried the sleep mode with another NodeMCU v3?

ISObob21, 2020-04-28:
I have the same problem with the Lolin NODEMCU V3, the wide unit. Load same code onto a DoIt ESP12E Devkit V2, 22mm wide unit,  and all is well.

allexb, 2020-07-21:
The same problem for me.
LoLin new NodeMcu V3 (ESP8266MOD)

worksasdesigned, 2020-09-23:
Same issue with the "xxxxMOD" Version. 12F is running perfectly smooth.  
Just took me 4 hours to find this post and another 5 min to exchange the WeMos with an other batch.

Craig1516, 2021-07-05:
I am revisiting two, or three, batches of ESP12's from 2019'ish.  Some work after deepsleep, some do not.  Double resets do work, but... the first reset ends badly with high power consumption (33 mA).  Sadly, my QC is sloppy and I have several protypes on several boards from several purchased batches.  And with 16 surface mounts these chips do not come off the PCB nicely.  Testing, testing, testing... so important.

ole1986, 2021-09-02:
I also noted the same.
While using deep sleep in a short period (1 min) it seem to work AS expected.

When I use deep sleep for 1h with a powerbank as power supply it never wake up.

5chufti, 2021-09-02:
did you check that it is not the powerbank that shuts down and hinders the esp to wake up?

ole1986, 2021-09-03:
I have now double checked while connecting to debugging computer for more than an hour.
Unfortunately it keeps staying in deep sleep.

Trying to reduce it to 30 minutes now!

Below is the code example and output
```
// where settingData.interval is 60 minutes
Serial.println("Going into deep sleep for " + String(settingData.interval) + " minutes");
ESP.deepSleep(settingData.interval * 60 * 1000000);
```
Output
````
Going into deep sleep for 60 minutes
````

Craig1516, 2021-09-03:
Seems like 30 minutes should work for you.  I think the max sleep time is a bit over 35 minutes.  The technicals (2^31) are discussed [here](https://forum.arduino.cc/t/esp8266-and-deep-sleep-time/677977/3).  

Maybe your powerbank as mentioned by Schufti?  A multimeter should show you if that is the problem.

Craig1516, 2021-09-03:
Here is [another ](https://panther.kapsi.fi/posts/2019-05-14_esp8266_deep_sleep)fun story.   He found that by swapping to a different ESP it solved his sleep problem.

ole1986, 2021-09-09:
As a side note. It's wasnt the powerbank but the esp when configuring it to approximately 1 hour.

When setting up to 30 min deep sleep and wake up works so far with the powerbank

FBMinis, 2022-11-23:
I have just landed here from Google and it was the powerbank that would allow the ESP-07S to deepsleep. After testing with a phone charger and 2xAA it worked.

amadeuspzs, 2024-02-11:
Can confirm this behaviour with some implementations of ESP8266MOD on a Wemos D1 Mini clone.

Trying on a NodeMCU (ESP-12F) worked, so it does look like, while frustrating and somewhat incomprehensible, different hardware implementations affect the correct functioning of deepSleep mode.

For anyone else facing this issue, try a different hardware implementation.

jtauscher, 2024-07-19:
Issue can be fixed by adding 10k resistor to second pin next to the 3.3V pin.
[image]

MrSp4nky, 2024-07-19:
@jtauscher Thanks! This fixed the issue.

5chufti, 2024-07-19:
do yourself a favour and use the technically correct method to connect rst and gpio16:
 a schottky diode, cathode (ring) to gpio16 (despite what the datasheet says);
this will greatly improve your flashing (and wakeup) experience.

lionel-gattegno, 2024-07-29:
Hi @5chufti can you please elaborate? how will it improve flashing and wakeup experience? (I'm brand new to electronics and super interested to learn...)

5chufti, 2024-08-01:
> Hi @5chufti can you please elaborate? how will it improve flashing and wakeup experience? (I'm brand new to electronics and super interested to learn...)

a) with hard link, external reset circuitry will 99.9% fail to reset ESP module (has to work against "strong HIGH" from gpio16 , so automatic flashing from IDE will work at all. 
b) with resistor, external reset circuitry will still have to work against "strong HIGH" from gpio16 through resistor.
Experience shows that experimental values for resistor may work with one batch of modules but fail on other.

with diode, both cases are solved electronically correct and compatible with 99.9% modules encountered.

jtauscher, 2024-08-02:
I don't think a diode will solve this issue. It's not a problem of RST or GPIO16.

5chufti, 2024-08-02:
there are modules with "suboptimal" internal circuitry on rst (RC network) which react erratically to the reset signal from gpio16 as allready discovered and discussed on gh in history. They suffer due to misshaped (thus shortened) pules on rst. Even this might improve with any action to keep pulse (edge) in shape!

jtauscher, 2024-08-02:
For my experience the pulse is too short or not "low enough". So maybe you are right but still I don't understand how a diode would help. With the additional R at GPIO7 it simply helps to get a decent LOW at RST. It worked for me on 2 modules so far. So it seems to be the simplest and yet reliable solution.

5chufti, 2024-08-02:
I really can't see how a pull-up on gpio7 can improve the timing related to gpio16/rst.
I think that there is a more complex underlying problem that keeps the esp from properly starting up after the wake-up reset.
One would have to decap the module and replace the flash for a known good one to get a more clear picture. Or experiment with differend flash modes or ...

simonadamprog, 2024-09-28:
Hi @mikegron,
Could you figure out what was the issue with the NodeMCU esp8266 esp12e?
I'm trying to figure out this deep sleep for weeks now and have read these comments multiple times. 😅
I've tried everything mentioned under this issue, no luck.
I have two of these beauties, Both are Amica NodeMCU v3s. And both are behaving the same.
D4 and D0 LED lights blink one when it wants to reset (just with wire between D0 and RST), but nothing happens after. When I place a shottky or 10kOhm between D0 and RST, the D0 led stays on (low), and not restarting. Tried a pullup and a pulldown on SDD0. Setting the D0 to WAKEUP_PULLDOWN, WAKEUP_PULLUP. All of this combination.
I just think now to don't go to deep sleep with them, just use them on normal mode, two eats max 500Wh per a month, I could live with that.
But still, it would be nice to figure out... And eat just 5-10Wh per month.

arwooldridge, 2024-09-28:
@5chufti I tried the Schottky diode (GPIO16 to RST), it did not cure my wake from sleep issue, but it does allow programming without having to remove the link.

5chufti, 2024-09-29:
Hi,
as I learned from others who had similar problems and disected their modules, there are some modules with incompatible circuitry (rc combination) at the reset pin that needs longer reset pulse than the gpio16 provides. This could only be "fixed" by an additional monoflop that costs more money and battery power than a new known good ESP8266 ...

arwooldridge, 2024-09-29:
Yes that's the conclusion I came to but where to get the genuine ones from? I guess it has to have the espressif logo or AI logo to be genuine? The ones I had were neither see attached, but the eBay seller ad showed the AI logo.

⁣Sent from TypeApp ​

5chufti, 2024-09-29:
depending on your needs, you could try sth like esp8266 D1 mini pro where all components are discrete (no module) and superfluous parts on reset could be removed.
Or maybe spend some more money and go for some reputable vendors on ebay or amazon.

arwooldridge, 2024-09-30:
Yes I tried the discrete D1 mini version and it wakes from sleep OK, but takes 110uA in sleep when powered from 3.3volts. I was hoping for much less with an esp8266 -12. I might try to remove redundant components or find some genuine esp8266's. It's a shame the market is probably flooded with fakes, most users would not realise if they don't use sleep.

⁣Sent from TypeApp ​

simonadamprog, 2024-10-01:
With logic analyzer I could gather this information (with schottky diode - ring facing to D0):
[image]
Looks D0 works fine, but RST is kinda messing around a little bit until it goes back to high again.

5chufti, 2024-10-03:
unfortunately a logic analyser is not a proper tool for this kind of problem as it will not show the ramp on the rst signal.
But apart from that, one would need to see the actual signal after the internal circuitry right on the pin of the chip.
As I allready wrote: if wake up does not work with a hard link D0-RST, there is nothing that can be done "from outside".
Get any of the boards that don't use modules so one can modify/remove interfering components or buy where you can be certain to get a board with quality module.

p.s.: all this is not related to the with improper load / incompatible flash ic (?) on sd_d0 fixed with resistor as shown several entries above...

Bly16, 2025-01-01:
Another weird thing I noticed, manual RST (button, but pin as well) works every second attempt, no idea why, So if i use my code, it fails at the first try and doesn't work further, but if I push a button it actually resets only for one more iteration etc. 
Am I the only one with this kind of issue?

nbbraun, 2025-03-28:
I've got five of these NodeMCU 8266. One of them works with the ESP.deepSleep() as it should. The other four won't,  they do what they should do as long as I do not try deepSleep. Can't figure out why.

[2 more replies]