esp8266/Arduino #5893: Web server send_P adds extra \0 at end of content

adrionics, 2019-03-18:
WebServer sends \0 at end of content when using send_P

SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1/BearSSL:6778687

Similar to https://github.com/esp8266/Arduino/issues/987

The following MCVE sketch serves an image using the SoftAP and adds the \0 causing the image to be corrupt.

```
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

ESP8266WebServer server(80);

//Image from https://www.svgrepo.com/svg/42915/hello-speech-bubble-handmade-chatting-symbol
const static char image[] = "<?xml version=\"1.0\" encoding=\"iso-8859-1\"?>"
"<!-- Generator: Adobe Illustrator 16.0.0, SVG Export Plug-In . SVG Version: 6.00 Build 0)  -->"
"<!DOCTYPE svg PUBLIC \"-//W3C//DTD SVG 1.1//EN\" \"http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd\">"
"<svg version=\"1.1\" id=\"Capa_1\" xmlns=\"http://www.w3.org/2000/svg\" xmlns:xlink=\"http://www.w3.org/1999/xlink\" x=\"0px\" y=\"0px\""
"  width=\"281.983px\" height=\"281.983px\" viewBox=\"0 0 281.983 281.983\" style=\"enable-background:new 0 0 281.983 281.983;\""
"   xml:space=\"preserve\">"
"<g>"
"  <g>"
"    <path d=\"M173.031,29.851c-58.7-7.3-140.1,2.5-165.4,65.9c-10.6,26.4-12.9,63.9,11.2,84.9c28.3,24.6,65.6,19.7,100.6,21.1l0,0"
"      c0.8,16.101,5.6,33.2,10.5,48.2c1.7,5.3,7.9,5.5,11,1.4c6.201-8.2,8.9-18.801,11.5-28.601c1.3-5,4.899-12.8,4.6-18.7"
"      c0.7,0,1.3,0,2.101-0.3c47-17.3,114.199-0.399,122.199-66.5C289.031,74.351,227.132,36.55,173.031,29.851z M263.531,157.45"
"      c-14.7,33.2-78.5,23-107.7,34.1c-3.1,1.2-4.3,4-4.199,6.601c-7.301,1.3-8.101,10.399-9.601,16.5c-1.4,5.5-2.7,11.3-4.6,16.899"
"      c-3.2-12.8-5.9-25.8-11.1-37.6c-1.2-2.7-4.5-2.2-6.1-0.3c-20.3-4.7-40-5-60.6-7.9c-61.1-8.399-52.1-75.7-23.7-109.8"
"      c32.5-39,98.5-40.1,144.201-32.4C232.031,52.351,289.632,98.45,263.531,157.45z\"/>"
"    <path d=\"M90.331,91.25c-1.6-4.3-7.4-5.6-10-1.3c-5,8.3-3.6,21.5-4.1,30.8c-0.3,5.1-0.7,10.6-1,16.1c-5.1,0.8-10.1,1.2-15.3,1.6"
"      c0-4.1-0.1-8.2-0.1-12.3c0-6.7,0.3-13.5,0.6-20.2c0.1-2.9,0.4-5.8,0.9-8.7c0.8-4.3,1.8-3.1-0.9-1.8c3.4-1.7,3.6-7.6,0-9.3"
"      c-21.3-9.6-12.4,67.3-11.7,77c0.5,7,10.5,7,10.8,0c0.3-5.101,0.3-10.101,0.3-15.2c5,0.3,10.1,0.4,15.2,0.2"
"      c0.2,5.5,0.9,10.8,2.5,15.7c0.6,1.899,2.2,3.899,4.3,4.3c2.1,0.399,4.4,0.3,6.1-1c1.2-0.9,1.3-2.2,0.8-3.101"
"      c1.1-3.899,0.3-7.8,0.1-12.199c-0.2-8.5,0.1-17.1,0.5-25.6C90.131,115.75,94.131,101.25,90.331,91.25z\"/>"
"    <path d=\"M125.131,92.25c-0.1-0.2-0.3-0.4-0.4-0.6c-1.7-2.3-3.6-2.9-6.3-2.6c-3.2,0.4-6.5,0.5-9.8,0.7c-1.6-0.9-3.9-0.4-4.7,1.4"
"      c-1.5,1.3-1.8,3.5-1.1,5.2c-3.6,22.5-3.9,46.1-4.1,68.699c0,3.7,2.9,5.9,6.2,6.2c8.5,0.9,19.8-0.7,19.1-11.7"
"      c-0.2-3.399-3.7-5.8-6.9-4c-2.4,1.4-4,2.4-5.8,2.9c0.1-4,0.2-8,0.4-12c1.4,0.2,2.8,0.3,4.1,0.5c2.8,0.4,4.7-0.3,6.4-2.6l0.1-0.101"
"      c1-1.3,1-4.1,0-5.399l-0.1-0.1c-1.7-2.2-3.6-3-6.4-2.6c-1.2,0.2-2.5,0.3-3.8,0.4c0.4-12.5,0.5-25-0.9-37.2"
"      c2.4,0.1,4.9,0.2,7.2,0.6c2.7,0.4,4.7-0.3,6.3-2.6c0.1-0.2,0.3-0.4,0.4-0.6C125.931,95.65,125.931,93.45,125.131,92.25z\"/>"
"    <path d=\"M155.331,154.25c-2.6-0.8-5.3-1.2-8-1.7c-0.399-19.299-1.699-38.899-4-58c-0.6-5.2-7.5-5.2-8.1,0"
"      c-2.3,20.9-0.6,42.4-0.6,63.3c0,2.899,2,5.399,4.7,6.199c5.4,1.5,10.4,2.7,15.9,1.101"
"      C160.831,163.55,160.632,155.851,155.331,154.25z\"/>"
"    <path d=\"M183.632,158.55c-1.301,0.601-2.4,1.4-3.5,1.9c0.1-21.8,0.699-43.9-4.5-65.2c-1-4.2-7.601-3.3-7.5,1"
"      c0.699,23.6-0.4,47-0.801,70.6c0,3,2,5.199,4.7,6.199c8,3,15.5,1.301,19.101-7C193.331,161.05,188.932,156.05,183.632,158.55z\"/>"
"    <path d=\"M220.731,98.65c-3.9-5.3-12-2-11.8,3.2c-0.2,0-0.3,0-0.601,0c-2.8-0.1-4.1,2.4-5.3,4.4c-3.899,6.7-5.8,15.2-7.6,22.6"
"      c-2.9,11.9-6.3,31.8,6.5,39.199c14.8,8.601,24.6-10.8,26.2-22.1C230.231,131.95,229.531,110.55,220.731,98.65z M214.831,149.05"
"      c-0.6,2.5-4.699,11.4-7.8,6.5c-5-8,0.101-24.199,2-32.5c1-4.3,2.101-8.6,3.3-12.9c0-0.1,0-0.2,0.101-0.3c2.8,7,3.5,15.5,3.8,22.5"
"      C216.531,137.851,216.132,143.55,214.831,149.05z\"/>"
"  </g></g><g></g><g></g><g></g><g></g><g></g><g></g><g></g><g></g><g></g><g></g><g></g><g></g><g></g><g></g><g></g></svg>";
  

void handleNotFound() {
  
//Does not work: (extra \0 at end of output)
  server.send_P(200, PSTR("image/svg+xml"), image, sizeof(image)); 

//The following "workaround" solves the problem:
//  server.send_P(200, PSTR("image/svg+xml"), image, sizeof(image) - 1); 

}

void setup(void) {
  Serial.begin(115200);
  WiFi.mode(WIFI_AP_STA);
  WiFi.softAP("esp8266");//
  delay(100);
  IPAddress apIP(192, 168, 4, 1);
  WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));

  server.onNotFound(handleNotFound);
  server.begin();

  Serial.println("");
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
}
```

adrionics, 2019-03-18:
It is also still present when using sendContent_P

```
      server.setContentLength(CONTENT_LENGTH_UNKNOWN);
      server.send(200, PSTR("image/svg+xml"), "");
      server.sendContent_P(image, sizeof(image));

```

earlephilhower, 2019-06-26:
I believe there's a misunderstanding here and not a bug.

`sendContent_P(char *data, int len)` is transmitting the exact same amount of data as `sendContent(char *data, int len)`, namely `len` bytes.

Your code makes `len` == `sizeof("string")`.  But, `sizeof(<any string>)` includes the terminal NUL.  That is, `sizeof("<str>") != strlen("<str>")`.  `sizeof("<str>")` == 6, `strlen("<str>")` == 5.

Instead, for ASCII data like this, just use sendContent_P(PSTR), which uses strlen_P to get the exact string len, not including the trailing 0.

closed 2019-06-26 by earlephilhower

adrionics, 2019-06-27:
Of course!! That makes sense, thank you so much for taking the time to give the detailed answer.