esp8266/Arduino #5896: While flashing binary files esp8266 from web server,it says Magic header does not start with 0xE9

vijayvikma, 2019-03-19:
Hello Experts,

I am flashing esp8266 using a web server.uploading bynary .bin files to esp8266 but it fails with magic byte not matching error.

[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'X-Powered-By: Express'
[HTTP-Client][handleHeaderResponse] RX: 'Access-Control-Allow-Origin: *'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 386432'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html; charset=utf-8'
[HTTP-Client][handleHeaderResponse] RX: 'ETag: W/"5e580-sbMqiSzVJszbBHAg13tl2+BG2to"'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Tue, 19 Mar 2019 20:21:44 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 386432
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 386432
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 3866624
[httpUpdate]  - current Sketch Size: 306208
[httpUpdate]  - current version: 1
:rch 536, 536
:rch 1072, 536
:rch 1608, 536
:close
[httpUpdate] runUpdate flash...
:pd 4, 2144, 239
:rpi 297, 4
[httpUpdate] Magic header does not start with 0xE9
[HTTP-Client][end] still data in buffer (1905), clean up.
:c 1, 536, 2144
:c 1, 536, 1608
:c 1, 536, 1072
:c0 1, 536
[HTTP-Client][end] tcp stop
:close

Can some please help since I have no clue what does it mean and how to check.
I did set up for web server refering below link.

earlephilhower, 2019-07-25:
Closing as there is no MCVE or even a detailed description.  My *guess* given the little in the bug is that you were trying to upload the `.elf` and not the `.bin`.  Please only upload `.bin`s as that is the only supported format.

closed 2019-07-25 by earlephilhower

infrafast, 2020-03-22:
Hello maybe you should not close it as I get exactly the same error, and I checked the file, it is a .bin and it start with 0xE9.
Looks like there is an issue with ESP8266
Mine is a LOLIN WEMOS D1 mini pro

connected with intervalmix, channel 1
dhcp client start...
ip:192.168.100.12,mask:255.255.255.0,gw:192.168.100.1
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 483312
[httpUpdate]  - MD5: 4fb2ad648eab6a8021b100fd5dce04d9
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 1613824
[httpUpdate]  - current Sketch Size: 483312
[httpUpdate]  - current version: 0.99.7
[httpUpdate] runUpdate flash...
[httpUpdate] Magic header does not start with 0xE9
ping 5, timeout 0, total payload 160 bytes, 5084 ms
[WS-Server] Server Started.
pm open,type:2 0