esp8266/Arduino #5899: An error occurred when I compiled the sample code Blinker_HELLO_WIFI

edanla, 2019-03-21:
#### The following error occurred when I was compiling the sample code Blinker_HELLO_WIFI. Who can help me see why

##### Arduino:1.8.8 (Windows 7), 开发板："Generic ESP8266 Module, 80 MHz, Flash, Disabled, ck, 26 MHz, 40MHz, DOUT (compatible), 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

``` c
In file included from C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/BlinkerESPMQTT.h:10:0,

                 from C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Blinker.h:122,

                 from C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\examples\Blinker_Hello\Hello_WiFi\Hello_WiFi.ino:40:

C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Adapters/BlinkerMQTT.h: In member function 'virtual char* BlinkerMQTT::lastRead()':

C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Adapters/BlinkerMQTT.h:599:17: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

     else return "";

                 ^

In file included from C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Blinker/BlinkerApi.h:67:0,

                 from C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/BlinkerESPMQTT.h:11,

                 from C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Blinker.h:122,

                 from C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\examples\Blinker_Hello\Hello_WiFi\Hello_WiFi.ino:40:

C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Blinker/BlinkerApiBase.h: In member function 'char* BlinkerBridge_key::getName()':

C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Blinker/BlinkerApiBase.h:162:29: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

                 else return "false";

                             ^

In file included from C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Blinker/BlinkerApi.h:68:0,

                 from C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/BlinkerESPMQTT.h:11,

                 from C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Blinker.h:122,

                 from C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\examples\Blinker_Hello\Hello_WiFi\Hello_WiFi.ino:40:

C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Blinker/BlinkerProtocol.h: In member function 'char* BlinkerProtocol::deviceName()':

C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Blinker/BlinkerProtocol.h:75:86: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

             char * deviceName() { if (isInit) return conn->deviceName(); else return ""; }

                                                                                      ^

C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Blinker/BlinkerProtocol.h: In member function 'char* BlinkerProtocol::authKey()':

C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Blinker/BlinkerProtocol.h:76:83: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

             char * authKey()    { if (isInit) return conn->authKey(); else return "";  }

                                                                                   ^

C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Blinker/BlinkerProtocol.h: In member function 'char* BlinkerProtocol::dataParse()':

C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Blinker/BlinkerProtocol.h:162:86: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

         char* dataParse()       { if (canParse) return conn->lastRead(); else return ""; }

                                                                                      ^

In file included from C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/BlinkerESPMQTT.h:11:0,

                 from C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src/Blinker.h:122,
[... 60 lines cut ...]

 BlinkerButton Button1("btn-abc");

                                ^

C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\examples\Blinker_Hello\Hello_WiFi\Hello_WiFi.ino:48:32: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

 BlinkerNumber Number1("num-abc");

                                ^

C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src\modules\WebSockets\WebSocketsClient.cpp: In member function 'void WebSocketsClient::connectedCb()':

C:\Users\xuewenliang\Documents\Arduino\libraries\blinker-library-master\src\modules\WebSockets\WebSocketsClient.cpp:706:68: warning: 'bool BearSSL::WiFiClientSecure::verify(const char*, const char*)' is deprecated (declared at C:\Users\xuewenliang\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:173): This is deprecated AxTLS API, check https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClientSecure.h#L25-L99 [-Wdeprecated-declarations]

         if(!_client.ssl->verify(_fingerprint.c_str(), _host.c_str())) {

                                                                    ^

c:/users/xuewenliang/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\XUEWEN~1\AppData\Local\Temp\arduino_build_249605/Hello_WiFi.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'

collect2.exe: error: ld returned 1 exit status

exit status 1
```

jandrassy, 2019-03-21:
only error is
c:/users/xuewenliang/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\XUEWEN~1\AppData\Local\Temp\arduino_build_249605/Hello_WiFi.ino.elf section `.irom0.text' will not fit in region `irom0_0_seg'
everithing else is const char* to char* conversion warning

devyte, 2019-03-21:
Closing because not related to our core.
And Espressif dropped support for 512KB flash quite some time ago. We allow it as a target for legacy reasons, but you're responsible for making your app fit on it.

closed 2019-03-21 by devyte