esp8266/Arduino #5901: 2.5.0 compatibility issue with Arduino Wire API

SergioGarciaCuevasGonzalez, 2019-03-21:
The Wire library included in version 2.5.0 breaks compatibility with the common Arduino Wire library.   On version 2.5.0, TwoWire::onReceive() has the following (incompatible) signature:

`void TwoWire::onReceive(void (*)(size_t));`

The expected signature is as follows:

`void TwoWire::onReceive(void (*)(int));`

Portable I2C slave reception code won't compile anymore because of this.

SergioGarciaCuevasGonzalez, 2019-03-21:
The following sketch won't compile with esp8266 2.5.0, but will compile with other Arduino ports:

```c++
#include <Wire.h>

const uint8_t slaveAddress = 0x10;

void receiveEvent(int howMany) {
  (void) Serial.print("receiveEvent(");
  (void) Serial.print(howMany, DEC);
  (void) Serial.print("): ");
  while (Wire.available() > 1)
  {
    (void) Serial.print(Wire.read(), DEC);
    (void) Serial.print(", ");
  }
  (void) Serial.println(Wire.read(), DEC);
}

void setup() {
  Serial.begin(115200);
  Wire.begin(slaveAddress);
  Wire.onReceive(receiveEvent);
}

void loop() {
}
```

d-a-v, 2019-03-21:
Thanks for the report. A fix will be proposed.

SergioGarciaCuevasGonzalez, 2019-03-21:
I deleted the template by accident.  I see I left out a lot of required information!  Please have a code example that compiles for other Arduino ports:

```cpp
#include <Wire.h>

const uint8_t slaveAddress = 0x10;

void receiveEvent(int howMany) {
  (void) Serial.print("receiveEvent(");
  (void) Serial.print(howMany, DEC);
  (void) Serial.print("): ");
  while (Wire.available() > 1)
  {
    (void) Serial.print(Wire.read(), DEC);
    (void) Serial.print(", ");
  }
  (void) Serial.println(Wire.read(), DEC);
}

void setup() {
  Serial.begin(115200);
  Wire.begin(slaveAddress);
  Wire.onReceive(receiveEvent);
}

void loop() {
}
```

PR #5969 (2019-04-26): restore proper arduino Client:: & Wire:: API
fixes #5749
fixes #5901 
fixes #5705 
~includes #5997~
closed 2019-04-26 by d-a-v