esp8266/Arduino #5902: Backport patch from lwip2 to fix lwip1.4 behaviour with MTU<1500

mcspr, 2019-03-21:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12
- Core Version:			[2019/03/21 git  8961bba]
- Development Env:		Arduino IDE
- Operating System:		Any

### Settings in IDE

- Module:  		Generic ESP8266 Module
- lwip Variant:   **v1.4**
- Flash Mode:			dout
- Flash Size: 		4MB
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200 (serial upload only)

### Problem Description

Using ESPAsyncTCP, ESPAsyncWebServer, AsyncWebSocket, lwip v1.4
It is impossible to receive data-chunk larger than MSS size via websockets if MTU of the client device is different from 1500. I can see data is sent to the router, but fragmenting does not work properly, so router refuses to send the data further. Surprisingly, HTML is sent properly.

I was able to find one related issue:
https://github.com/esp8266/Arduino/issues/2437

This was already fixed in lwip2. Related issue + fix:
http://savannah.nongnu.org/bugs/?46384
http://git.savannah.gnu.org/cgit/lwip.git/commit/src?id=8e8571da6a6771f9d2d82bbd0b5a6c27474ce0fc
Can we apply this patch to the local tree here and rebuild `liblwip_gcc.a` so pre built option works too?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

Gist of the sketch, to avoid blowing up the issue with long strings:
https://gist.github.com/mcspr/00e95d3ea8d680aeaa04cbe8a707ba8d

### Debug Messages

In browser, using lwip v1.4:
```
SDK:2.2.2-dev(c0eb301)/Core:unix-2.6.0-dev=-194000/lwIP:1.4.0rc2/BearSSL:6778687
client connected
```
Same log with lwip2 low memory:
```
SDK:2.2.2-dev(c0eb301)/Core:unix-2.6.0-dev=-194000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-2-ga501b57/BearSSL:6778687
client connected
aaaaaaaaaaaaaa...
aaaaaaaaaaaaaa...
```

d-a-v, 2019-03-21:
Out of curiosity, what is preventing you from using lwIP-v2 ?

TBH, the primary goal of lwip2 was exactly to avoid your situation:
being able to use latest sources, update when bug-fix are posted in the original library.

mcspr, 2019-03-22:
I happened to notice this when trying to fix issues with the app running 2.3.0, and testing with other versions happen to pinpoint LWIP1 as the culprit. Since it is a (still) selectable menu option , I thought it might warrant a patch. (although, I have missed how old lwip source actually is...)

Also, TBH, I would agree that using a single library source is much easier :) Amazing work on that.

d-a-v, 2019-03-22:
@earlephilhower @devyte Should we mark lwIP-v1.4 as deprecated for the next release ?
(rtos-sdk, esp32, both have lwip-2.0.3)

devyte, 2019-03-22:
Yes, I'd like to get rid of it in next major.
But how do you deprecate a lib?

mcspr, 2020-09-01:
Resolved via https://github.com/esp8266/Arduino/pull/7436, no more lwip1.4

closed 2020-09-01 by mcspr