esp8266/Arduino #5903: After including ESP8266httpUpdate.h getting huge stacktrace

rafalfigura, 2019-03-21:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP8285  - nodemcu]
- Core Version:			[latest]
- Development Env:		[Platformio]
- Operating System:		[Windows & MacOS]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other] ?
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [other] (921600)

### Problem Description

After I include 
#include "ESP8266httpUpdate.h"
Im getting huge stacktrace about WiFiServerSecureBearSSL.h file

When I include that file, everything works as expected (need to comment all update code like t_httpUpdate_return etc)

```cpp

#ifndef Updater_h
#define Updater_h
#include "Arduino.h"
#include "ESP8266httpUpdate.h"

extern String FW_VERSION;

class Updater
{
  public:
    Updater();
    void checkAndUpdateOnce();
    private:
};
#endif

Updater.cpp:
#include "Updater.h"
Updater::Updater() {}
void Updater::checkAndUpdateOnce()
{
t_httpUpdate_return ret = ESPhttpUpdate.update(F("url"), FW_VERSION);
}
```

### Debug Messages

Here is this wall of text https://pastebin.com/DX5inLCw

earlephilhower, 2019-04-11:
The pastebin is not a stack trace but a compile error.  The MCVE given compiles fine (assuming you remove the "Updater.cpp:" line).

Can you supply a real MCVE that fails when compiled from the IDE?

rafalfigura, 2019-04-14:
@earlephilhower 
Yes, it was compiler error. I misspelled include in Updater.cpp file. Please close/remove that issue ;)

closed 2019-04-14 by earlephilhower