esp8266/Arduino #5905: ESP8266 hangs when mobile device connects to webserver

runeazn, 2019-03-21:
---------------------------- Delete above -----------------------------

### Basic Infos

- [ x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ x] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[2.0.4]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

When I connect to my local webpage from a desktop browser on as many different computers as I want, I am able to load my page. However, once I load it from my smartphone all the pages on my desktop become non-responsive and only load when I refresh my mobile page. Closing the page on my mobile has no influence. 
When desktop clients connect it ends with client disconnected. 
However when I connect from my smartphone, it ends also with client disconnect followed with New client and then hangs??
While I expect it to keep looping over loop(). I am not sure why this happens to solely my chrome smartphone browser.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>

// Replace with your network credentials
const char* ssid     = "name";
const char* password = "pass";

void setup() {
Serial.begin(115200);
Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid,password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  // Print local IP address and start web server
  Serial.println("");
  Serial.println("WiFi connected.");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  server.begin();
}

void loop() {
WiFiClient client = server.available();
 if (client) {                             // If a new client connects,
    Serial.println("New Client.");          // print a message out in the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()||client.available()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out the serial monitor
        header += c;
        if (c == '\n') {                    // if the byte is a newline character
          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println("Connection: close");
            client.println();
            
            client.println("<!DOCTYPE html><html>");
            client.println("<head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1\">");

            waitTime = (myTime.currtime-myTime.prevtime)/1000;//we needs ms->s, note that division will floor so waitTime is always more than needed, which is failsafe
            Serial.println(waitTime);
            if(waitTime>14||waitTime<0){
              waitTime = maxTime;
            }
            Serial.println(waitTime);
            myString = String(waitTime);

            // turns the GPIOs on and off
            if (header.indexOf("GET /door/opening") >= 0) {
              client.println("<meta http-equiv=\"refresh\" content=\""+ myString +";url=/door/open\"");
              Serial.println("Door is opening");
              state = "opening";
              toggleDoor(output26);
              myTime.prevtime = millis();
            }             
            //client.println("<html><head><meta http-equiv="Content-Type" content="text/html;charset=utf-8">");
            client.println("<link rel=\"icon\" href=\"data:,\">");
            // CSS to style the on/off buttons 
            // Feel free to change the background-color and font-size attributes to fit your preferences
            client.println("<style>html { font-family: Helvetica; display: inline-block; margin: 0px auto; text-align: center;}");
            client.println(".button { background-color: #4CAF50; border: none; color: white; padding: 16px 40px;"); //open button
            client.println("text-decoration: none; font-size: 30px; margin: 2px; cursor: pointer;}");
            client.println(".button2 {background-color: #555555;}"); //close button
            client.println(".button3 {background-color: #ff2d00;}</style></head>"); //stop button
            
            // Web Page Heading
            client.println("<body><h1>Garage door</h1>");
            
            // Display current state, and ON/OFF buttons for GPIO 26  
            client.println("<p>The door is " + state + "</p>");
            // If the state is off, it displays the ON button       
            if (state=="closed") {
              client.println("<p><a href=\"/door/opening\"><button class=\"button\">OPEN</button></a></p>");
            }
            client.println("</body></html>");
            
            // The HTTP response ends with another blank line
            client.println();
            // Break out of the while loop
            break;
          } else { // if you got a newline, then clear currentLine
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }
      }
    }
    // Clear the header variable
    header = "";
    // Close the connection
    //client.print('A');
    client.stop();
    Serial.println("Client disconnected.");
    Serial.println("");
  }
}

d-a-v, 2019-03-21:
If a web/tcp client doesn't close the connection, your loop gets stuck.

You can also use arduino API's timed-read
```
char c;
size_t len = client->readBytes(&c, 1); // wait, can timeout (client->setTimeout(mytimeout))
if (len == 0) // nothing has been sent for a while, consider the connection dead, close, exit
```

Or use your own timeout test
```
esp8266::polledTimeout::oneShot toolong(2000); // ms
```
when data is received
```
toolong.reset();
```
once bored:
```
if (toolong) // === if (toolong.expired())
   close, get out from this hole
```

Or better, use a `WiFiClient` list, treat them in parallel, so you don't get stuck with one client.

```
loop()
{
  if (server->hasClient())
    add-client-to-the-list

  for (auto& client: in-the-list)
  {
    check-if-dead-then-remove-it-from-the-list or
    use-data-if-available
  }
}
```

runeazn, 2019-03-21:
> If a web/tcp client doesn't close the connection, your loop gets stuck.
>
> You can also use arduino API's timed-read
>
> ```
> char c;
> size_t len = client->readBytes(&c, 1); // wait, can timeout [...]

If I look at my code, I think it implements this already but as you say this you think differently. Could you please how my code does not do this?
As far as I know, whenever it sees a client it will disconnect it rapidly, only it happens to be that for mobile devices it will also create a new client right after it disconnects, which seems weird for me.

d-a-v, 2019-03-21:
You code does (if I read it well):
```
while (connected or data-available)
{
  if (data-available)
    do-something
}
```

When the tcp/web client does nothing but leaves the connection opened, you are stuck in your loop.

runeazn, 2019-03-21:
Alright, I just disconnect twice in case it happens. Still doesn't explain why visiting the page with chrome will create 2 clients but ah well. Thanks for your help, I indeed missed the control when the client is not available but still connected (which happens with chrome where it creates then drops, become unavailable and then creates another client which cant be dropped).

d-a-v, 2019-03-21:
This is the kind of loop I describe

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiTelnetToSerial/WiFiTelnetToSerial.ino

runeazn, 2019-03-22:
Thanks!
What I don't fully understand and this is more html related or tcp.
Lets say I visit this website, then the esp reads me as a client and will go through my script. Afterwards it's done and then it will stop but create a new client.

Changing some code, I think I am getting the root cause of my misunderstanding.
So I implemented that if a client doesn't respond I just close the connection. What I see is an empty page if I refresh.

So just hypothetically speaking, is chrome causing 2 client request such that it can have one active client and once client1 is closed it can do its further requests on client2?

devyte, 2019-03-22:
Or OP could use the official webserver, or even the async webserver, instead of reinventing the wheel.

Closing as not a core issue.

closed 2019-03-22 by devyte

jandrassy, 2019-03-22:
the second connection from browser is for favicon