esp8266/Arduino #5908: any known issues in connecting to wifi in 2.5.0?

mamama1, 2019-03-22:
- Hardware:					ESP-12/nodemcuv2
- Core Version:			2.5.0
- Development Env:		Platformio
- Operating System:		Windows

### Settings in IDE

- Module:  			nodemcuv2
- Flash Mode:			dunno
- Flash Size: 		2MB
- lwip Variant:   v2 Lower Memory (2.1.2)
- Reset Method:		dunno
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

Hi

is there any obvious reason why connecting to wifi might not work with 2.5.0 but works flawlessly with 2.4.2?
with 2.5.0, wifi status starts at 6, then jumps to 1 and stays there, sometimes after many minutes it goes to 3 (connected) for one or two seconds (2 pings or so come through) and then it disconnects again.
@ 2.4.2 it connects instantly and stays connected, no packet loss whatsoever.

I wasn't able to produce a minimal sketch yet, unfortunately which reproduces this issue. My software is quite big and uses quite some amount of resources so I'm not sure whether the issue lies there. Just curious why everything's working with 2.4.2.
DATA:    [=====     ]  50.6% (used 41412 bytes from 81920 bytes)
PROGRAM: [====      ]  37.3% (used 390040 bytes from 1044464 bytes)

I'm using a crapload of libraries/includes in this project, maybe it gives a clue what could be the issue. The dnx* libraries are either local "forks" or self written libraries.

I'm not doing huge operations in async callbacks, I'm not defining any json buffers globally. I don't get any exceptions or crashes.

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPUpdate.h>
#include <ESP8266HTTPClient.h>
#include <RFM69.h>
#include <RFM69_ATC.h>
#include <Hash.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <SPI.h>
#include <dnxHC5.h>
#include <dnxESPWiFi.h>
#include <ArduinoJson.h>
#include <Wire.h>
#include "dnxExtEEPROM.h"
#include <PubSubClient.h>
#include <dnxNTPClient.h>
#include <WiFiUdp.h>
#include <vector>
#include <static/index.html.h>

I'm going to release the whole source code sooner or later anyways but I'll keep trying to create a minimal sketch which raises this issue for the sake of more easier debugging.

Thanks for any advice!

mamama1, 2019-03-22:
This is how it behaves from boot:

SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1/BearSSL:6778687
Connecting to: >apfelzimtgulasch<
using PSK: >mySecretPSK;-)<
sleep disable
del if0
usl
mode : null
mode : sta(5c:cf:7f:cc:23:92)
add if0
WiFi status: 6
WiFi status: 6
WiFi status: 6
scandone
state: 0 -> 2 (b0)
WiFi status: 6
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt
WiFi status: 6

connected with apfelzimtgulasch, channel 6
dhcp client start...
WiFi status: 6
WiFi status: 6
WiFi status: 6
WiFi status: 6
WiFi status: 6
WiFi status: 6
WiFi status: 6
WiFi status: 6
WiFi status: 6
pm open,type:0 0
WiFi status: 6

and after that, it goes on and on and on with status 6 (checking every second) and maybe eventually connects for 1 or 2 seconds as mentioned and disconnects again.

d-a-v, 2019-03-23:
A closed-source firmware update issue .
Use 2.4.2 or git master.

mamama1, 2019-03-23:
hi

thanks for your reply. do you have any reference or further information to that, under which circumstances this issue comes up? i guess it has been tried already to make espressif aware of this? do you know whether the ESP32 has the same issue?
and last: what version will git master be, why will git master work and 2.5.0 does not?

thanks again!

d-a-v, 2019-03-23:
see #5763
2.5.1 will be released next month

mamama1, 2019-03-23:
>Some boards show erratic behavior
so this is related to the board itself, like MCU manufacturing lot or is it more an issue of the environment (wifi infrastructure, I use ubiquiti)...?

d-a-v, 2019-03-23:
I have a working board and a non working board with 2.5.0. I don't know the 
reason.
Both are reliable with git master.
It has to be noted that 2.5.0 is also quite slower.

mamama1, 2019-03-23:
very interesting. is there anything on the horizon from espressif?
are there any obvious differences between the boards? esp8266ex manufacturing date, different flash chip, order date, board manufacturer, different 3.3V LDO, ...?

jandrassy, 2019-03-23:
If you have a board with this behaviour, try to flash esp_init_data_default.bin and initialize RF_CAL parameter area. The file is with AT firmware. Instructions are in AT firmware reference. If you are on Windows, use Espressif Flash Download Tool. It tunes the values in the bin file. You can find AT firmware and the tool on Espressif's site.

mamama1, 2019-03-23:
thanks, what kind of black magic is that? or rather: does this tune settings which are not touched usually when using the arduino framework?

mamama1, 2019-03-23:
OK, so I did this:
[image]

and then reflashed my firmware with core 2.5, nothing changed unfortunately.
then reflashed with 2.4.2 and it instantly connected to wifi again. is there any point in reporting to espressif or are they already aware of this/do they even care?

jandrassy, 2019-03-23:
blank.bin 0x1FB000 Initializes RF_CAL parameter area.

mamama1, 2019-03-23:
right, sorry, missed that.

redone it:

[image]

same behaviour, unfortunately...

d-a-v, 2019-03-23:
> is there any point in reporting to espressif 

read the link above.
it is an unofficial firmware that had been tested for some month before release. The issue was found after.
there's no point telling them.
If you want to test another firmware use git master there's their latest nonos-sdk-2.2.x optionally available.

mamama1, 2019-03-23:
i did read the link, but i do not quite understand. nonos-sdk is an unofficial firmware from espressif? so they are making "little" effort (because you are saying reporting doesn't make sense) + keeping the code closed source?

d-a-v, 2019-03-23:
Sorry, I was meaning unofficial "version" of the firmware.

mamama1, 2019-03-23:
ooh ok so they've probably released newer versions which in turn haven't been implemented into the arduino core yet, apart from the master branch? do you know how to use the master branch in platformio or wheter that's even possible as they seem to have own releases?

arantius, 2019-06-15:
I've got an old project I'm dusting off.  I'm experiencing exactly this.  Versions 2.3.0 through 2.4.2 work fine.  Versions 2.5.0, 2.5.1, 2.5.2 do not connect to wifi as a station.

Do above references to master mean that maybe 2.5.3 will fix this?  Either way I'm sticking with 2.4 for now.

jandrassy, 2019-06-15:
> I've got an old project I'm dusting off. I'm experiencing exactly this. Versions 2.3.0 through 2.4.2 work fine. Versions 2.5.0, 2.5.1, 2.5.2 do not connect to wifi as a station.
> 
> Do above references to master mean that maybe 2.5.3 will fix this? Either way I'm sticking with 2.4 for now.

WPA or WEP?

arantius, 2019-06-15:
WEP in this case.

jandrassy, 2019-06-15:
> WEP in this case.

here too https://github.com/esp8266/Arduino/issues/6138#issuecomment-500131554

d-a-v, 2019-06-17:
@arantius what is your WEP password like ? (passphrase, hexkey)

Can you try with, before calling `WiFi.begin(...)` :
```
WiFi.enableInsecureWEP();
```

devyte, 2019-12-18:
No feedback in a long time. Closing.

closed 2019-12-18 by devyte