esp8266/Arduino #5910: tests/host build problem - SdFat.h: No such file or directory

balee, 2019-03-24:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[N/A (tests/host issue)]
- Core Version:			[2019-03-23]
- Development Env:		[N/A (tests/host issue)]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[N/A (tests/host issue)]
- Flash Mode:			[N/A (tests/host issue)]
- Flash Size: 		[N/A (tests/host issue)]
- lwip Variant:   [N/A (tests/host issue)]
- Reset Method:		[N/A (tests/host issue)]
- Flash Frequency:	[N/A (tests/host issue)]
- CPU Frequency:		[N/A (tests/host issue)]
- Upload Using:		[N/A (tests/host issue)]
- Upload Speed:   [N/A (tests/host issue)]

### Problem Description

Building Blink example (among others) fails in tests/host in latest head, while it is working in 2.5.0.

Reproduce:

> $ git clone https://github.com/esp8266/Arduino.git
> Cloning into 'Arduino'...
> remote: Enumerating objects: 10, done.
> remote: Counting objects: 100% (10/10), done.
> remote: Compressing objects: 100% (10/10), done.
> remote: Total 21483 (delta 2), reused 2 (delta 0), pack-reused 21473
> Receiving objects: 100% (21483/21483), 71.04 MiB | 7.01 MiB/s, done.
> Resolving deltas: 100% (12813/12813), done.
> $ cd Arduino/tests/host/
> ~/Arduino/tests/host$ make D=1 ../../libraries/esp8266/examples/Blink/Blink
> Makefile:25: Cannot compile in 32 bit mode, switching to native mode
> Makefile:35: compiling in native mode
> make -f Makefile MKFLAGS=-Wextra INODIR=../../libraries/esp8266/examples/Blink/ INO=Blink bin/Blink/Blink
> make[1]: Entering directory '/home/balee/Arduino/tests/host'
> Makefile:25: Cannot compile in 32 bit mode, switching to native mode
> Makefile:35: compiling in native mode
> C   common/md5.c.o
> C   common/noniso.c.o
> C++ common/Arduino.cpp.o
> C++ common/spiffs_mock.cpp.o
> C++ common/sdfs_mock.cpp.o
> In file included from common/sdfs_mock.cpp:17:0:
> common/../../../libraries/SDFS/src/SDFS.h:36:10: fatal error: SdFat.h: No such file or directory
>  #include <SdFat.h>
>           ^~~~~~~~~
> compilation terminated.
> Makefile:216: recipe for target 'common/sdfs_mock.cpp.o' failed
> make[1]: *** [common/sdfs_mock.cpp.o] Error 1
> make[1]: Leaving directory '/home/balee/Arduino/tests/host'
> Makefile:318: recipe for target '../../libraries/esp8266/examples/Blink/Blink' failed
> make: *** [../../libraries/esp8266/examples/Blink/Blink] Error 2

The same in 2.5.0:

> ~/Arduino/tests/host$ cd ../..
> ~/Arduino$ git checkout 2.5.0
> Note: checking out '2.5.0'.
> 
> You are in 'detached HEAD' state. You can look around, make experimental
> changes and commit them, and you can discard any commits you make in this
> state without impacting any branches by performing another checkout.
> 
> If you want to create a new branch to retain commits you create, you may
> do so (now or later) by using -b with the checkout command again. Example:
> 
>   git checkout -b <new-branch-name>
> 
> HEAD is now at 951aeffa Release 2.5.0 (#5724)
> ~/Arduino$ cd tests/host/
> ~/Arduino/tests/host$ make D=1 ../../libraries/esp8266/examples/Blink/Blink
> Makefile:24: Cannot compile in 32 bit mode, switching to native mode
> Makefile:34: compiling in native mode
> make -f Makefile MKFLAGS=-Wextra INODIR=../../libraries/esp8266/examples/Blink/ INO=Blink bin/Blink/Blink
> make[1]: Entering directory '/home/balee/Arduino/tests/host'
> Makefile:24: Cannot compile in 32 bit mode, switching to native mode
> Makefile:34: compiling in native mode
> C   common/md5.c.o
> C   common/noniso.c.o
> C   ../../cores/esp8266/core_esp8266_noniso.c.o
> C   ../../cores/esp8266/spiffs/spiffs_cache.c.o
> C   ../../cores/esp8266/spiffs/spiffs_check.c.o
> C   ../../cores/esp8266/spiffs/spiffs_gc.c.o
> C   ../../cores/esp8266/spiffs/spiffs_hydrogen.c.o

... and works

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

N/A (tests/host issue)

### Debug Messages

N/A (tests/host issue)

d-a-v, 2019-03-24:
Did you `git submodule update --init` ?

balee, 2019-03-24:
Oops, of course not, I didn't reach that part in the readme maybe because it is after dealing with Arduino IDE and I didn't want to use that. My bad, sorry...

closed 2019-03-24 by balee

MichaelBrunn3r, 2019-04-18:
Can we add this as a disclaimer in the test README or is this too obvious?

d-a-v, 2019-04-18:
Would you make a PR with this information in this readme ?