esp8266/Arduino #5911: Feature Request... Synchronize the WiFi (And WiFiMulti) APIs between ESP 8266 and ESP 32

dachshund-digital, 2019-03-25:
Feature Request... Synchronize the WiFi (And WiFiMulti) APIs between ESP 8266 and ESP 32.  This would be a good thing.  I just finished writing programming logic identical on ESP 32 and ESP 8266 devices, and frankly there is not good reason the APIs differ.  It is quite user unfriendly.  I wasted a lot of time having to use #defines and such to deal with the variances between the APIs in question.

jandrassy, 2019-03-25:
what differences? the API is defined by Arduino networking libraries (Ethernet, WiFi). 
you mean this problem https://github.com/esp8266/Arduino/issues/5749?

d-a-v, 2019-04-09:
@dachshund-digital please elaborate

dachshund-digital, 2019-04-09:
You are kidding right?  Sorry, but I find your request a bit of surprise.  If you compare the ESP32 and ESP8266 APIs the deviations are obvious.

A common API, is a common API, meaning function calls named the same, function the same, and support the same parameters per API function.  This is not done at this time, meaning that custom code must be written to do the same basic tasks, those when writing code for ESP32 and ESP8266 you have to deviate code logic per device.  As was noted before hostname() versus sethostname() is one such example, there are several more just in WiFi and WiFiMulti APIs.

d-a-v, 2019-04-09:
> the official APIs should be consistent and aligned

Where are official api specifications ?