esp8266/Arduino #5913: Streaming audio(.3gp)from FIREBASE to NODEMCU to SPEAKER #REALTIME

AbeeB, 2019-03-25:
How do I fetch audio saved as .3gp in firebase and send to nodemcu and then to speaker?

#i used MIT app Inventor to create app and store the recording (.3gp) in Firebase

would I need an arduino?or some other hardware to do the operation.What would be the procedure.

Please sort this query friends...

closed 2019-03-25 by devyte

AbeeB, 2019-03-27:
can u specify the solution in detail sir

d-a-v, 2019-03-27:
Maybe [ESP8266Audio](https://github.com/earlephilhower/ESP8266Audio) is what you need. You will not get details here or there. Use discussion forum.

Quoting [Policy](https://github.com/esp8266/Arduino/blob/master/POLICY.md):

> Questions of type "**How do I**..." or "Can you please help me with..." or "Can the ESP do..." won't be handled here. Such questions should be directed at a discussion forum, like esp8266.com or stackoverflow. All issues of this type will be closed [...]

AbeeB, 2019-03-27:
Ok sir I'll see towards it
Thanku sir

freddymagne, 2021-09-07:
sad