esp8266/Arduino #5918: ESP8266 VERY SLOW TCP SOCKET TRANSFER SPEED

mipoat, 2019-03-26:
I'm trying to send data from ESP8266 to a smartphone via TCP sockets.
Hardware: NodeMCU with ESP8266MOD, various Android and iOS phones

I'm sending 1000 blocks of 1024bytes from ESP8266 to the smartphone and can achieve data rates around 1Mbit/s which I find extremely slow. Using the exact same ESP8266 firmware but a Windows10 PC or Raspberry Pi as receiver, I see data rates of 2 Mbit/s. Still not fast, but at least twice the speed I get if the receiver is a smartphone. Oddly enough the data rate is the same for iOS and Android phones.

I've tried already different buffer sizes (below and above the MTU of 1460), WiFi channels, WiFi output power without any significant change in the overal data rate.
I'd be glad if somebody could have a look at the Arduino- and Android-project to tell me if I'm doing something wrong and how to achieve faster transfer speed: https://www.dropbox.com/s/iu8egzqqskcfd ... o.zip?dl=0

I have decided to setup the smartphone as a hotspot, to rule out the WiFi router as bottleneck.
(Using the WiFi router actually reduced the speed a little bit)
If you want to try the project, please setup the Android phone as hotspot and enter the hotspot credentials in the Arduino project:
const char* ssid = "??????????";
const char* password = "??????";
const char* host = "192.168.43.1"; // please enter here the IP address of the hotspot

d-a-v, 2019-03-26:
Are Windows and RPi both on ethernet ? (esp8266 -wifi- AP -**eth**- computer)
Your phones are WiFi (esp8266 -wifi- AP -**wifi**- phone)
That might explain it. Have you tried with your phones connected to a 5G AP ?
(esp can only use 2.4G APs, some APs provide both frequencies)

Your dropbox URL is broken (can't you just copy paste your MCVE here?)

mipoat, 2019-03-26:
The phones are generating the WiFi hotspot and since the ESP can handle only 2,4GHz this was used.
There is no AP between ESP and phone.

Yes you are right: Windows and RPi were both connected via ethernet to the AP, the AP had a 2,4GHz link to the ESP.

I'm very sorry for the broken dropbox URL: https://www.dropbox.com/s/iu8egzqqskcfdy7/TCPsocketDemo.zip?dl=0

d-a-v, 2019-03-26:
You don't have the same AP on each case, so this is hard to tell.
Beside, your phone AP is not very configurable, is it ?
An easy way to check if it comes from TCP or WiFi is to make an UDP BW test.

mipoat, 2019-03-26:
I already used the AP between phone and ESP, this has dropped the datarate a little bit like 0,9 -0,8 kbit/s.

Maybe my question was bad afterall: the main problem is not the difference between phone and Windows/RPi but the overall extremely bad transfer speed from ESP to phone.
Even 2 Mbit/s are very slow.

devyte, 2019-03-26:
@d-a-v didn't the wifi stress test measure about 1Mbps throughput for our WiFiClient, and just slightly above that for the async tcp?

mipoat, 2019-04-04:
I would be very interested to understand the test-setup for a unidirectional TCP socket transfer with speeds of more than 5Mbit/s from ESP8266 to any other device using 2.5.0 arduino ES8266 package.

This would be more than double the speed I see with my extremly simple sending loop (see above).
I'm talking about very close proximity of the 2 devices and no other 2,4GHz device within reach.

d-a-v, 2019-04-04:
The 5Mbits/s result was during testing tcp stability. No useful data were transferred, data were swallowed and discarded immediately on the receiver side (so acks were not delayed) , nothing else in the sketch main loop, and option "lwIP-v2 higher bandwidth" (MSS=1460).

Also (and I don't know if it has a Bw effects) 2.5.0 is slower than 2.4.2 or git.

kdervis, 2020-09-02:
Hi, can i see code please