esp8266/Arduino #5920: Watchdog Error when restarting the ESP8266

StefanVerresen, 2019-03-26:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP8266 device]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

I have made a sketch that connects my NodeMCU to a MQTT-broker, the NodeMCU generates a topic and stores it in the EEPROM. Then it subscribes to that topic and listens to all incomming data from that topic. 

Now when I try to update my sketch to also save the last data it received, It starts to bug and crash. Every time my NodeMCU receives data, it writes that data to the EEPROM, it works normally when the NodeMCU keeps on running but when I try to restart the system ( ESP.restart() ); then the watchdog crash comes up.

```cpp
-> ets Jan  8 2013,rst cause:2, boot mode:(1,6)
-> 
-> 
-> ets Jan  8 2013,rst cause:4, boot mode:(1,6)
-> 
-> wdt reset
**Nothing...**
```

I am doing the restart to replicate a failure of power to my NodeMCU.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <WiFiManager.h>
#include <PubSubClient.h>
#include <EEPROM.h>
  
const char* mqtt_server = "xxx";
String BrokerID;
char LastCommand;

WiFiClient espClient;
PubSubClient client(espClient);
WiFiManager wifiManager;

void save_lastcommand(char message){
  EEPROM.write(14, message);
  EEPROM.commit();
}

//write Topic into EEPROM (0->13)
void save_credentials(String brokercredentials){
  brokercredentials += "EduBox-" + String(random(0xfffffff), HEX);
  for (int i = 0; i <= brokercredentials.length(); i++) {
    EEPROM.write(i,brokercredentials[i]);
  }
  BrokerID = brokercredentials;
  EEPROM.commit();
  Serial.println("Credentials saved");
}

//read Topic from EEPROM (0->13) and store it into a variable
void load_credentials(){
  char variabele[14] ;
  for (int i = 0; i <= strlen(variabele); i++) {
    variabele[i] = char(EEPROM.read(i));
  }
  BrokerID = variabele;
  Serial.println("Credentials loaded");
}

//If the first place is not empty, read out the full topic
void setup_broker(){
  if(EEPROM.read(0) != 255){    
    load_credentials();
    Serial.println("You already have a broker");
  }
  else { //create a new topic and store it in the EEPROM
    save_credentials(BrokerID);
    Serial.println("Creating a new Topic...");
  }
  delay(2000);
  Serial.print("Id van broker: ");
  Serial.print(BrokerID);
  Serial.println("");
  randomSeed(micros());
}

void setup_wifi() {
 // basic wifi configuration
}

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  
  if ((char)payload[0] == 'o') {
    LastCommand = (char)payload[0];
    save_lastcommand(LastCommand); //save 'o' in the EEPROM
    digitalWrite(2, HIGH);
  } else if((char)payload[0] == 'c') {
    LastCommand = (char)payload[0];
    save_lastcommand(LastCommand); //save 'c' in the EEPROM
    digitalWrite(2, LOW);
    ESP.restart(); //replicate power of failure
  }
  else {
    Serial.print("  <-- Is not a valid command");
  }
  Serial.println("");
  delay(1000);
}

void reconnect() {
// Basic reconnecting to broker
}

void setup() {
  Serial.begin(115200);
  EEPROM.begin(512);
  pinMode(2, OUTPUT);
  load_lastcommand();
  setup_wifi();
  setup_broker();
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
}

void loop() {
  if (!client.connected()) {
    reconnect();
  }
  client.loop();
}

```

### Debug Messages

**Here below the Watchdog Crash:**

[image]

devyte, 2019-03-26:
After flashing over serial, are you physically resetting the ESP by pushing the reset button? If you do not physically reset after flashing over serial, and your sketch calls ESP.restart(), the behavior you describe will occur.

StefanVerresen, 2019-03-26:
No I am not pushing the reset button physicially,
Is there a possibility to replicate a failure of power?
In my project I am trying to save every incomming command in the EEPROM and I want to get that command back when the NodeMCU had a power failure

Any suggetions to replicate that? or Is my code above working correctly?

devyte, 2019-03-27:
Ref: https://arduino-esp8266.readthedocs.io/en/latest/faq/readme.html#i-have-observed-a-case-when-esp-restart-doesn-t-work-what-is-the-reason-for-that

Closing due to user error.

devyte, 2019-03-27:
You flash your sketch over serial, you press the reset button, then you test whether your sketch works or not.

closed 2019-03-27 by devyte