esp8266/Arduino #5923: Creating IPv6 only network

tulika-agrawal, 2019-03-27:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			ESP8266
- Core Version:	        latest git
- Development Env:	Arduino IDE
- Operating System:	Ubuntu

### Settings in IDE

- Module:  		Nodemcu
- Flash Size: 		4MB
- lwip Variant:          v6 Lower Memory
- Reset Method:	nodemcu
- Flash Frequency:	[40Mhz]
- CPU Frequency:	80Mhz
- Upload Using:	SERIAL
- Upload Speed:      115200(serial upload only)

### Problem Description

I want to create a complete ipv6 network. But I,m not getting how to disable the ipv4 addresses. Can you please provide the solution for this?

#include "lwip/opt.h"
#include "lwip/def.h"

#include "lwip/ip4_addr.h"
#include "lwip/ip6_addr.h"
#define LWIP_IPV4 0 
#define LWIP_IPV6 1 
#define LWIP_DHCP 0 
const char *ssid ="ssid_name";
const char *password = "pass";

void setup() {
  // put your setup code here, to run once:

  Serial.begin(115200);
  WiFi.begin(ssid,password);
  while(WiFi.status() != WL_CONNECTED ){
      delay(500);
      Serial.println("Connecting to wifi");
    }
    Serial.println("Connected to Wifi");
    for (bool configured = false; !configured;) {
    for (auto addr : addrList)
      if ((configured = ! addr.isLocal()
                         &&addr.isV6() // uncomment when IPv6 is mandatory
                         && addr.ifnumber() == STATION_IF
          )) {
        break;
      }
      Serial.println('.');
    delay(500);
      }
    for (auto a : addrList) {
    Serial.printf("IF='%s' IPv6=%d local=%d hostname='%s' addr= %s\n",
               a.ifname().c_str(),
               a.isV6(),
               a.isLocal(),
               a.ifhostname(),
               a.toString().c_str());
    }
}

void loop() {
  // put your main code here, to run repeatedly:

}

d-a-v, 2019-03-28:
We can't right now.
nonos-sdk on which this core is only based on and needs IPv4.
Nothing prevents your sketch from using IPv6 only, everything should work.

```
#define LWIP_IPV4 0 
#define LWIP_IPV6 1 
#define LWIP_DHCP 0 
```
This is useless, those defines must be changed in `tools/sdk/lwip2` submodule and you will not be able to properly compile it with `LWIP_IPV4 0`.

```
while(WiFi.status() != WL_CONNECTED ){
      delay(500);
      Serial.println("Connecting to wifi");
    }
```
I think you don't need this if you aren't going to use IPv4 at all.

tulika-agrawal, 2019-04-01:
Yes, Sir. I'm getting IPv6 address and able to connect to WiFi. But I am unable to understand your comment exactly "Nothing prevents your sketch from using IPv6 only, everything should work". If it won't compile correctly with LWIP_IPV4 0, then how can we create IPv6 only.