esp8266/Arduino #5925: Is there a possibility to connect a ESP8266 to the Google Firestore without using Google IOT platform

StefanVerresen, 2019-03-28:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP8285 device]
- Core Version:			[28/03]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[other]
- Flash Size: 		        [4MB]
- Reset Method:		[nodemcu]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:              [115200]

### Problem Description

I am working on a project whereby I want to connect my ESP8266 (NodeMCU) to the google Firestore (not the RT-Firebase) to read / retrieve some data and maybe write something to it aswell.

I have already tried with a simple REST API GET-request but That kept on failing. Any guides, help, examples?

Here below my code of a simple Get-request that keeps on giving an error

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

void setup() {
 // standard setup
}

void loop() {
  if(WiFi.status()== WL_CONNECTED){

  http.begin("https://firestore.googleapis.com/v1beta1/projects/edubox-49528/databases/(default)/documents/nodes?&key=[API-KEY]");
  int httpCode = http.GET(); //Send the request
  Serial.println(httpCode);
  if (httpCode > 0) {
    String payload = http.getString(); //Get the request response payload
    Serial.println(payload); //Print the response payload
  }
  else {
    Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
  }
  http.end();
  } else {
    Serial.println("Error in WiFi connection");   
  }
  delay(10000);

}

```

The error:
### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
[HTTP] GET... failed, error: connection refused
```

closed 2019-03-29 by devyte

StefanVerresen, 2019-03-29:
I do not get why this is closed...

liebman, 2019-03-29:
Probably this: https://github.com/esp8266/Arduino/blob/master/POLICY.md. See number 3.