esp8266/Arduino #5926: Connect to wifi and upload to thingSpeak from ESP8266 with info stored in SPIFFS after power outage

gbenna, 2019-03-29:
I have an ESP8266 sketch that will allow users to input their ip address and password and their thingSpeak write api when they power up the device (ESP8266 connected to Arduino UNO v3).  I am using the attached code which I use the Arduino IDE to send to device.  I use Flash Size: 512K (128K SPIFFS) when uploading to the ESP8266 because of the SPIFFS in the code and the FS is mounted and the custom parameters are supposedly saved to the config file. Anyway the serial printer says they are.  
Once the ip address and password along with api.thingspeak.com and write api are entered and saved the ESP8266 restarts and connects and starts to upload the two sensor readings to ThingSpeak.  This is great!!
However,  I want my project to be self sufficient (batteries and solar charger) so I am trying to cut power usage as much as possible.  The code that is on the Arduino turns on the ESP8266 every hr. when the pool is slowly dropping because of evaporation and every two minutes when the pool is filling.  
The problem is that when I cut power to the ESP8266 and restart it nothing happens.  The blue light on the ESP8266 blinks but it does not connect to the wifi and to thingSpeak and upload the sensor readings.  NOTHING HAPPENS.
So my question is:
How do is connect to the wifi inside the (void) loop and get the information stored in the SPIFFS to connect to ThingSpeak and upload to my write API.  I don't understand why store information in the SPIFFS if you can't retrieve it.
Also I'm not really sure how to use the WiFi reset pin 13.

Here is my code.

    
   ```
 #include <FS.h>
    
    #include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
    
    //needed for library
    #include <DNSServer.h>
    #include <ESP8266WebServer.h>
    #include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
    #include <ArduinoJson.h>
    
    //NEW STUFF START
    
    char apiKey[20]="";
    WiFiClient client;
    
    char defaultHost[100] = "api.thingspeak.com";  //Thing Speak IP address (sometime the web address causes issues with ESP's :/
        long itt = 500;
        long itt2 = 500;
    
    const byte wifiResetPin = 13;
    int interruptPinDebounce = 0;
    long debouncing_time = 1000;
    volatile unsigned long wifiResetLastMillis = 0;
    
    
    bool shouldSaveConfig = false;
    
    void saveConfigCallback () {
      Serial.println("Should save config");
      shouldSaveConfig = true;}
    
      void handleWifiReset(){
        if(millis()<wifiResetLastMillis){
          wifiResetLastMillis = millis();
        }
        if((millis() - wifiResetLastMillis)>= debouncing_time){
          Serial.println("Clearing WiFi data resetting");
          WiFiManager wifiManager;
          wifiManager.resetSettings();
          SPIFFS.format();
          ESP.reset();
          delay(1000);
        }
        wifiResetLastMillis = millis();
      }
      
    void setup() {
      WiFiManager wifiManager;
        // put your setup code here, to run once:
        Serial.begin(115200);
        pinMode(wifiResetPin, INPUT_PULLUP);
        attachInterrupt(digitalPinToInterrupt(wifiResetPin), handleWifiReset,FALLING);
    
        //NEW STUFF START
        //clean FS, for testing
      //SPIFFS.format();
    
      //read configuration from FS json
[... 121 lines cut ...]
            String postStr = apiKey;
            postStr +="&field1=";
            postStr += String(itt);
            postStr +="&field2=";
            postStr += String(itt2);
            postStr += "\r\n\r\n\r\n";
    
            client.print("POST /update HTTP/1.1\n");
            client.print("Host: api.thingspeak.com\n");
            client.print("Connection: close\n");
            client.print("X-THINGSPEAKAPIKEY: "+String (apiKey)+"\n");
            client.print("Content-Type: application/x-www-form-urlencoded\n");
            client.print("Content-Length: ");
            client.print(postStr.length());
            client.print("\n\n\n");
            client.print(postStr);
            Serial.println("% send to Thingspeak");
        }
        client.stop();
        Serial.println("Waiting…");
        delay(55000);
    }

```
HELP PLEASE!!

gbenna, 2019-03-29:
I added 
```
WiFiManager wifiManager;
wifiManager.autoConnect("AutoConnectAP");
    Serial.println("Connected");
```
to my void loop and now when I disconnect and power back up the ESP8266 it connects to the wifi and I am assuming it attached to thingSpeak since that is hard coded.  My biggest problem now it to access the SPIFFS files and get the thingSpeak write api so I can put it into the code and send the information to my channel.  Does anyone know where there is a good example/tutorial on how to access and retrieve data from SPIFFS?  Or if someone could write an example it would help a lot.
Serial monitor is posting stuff from the void set up area Mounting file system and adding parameters which there aren't any.  Explanations ?

devyte, 2019-03-30:
Look at the examples included with the core.
This is not the right place to ask for help with your particular project. This is an issue tracker, meant for tracking issues in the core. If you need help with usage, please readthedocs or seek peer help at a community forum.
Closing per issue POLICY doc and ISSUE_TEMPLATE instructions.

closed 2019-03-30 by devyte