esp8266/Arduino #5928: Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json

boerti71, 2019-03-29:
Hi

I can't longer download the ESP JSON as I got the following error:
Error downloading http://arduino.esp8266.com/stable/package_esp8266com_index.json

not sure what is struggling, I didn't change the link in preferences, the other JSON are working:
https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json
https://dl.espressif.com/dl/package_esp32_index.json
http://arduino.esp8266.com/stable/package_esp8266com_index.json  **=> not working**

boerti71, 2019-03-29:
when I download the zip according to the JSON file/link and including them into arduino IDE, both are genereating errors:
https://github.com/esp8266/Arduino/releases/download/2.0.0/esp8266-2.0.0.zip 
=> Specified folder/zip file does not contain a valid library

downloading and includeing zip file from: https://arduino-test.esp8266.com/Arduino/versions/2.0.0/
=> Specified folder/zip file does not contain a valid library

devyte, 2019-03-29:
I can't reproduce this, dl of the latest stable json works for me.

Closing as can't reproduce.

BTW:
> https://github.com/esp8266/Arduino/releases/download/2.0.0/esp8266-2.0.0.zip

V2.0.0? Really?

closed 2019-03-29 by devyte

boerti71, 2019-03-29:
check out the JSON file: http://arduino.esp8266.com/stable/package_esp8266com_index.json
{
  "packages": [
    {
      "maintainer": "ESP8266 Community", 
      "help": {
        "online": "http://esp8266.com/arduino"
      }, 
      "websiteURL": "https://github.com/esp8266/Arduino", 
      "platforms": [
        {
          "category": "ESP8266", 
          "help": {
            "online": "http://esp8266.github.io/Arduino/versions/2.0.0/"
          }, 
          "url": **"https://github.com/esp8266/Arduino/releases/download/2.0.0/esp8266-2.0.0.zip",** 
          "checksum": "SHA-256:00911ee9f5d4802b279fd81a5a8177fd6fef4888d04ebe93fc3952bac85d909c", 
          "name": "esp8266", 
          "version": "2.0.0", 
          "architecture": "esp8266", 
          "archiveFileName": "esp8266-2.0.0.zip", 
          "size": "5606497", 
          "toolsDependencies": [
            {
              "packager": "esp8266", 
              "version": "0.4.6", 
              "name": "esptool"
            }, 
            {
              "packager": "esp8266", 
              "version": "1.20.0-26-gb404fb9-2", 
              "name": "xtensa-lx106-elf-gcc"
            }, 
            {
              "packager": "esp8266", 
              "version": "0.1.2", 
              "name": "mkspiffs"
            }
          ],

VeeruSubbuAmi, 2019-10-26:
Find the fix here.
https://electronicsinnovation.com/fix-error-downloading-http-arduino-esp8266-com/