esp8266/Arduino #5931: ESP-12E - weird exception with Blinky

bluetiger9, 2019-03-31:
Hi,

I'm getting a weird exception *(see bellow)* when trying to run the Blinky example on a ESP-12E.

**Config**:
[image]

**ESP8266 core**: 2.5.0

**Serial Output**:
```
Exception (0):
epc1=0x40105770 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffe8c0 end: 3fffffb0 offset: 01a0
3fffea60:  402233c8 00000470 00000000 40100460  
3fffea70:  40223ba9 00000005 ffffffff 40223ba4  
3fffea80:  3ffee994 00001000 40101052 000003fd  
3fffea90:  000000ed 00000020 3fffff10 00000000  
3fffeaa0:  3ffeea34 00000000 00000003 00000000  
3fffeab0:  ffffffff ffffffff ffff0002 00000000  
3fffeac0:  00000000 00000000 00000000 00000000  
3fffead0:  00000000 00000000 00000000 00000000  
3fffeae0:  ffffffff 00ffffff 00000000 00000000  
3fffeaf0:  00000000 00000000 00000000 00000000  
3fffeb00:  00000000 00000000 00000000 00000000  
3fffeb10:  00000000 00000000 00000000 00000000  
3fffeb20:  00000000 ff000000 ffffffff ffffffff  
3fffeb30:  ffffffff ffffffff ffffffff ffffffff  
3fffeb40:  ffffffff ffffffff ffffffff ffffffff  
3fffeb50:  ffffffff ffffffff ffff00ff ffffffff  
3fffeb60:  0000000a 5f505345 36363534 00003943  
3fffeb70:  00000000 00000000 00000000 00000000  
3fffeb80:  00000000 00000000 00000000 00000000  
3fffeb90:  00000000 00000000 00000000 00000000  
3fffeba0:  00000000 00000000 00000000 00000000  
3fffebb0:  00000000 00000000 00000000 00000000  
3fffebc0:  00000000 ffffffff ffffffff ffffffff  
3fffebd0:  ffffffff ffffffff ffffffff ffffffff  
3fffebe0:  ffffffff 000001ff ffffff04 ffffff01  
3fffebf0:  ffffffff ffffffff ffffffff ffffffff  
3fffec00:  ffffffff ffffffff ffffffff ffffffff  
3fffec10:  ffffffff ffffffff ffffffff ffffffff  
3fffec20:  ffffffff ffffffff ffffffff ffffffff  
3fffec30:  ffffffff ffffffff ffffffff ffffffff  
3fffec40:  ffffffff ffffffff ffffffff ffffffff  
3fffec50:  ffffffff ffffffff ffffffff ffffffff  
3fffec60:  ffffffff ffffffff ffffffff ffffffff  
3fffec70:  ffffffff ffffffff ffffffff ffffffff  
3fffec80:  ffffffff ffffffff ffffffff ffffffff  
3fffec90:  ffffffff ffffffff ffffffff ffffffff  
3fffeca0:  ffffffff ffffffff ffffffff ffffffff  
3fffecb0:  ffffffff ffffffff ffffffff ffffffff  
3fffecc0:  ffffffff ffffffff ffffffff ffffffff  
3fffecd0:  ffffffff ffffffff ffffffff ffffffff  
3fffece0:  ffffffff ffffffff ffffffff ffffffff  
3fffecf0:  ffffffff ffffffff ffffffff ffffffff  
3fffed00:  ffffffff ffffffff ffffffff ffffffff  
3fffed10:  ffffffff ffffffff ffffffff ffffffff  
3fffed20:  ffffffff ffffffff ffffffff ffffffff  
3fffed30:  ffffffff ffffffff ffffffff ffffffff  
3fffed40:  ffffffff ffffffff ffffffff ffffffff  
3fffed50:  ffffffff ffffffff ffffffff ffffffff  
3fffed60:  ffffffff ffffffff ffffffff ffffffff  
3fffed70:  ffffffff ffffffff ffffffff ffffffff  
3fffed80:  ffffffff ffffffff ffffffff ffffffff  
3fffed90:  ffffffff ffffffff ffffffff ffffffff  
[... 289 lines cut ...]
<<<stack<<<
```

**ESP Exception Decoder output**:
```
Exception 0: Illegal instruction
PC: 0x40105770
EXCVADDR: 0x00000000

Decoding stack results
0x402233c8: dns_recv at core/dns.c line 1286
0x40100460: umm_make_new_block at /home/bluetiger/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/umm_malloc/umm_malloc.c line 1154
0x40223ba9: tcp_free_acked_segments at core/tcp_in.c line 1101
0x40223ba4: tcp_free_acked_segments at core/tcp_in.c line 1101
0x40201c97: base64_encode_block at /home/bluetiger/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/libb64/cencode.c line 59
0x40100b50: timer1Interrupt at /home/bluetiger/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_waveform.c line 255
0x4010029c: umm_assimilate_up at /home/bluetiger/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/umm_malloc/umm_malloc.c line 1175
0x40100249: umm_disconnect_from_free_list at /home/bluetiger/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/umm_malloc/umm_malloc.c line 1164
0x401001fc: pvPortMalloc at /home/bluetiger/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/heap.c line 153
```

Any idea what could cause this?

Thanks,
Attila

JiriBilek, 2019-03-31:
Please make sure you have stable power supply. Many weird resets are caused by it (my bitter experience).

bluetiger9, 2019-03-31:
@JiriBilek, already checked this. The power supply is stable, 3.3V...

JiriBilek, 2019-03-31:
You have power supply from an independent source, not from USB, I suppose.
Then you can try to clear the whole flash memory (Arduino IDE - Tools - Erase Flash - All Flash Contents)

bluetiger9, 2019-03-31:
Two more Exception (0), but with different a Sketch (eeprom_clear). The weird thing is that both at the same PC address as before (0x40105770).

*1.* Flash Size = 4M (3M SPIFSS)
```
Exception 0: Illegal instruction
PC: 0x40105770
EXCVADDR: 0x00000000

Decoding stack results
0x40100460: vPortFree at /home/bluetiger/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/heap.c line 59
```

```
Exception (0):
epc1=0x40105770 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffe8c0 end: 3fffffb0 offset: 01a0
3fffea60:  402233c8 00000470 00000000 40100460  
3fffea70:  40223ba9 00000005 ffffffff 40223ba4  
3fffea80:  3ffee994 00001000 40101052 000003fd  
3fffea90:  000000ed 00000020 3fffff10 00000000  
3fffeaa0:  3ffeea34 00000000 00000003 00000000  
3fffeab0:  ffffffff ffffffff ffff0002 00000000  
3fffeac0:  00000000 00000000 00000000 00000000  
3fffead0:  00000000 00000000 00000000 00000000  
3fffeae0:  ffffffff 00ffffff 00000000 00000000  
3fffeaf0:  00000000 00000000 00000000 00000000  
3fffeb00:  00000000 00000000 00000000 00000000  
3fffeb10:  00000000 00000000 00000000 00000000  
3fffeb20:  00000000 ff000000 ffffffff ffffffff  
3fffeb30:  ffffffff ffffffff ffffffff ffffffff  
3fffeb40:  ffffffff ffffffff ffffffff ffffffff  
3fffeb50:  ffffffff ffffffff ffff00ff ffffffff  
3fffeb60:  0000000a 5f505345 36363534 00003943  
3fffeb70:  00000000 00000000 00000000 00000000  
3fffeb80:  00000000 00000000 00000000 00000000  
3fffeb90:  00000000 00000000 00000000 00000000  
3fffeba0:  00000000 00000000 00000000 00000000  
3fffebb0:  00000000 00000000 00000000 00000000  
3fffebc0:  00000000 ffffffff ffffffff ffffffff  
3fffebd0:  ffffffff ffffffff ffffffff ffffffff  
3fffebe0:  ffffffff 000001ff ffffff04 ffffff01  
3fffebf0:  ffffffff ffffffff ffffffff ffffffff  
3fffec00:  ffffffff ffffffff ffffffff ffffffff  
3fffec10:  ffffffff ffffffff ffffffff ffffffff  
3fffec20:  ffffffff ffffffff ffffffff ffffffff  
3fffec30:  ffffffff ffffffff ffffffff ffffffff  
3fffec40:  ffffffff ffffffff ffffffff ffffffff  
3fffec50:  ffffffff ffffffff ffffffff ffffffff  
3fffec60:  ffffffff ffffffff ffffffff ffffffff  
3fffec70:  ffffffff ffffffff ffffffff ffffffff  
3fffec80:  ffffffff ffffffff ffffffff ffffffff  
3fffec90:  ffffffff ffffffff ffffffff ffffffff  
3fffeca0:  ffffffff ffffffff ffffffff ffffffff  
3fffecb0:  ffffffff ffffffff ffffffff ffffffff  
3fffecc0:  ffffffff ffffffff ffffffff ffffffff  
3fffecd0:  ffffffff ffffffff ffffffff ffffffff  
3fffece0:  ffffffff ffffffff ffffffff ffffffff  
3fffecf0:  ffffffff ffffffff ffffffff ffffffff  
3fffed00:  ffffffff ffffffff ffffffff ffffffff  
3fffed10:  ffffffff ffffffff ffffffff ffffffff  
3fffed20:  ffffffff ffffffff ffffffff ffffffff  
3fffed30:  ffffffff ffffffff ffffffff ffffffff  
3fffed40:  ffffffff ffffffff ffffffff ffffffff  
3fffed50:  ffffffff ffffffff ffffffff ffffffff  
3fffed60:  ffffffff ffffffff ffffffff ffffffff  
3fffed70:  ffffffff ffffffff ffffffff ffffffff  
[... 144 lines cut ...]
0x402053b3: memmem_P at ../../../../../../../dl/newlib-xtensa/newlib/libc/sys/xtensa/string_pgmspace.c line 201
0x40205a36: _dtoa_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdlib/dtoa.c line 405
0x40205a1b: _dtoa_r at ../../../../../../dl/newlib-xtensa/newlib/libc/stdlib/dtoa.c line 420
0x401001fc: app_entry() at /home/bluetiger/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 246
```

Using `esptool.py`, I verified that the flash content is correct:
```
$ python3 esptool.py --port /dev/ttyUSB0 verify_flash --diff yes 0x00000 /tmp/arduino_build_954934/eeprom_clear.ino.bin
esptool.py v2.7-dev
Serial port /dev/ttyUSB0
Connecting......
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:45:66:c9
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Verifying 0x40340 (262976) bytes @ 0x00000000 in flash against /tmp/arduino_build_954934/eeprom_clear.ino.bin...
-- verify OK (digest matched)
Hard resetting via RTS pin...
```

bluetiger9, 2019-03-31:
And one more, this time 9, LoadStoreAlignmentCause:
```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x40105776
EXCVADDR: 0x00000705

Decoding stack results
0x40201d5b: init_done() at /home/bluetiger/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/core_esp8266_main.cpp line 184
```
The PC address (`0x40105776`), is different, but is pretty close to the previous one (`0x40105770`).

Serial output:
```
Exception (9):
epc1=0x40105776 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000705 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffe8c0 end: 3fffffb0 offset: 01a0
3fffea60:  402234dc 00000470 00000000 401003d8  
3fffea70:  40223cb9 00000005 ffffffff 40223cb4  
3fffea80:  3ffeea54 00001000 4010109a 000003fd  
3fffea90:  000000ed 00000020 3fffff10 00000000  
3fffeaa0:  3ffeeaf4 00000000 00000003 00000000  
3fffeab0:  ffffffff ffffffff ffff0002 00000000  
3fffeac0:  00000000 00000000 00000000 00000000  
3fffead0:  00000000 00000000 00000000 00000000  
3fffeae0:  ffffffff 00ffffff 00000000 00000000  
3fffeaf0:  00000000 00000000 00000000 00000000  
3fffeb00:  00000000 00000000 00000000 00000000  
3fffeb10:  00000000 00000000 00000000 00000000  
3fffeb20:  00000000 ff000000 ffffffff ffffffff  
3fffeb30:  ffffffff ffffffff ffffffff ffffffff  
3fffeb40:  ffffffff ffffffff ffffffff ffffffff  
3fffeb50:  ffffffff ffffffff ffff00ff ffffffff  
3fffeb60:  0000000a 5f505345 36363534 00003943  
3fffeb70:  00000000 00000000 00000000 00000000  
3fffeb80:  00000000 00000000 00000000 00000000  
3fffeb90:  00000000 00000000 00000000 00000000  
3fffeba0:  00000000 00000000 00000000 00000000  
3fffebb0:  00000000 00000000 00000000 00000000  
3fffebc0:  00000000 ffffffff ffffffff ffffffff  
3fffebd0:  ffffffff ffffffff ffffffff ffffffff  
3fffebe0:  ffffffff 000001ff ffffff04 ffffff01  
3fffebf0:  ffffffff ffffffff ffffffff ffffffff  
3fffec00:  ffffffff ffffffff ffffffff ffffffff  
3fffec10:  ffffffff ffffffff ffffffff ffffffff  
3fffec20:  ffffffff ffffffff ffffffff ffffffff  
3fffec30:  ffffffff ffffffff ffffffff ffffffff  
3fffec40:  ffffffff ffffffff ffffffff ffffffff  
3fffec50:  ffffffff ffffffff ffffffff ffffffff  
3fffec60:  ffffffff ffffffff ffffffff ffffffff  
3fffec70:  ffffffff ffffffff ffffffff ffffffff  
3fffec80:  ffffffff ffffffff ffffffff ffffffff  
3fffec90:  ffffffff ffffffff ffffffff ffffffff  
3fffeca0:  ffffffff ffffffff ffffffff ffffffff  
3fffecb0:  ffffffff ffffffff ffffffff ffffffff  
3fffecc0:  ffffffff ffffffff ffffffff ffffffff  
3fffecd0:  ffffffff ffffffff ffffffff ffffffff  
3fffece0:  ffffffff ffffffff ffffffff ffffffff  
3fffecf0:  ffffffff ffffffff ffffffff ffffffff  
3fffed00:  ffffffff ffffffff ffffffff ffffffff  
3fffed10:  ffffffff ffffffff ffffffff ffffffff  
3fffed20:  ffffffff ffffffff ffffffff ffffffff  
3fffed30:  ffffffff ffffffff ffffffff ffffffff  
3fffed40:  ffffffff ffffffff ffffffff ffffffff  
3fffed50:  ffffffff ffffffff ffffffff ffffffff  
3fffed60:  ffffffff ffffffff ffffffff ffffffff  
3fffed70:  ffffffff ffffffff ffffffff ffffffff  
[... 268 lines cut ...]
3ffffe40:  40001f46 00000009 feefeffe feefeffe  
3ffffe50:  61313539 61666665 40001e00 3ffffea1  
3ffffe60:  3fffff20 00000007 feefeffe 40001e4b  
3ffffe70:  3ffffea0 00000000 00000000 00000000  
3ffffe80:  00000000 00000000 0000001f 4010139d  
3ffffe90:  00000000 3fffff00 3fffff00 3fffff2e  
3ffffea0:  40002514 3fffdd3c 00000002 3fffffb0  
3ffffeb0:  3fffff20 00000008 00000008 00000001  
3ffffec0:  00000000 951aeffa 00000000 3ffffe58  
3ffffed0:  00000000 00004a01 00000000 fffffffe  
3ffffee0:  00000000 400042db ffffffff 4010f098  
3ffffef0:  40004b31 40001878 00000002 3fffffb0  
3fffff00:  4010f15c 951aeffa 00000000 20000000  
3fffff10:  00040334 00000004 3fffff01 3ffe8500  
3fffff20:  400304e9 4010016c 3ffe8500 000002f0  
3fffff30:  00000000 0000002d 00000000 3ffffec2  
3fffff40:  4010f355 000000e3 3fffff50 4010f35a <
3fffff50:  00000009 d1fd28ef ffffffff 00000000  
3fffff60:  59151fc5 df7ecbe7 88413307 ededb0d6  
3fffff70:  009dbddd a59d6d21 4d88a75f 8df8118e  
3fffff80:  fb07b730 553077e0 16097f1e 1fdb0092  
3fffff90:  ea69e82e 05f02b2b d8e1d0a6 c23fa235  
3fffffa0:  ef172cbe a75f2e44 ae08b695 e2a7509f  
<<<stack<<<
```

JiriBilek, 2019-04-01:
It is really strange that the exception occurs nearly at the same address. Could you try another module with the same configuration, to be sure the hardware is ok?

bluetiger9, 2019-04-01:
It is not excluded to be a hardware problem. The board draw about 390mA @ 3.3V, which is a little bit much. Unfortunately, I don't have a spare module right now to test it.

I will close the issue for now, and I will it reproduces on a second board.

closed 2019-04-01 by bluetiger9

JiriBilek, 2019-04-01:
It is really like "pocket heating" :)
My ESP8266s draw typically 17 mA when no wifi runs. With wifi running they draw over 50 mA with spikes about 200 mA.

bluetiger9, 2019-04-01:
So, did an _"autopsy"_ on the module :). Apparently, the ESP8266 chip got a little bit fried :D. The resistance between the pins GND(33) and VDDA (30) of the IC is ~ 15 Ω. This, I think explains the high current consumption...