esp8266/Arduino #5934: How to disable CCA(CSMA) for esp8266

mmcruirong, 2019-03-31:
### Basic Infos

- [√] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [√] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [√] I have tested that the issue is present in current master branch (aka latest git).
- [√] I have searched the issue tracker for a similar issue.
- [√] If there is a stack dump, I have decoded it.
- [√] I have filled out all fields below.

#### Platform

- Hardware:				other
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Hi I am currently working on ESP8266(ESP-WROOM-02D). My project requires me to disable CCA (CSMA) or increase the channel detection threshold for the chip. I have searched for the solution, there are some hint here: https://electronics.stackexchange.com/questions/323911/disabling-csma-ca-protocol-in-esp8266-or-any-other-wifi-ic. (There is a way to disable it using rom_chip_v5_disable_cca() function defined at address 0x400060d0 in the boot ROM of esp8266 (/tools/sdk/ld/eagle.rom.addr.v6.ld file in esp8266 Arduino core gives idea about it.)

But I dont know how to call that ROM function in ardunio IDE. If it's not feasible in Ardunio now by calling that rom function, is there any other ways I can disable CCA? Could anyone help me with this issue?

Thanks for all the help!

d-a-v, 2019-03-31:
You can just declare and call it:
```
extern "C" void rom_chip_v5_disable_cca();

void setup()
{
  rom_chip_v5_disable_cca();
  ...
```

devyte, 2019-04-01:
@d-a-v is correct, because there is already an entry in the .ld with the function address.

closed 2019-04-01 by devyte