esp8266/Arduino #5936: Unable to use ESP8266 Sketch Data Upload

mikekgr, 2019-03-31:
### Basic Infos

- [Y ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [Y ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [Y ] I have tested that the issue is present in current master branch (aka latest git).
- [N ] I have searched the issue tracker for a similar issue.
- [N ] If there is a stack dump, I have decoded it.
- [ Y] I have filled out all fields below.

#### Platform

- Hardware:					[WeMos D1 mini]
- Core Version:			[latest git 31/03/2018]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows 7 pro 64bit]

### Settings in IDE

- Module:  			[Wemos D1 mini]
- Flash Mode:			[Tested: qout, dout]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[Both tested: ck, nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200, serial upload only)

### Problem Description

Detailed problem description goes here.

Impossible to use Tools => ESP8266 Sketch Data Upload
almost immediately I got the error: SPIFFS Error: esptool not found!

Arduino: 1.8.8 (Windows 7), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled, nodemcu, 26 MHz, 40MHz, QIO (fast), 4M (1M SPIFFS), 2, nonos-sdk 2.2.2-190313 (testing), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

SPIFFS Error: esptool not found!

Please help me to solve the problem.

Thanks and Best Regards,
Mike Kranidis

earlephilhower, 2019-04-01:
You need to update the uploader. There is an open but to update the discs and shows where you can download the new version.

mikekgr, 2019-04-01:
> You need to update the uploader. There is an open but to update the discs and shows where you can download the new version.

Dear @earlephilhower  Thanks a lot for your reply.
I am not sure that I understood what I need to do and how can update the things. Can you give me more details?

Thanks.

Jeroen88, 2019-04-01:
@earlephilhower I have the same issue. I have updated the updater to the .py version.I also did a git submodule update --init.
I suspect that the filename checking in the data upload java source is not compliant with the changed updater, but I am not familiar enough with java to do a PR. Could you take a look?

mikekgr, 2019-04-01:
after having checked I found this thread:
https://github.com/esp8266/arduino-esp8266fs-plugin/issues/54

where this zip is working for me:
https://github.com/esp8266/arduino-esp8266fs-plugin/files/2936951/ESP8266FS.zip

Thanks

Craig1516, 2019-04-07:
@ earlephilhower and @ mikekgr:  Congratulations (mike) and thanks for the zip (Mike and Earl).  So are you using the Sketch Data Upload from the Arduino IDE menu Tools/ESP8266 Sketch Data Upload?  I have included the esp8266.jar file in the Arduino file structure (found  through "File/Preferences/Sketchbook".  I have a new Arduino load (1.8.9) today.  And I've got nothing for a menu selection to invoke Sketch Data Upload.   I had this all working yesterday, but updated ArduinoIDE today... all gone.  Thanks.

mikekgr, 2019-04-07:
Dear @Craig1516,
I use Arduino 1.8.8 ( Windows 7 OS ) and I have the proper menu for Sketch Data Upload. If you don't check that the relevant jar file is copied to the correct path. If did all perfect and still no Sketch Data Upload menu then perhaps something wrong with Arduino 1.8.9 ?
Check them all and report back.

Craig1516, 2019-04-07:
Dear @mikekgr,  There were two paths for me:  
1. C:\Users\Craig\Documents\Arduino\ESP8266FS\tool\esp8266fs.jar
2.  C:\Program Files (x86)\Arduino\ESP8266FS\tool\esp8266fs.jar

I tried both in 1.8.9, but nothing is there... or I'm looking in the wrong places.    So easy befoe as Sketch Data Upload would just jump on the menu bar and I'd forget about it.  Well now I've got a SPIFFS file that needs to be managed.    

Maybe  I slide back to a 1.86 or 1.84 to see how it goes?  Perhaps? Thanks for commenting.

Craig1516, 2019-04-07:
@mikekgr  I did find that in Arduino 1.8.9 that a sketch is able to build a json file.  I can't remember exactly but I don't think this ability existed in 1.8.1 as any sketch that was looking for a json file just would not compile.  So maybe now with 1.8.9 and dependent library updates, Sketch Data Upload is considered obsolete.  I've spent days looking.  So for the benefit of others I leave this comment.  Thx.

earlephilhower, 2019-04-11:
The uploader .java file is present now in the esp8266fs release web page.

@Jeroen88 you'll need to do
````
git submodule init
git submodule update
````
The `init` alone just makes an empty directory, while the `update` will fetch from the separate tools' repositories into the build tree.

Jeroen88, 2019-04-11:
@earlephilhower That did not work. I did the init & update but still:

`SPIFFS Error: esptool not found!`

I did a fresh install of the core 10 days ago or so. I am using Arduino IDE 1.8.7. In the tools menu the item 'ESP8266 Data Upload' is present. I am using board 'NodeMCU 1.0 (ESP-12E module)', 4M (1M SPIFFS).

Uploading a sketch is no problem, so the ESP tool is present in the right directory.

Any ideas?

Pilooz, 2019-06-05:
Hi,
I don't know if that could help, but, I faced the problem of `SPIFFS Error: esptool not found!`with Arduino IDE 1.8.9.
My version of `esp8266fs.jar` was compiled in January the 12 (this is the 0.2.0 version I thinks) but was not working.
As @mikekgr said above, I replace my version by this one :
https://github.com/esp8266/arduino-esp8266fs-plugin/files/2936951/ESP8266FS.zip
 and restart Arduino IDE. And... All is working now.

#### Environnement : 
```
- Arduino IDE 1.8.9
- LOLIN(WEMOS) D1 Mini
- Flash Size : 4M (3M SPIFFS)
- version of SPIFFS uploader tool : 
https://github.com/esp8266/arduino-esp8266fs-plugin/files/2936951/ESP8266FS.zip
```
Hope this helps

HristoTraykov18, 2019-06-07:
Hello,

in Arduino IDE 1.8.9, installing the 0.4.0 esp8266fs plugin version (the way it's instructed in the documentation) fixed it for me.

Hope that helps someone

chuck71, 2019-07-06:
I had the same issue, but after couple miserable days trying to figure out, I can conlude that Arduino IDE 1.8.9 works perfectly fine only with esp8266fs plug ins version 0.4.0

BOBILLIER, 2019-07-08:
Hi. I can't download any files to my ESP8266 spiffs memory . I have same configuration than chuck71, but i working with esp8266-12 (4M-3M SPIFFS). I program it , with simple USB-Serial bridge with just TX and RX line (no hand-check signals). Normally it's enough for download sketch, but i'm not sure it's enough for esp8266fs.jar. 
Error Message:

> 
[SPIFFS] data    : C:\Users\bobiller\Documents\Arduino\Test\Test_FTPServer_Dac_Bat31\data
[SPIFFS] size    : 3052
[SPIFFS] page    : 256
[SPIFFS] block   : 8192
/cours_grafcet.pdf

[SPIFFS] upload  : C:\Edruino\Temp/Test_FTPServer_Dac_Bat31.spiffs.bin
[SPIFFS] address  : 0x100000
[SPIFFS] reset    : ck
[SPIFFS] port     : COM36
[SPIFFS] speed    : 115200
[SPIFFS] python   : C:\Users\bobiller\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1\python.exe
[SPIFFS] uploader : C:\Users\bobiller\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\tools\upload.py

esptool.py v2.6

Serial port COM36

Connecting........_____....._____....._____....._____....._____....._____....._____

Traceback (most recent call last):

  File "C:\Users\bobiller\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\tools\upload.py", line 25, in <module>

    esptool.main(fakeargs)

  File "C:/Users/bobiller/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2653, in main

    esp.connect(args.before)

  File "C:/Users/bobiller/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 468, in connect

    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))

esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

SPIFFS Upload failed!

d00m3dd00d, 2019-07-12:
Placing the .zip into %userdir%]Documents\Arduino\tools\ESP8266FS fixed the issue for me.

File structure should appear such as:
C:\Users\YourUsername\Documents\Arduino\tools\ESP8266FS\tool\esp8266fs.jar

BOBILLIER, 2019-07-15:
You  are right. Thanks. 
One another thing i forget to do , is to place ESP in program mode , like when i try to download a sketch to it.

earlephilhower, 2020-01-01:
Closing as fixed.  We've documented the issue in the latest release, and the SDU is updated in @me-no-dev 's repo.

closed 2020-01-01 by earlephilhower