esp8266/Arduino #5938: Can't set sntp_update_delay

bperrybap, 2019-04-01:
#### Platform

- Hardware:				[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			2.5.0 installed from the IDE boards manager
- Development Env:		Arduino IDE 1.8.9
- Operating System:		Linux Mint 17.3

### Settings in IDE

- Module/board:  			LOLIN(WEMOS) D1 R2 & mini

I am doing some testing with time.c functions and ntp and I'd like to be able to set the sntp update delay. Is there anyway to do this?
There is (or at least used to be)  a function to do this **sntp_set_update_delay()** in ntp.c but it does not seem to be available.
The declaration does not seem to be present in sntp.h which causes a compilation failure.
If I create the appropriate C external declaration in the sketch:
```
extern "C" void sntp_set_update_delay(uint32 ms);
```
the sketch will compile but will fail to link with:
```
undefined reference to `sntp_set_update_delay'
```
What also seems strange is that code down in ~/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/lwip/src/core/sntp.c does not seem to be used when building using the Arduino IDE but yet is included with the esp8266 package.
I don't understand the "lwip" vs "lwip2" stuff  but it appears that lwip2 is being used by the Arduino core file sntp_lwip2.c 
Was sntp_set_update_delay() dropped in lwip2 ?

It would be nice to be able to set the NTP poll/update interval.
Perhaps sntp_set_update_delay() could set the delay for all the servers, and sntp_setserver_update_delay() as described in issue #5879 could set the delay for individual servers.

d-a-v, 2019-04-01:
> I don't understand the "lwip" vs "lwip2" stuff but it appears that lwip2 is being used by the Arduino core file sntp_lwip2.c

lwip (selected in Tool's menu "lwIP-v1.4") is the legacy espressif patched version of original lwIP-v1.4RC2.
lwip2 is default and using the latest release of lwIP (lwIP-v2.1.2)

The sntp code you mention is an espressif addition on lwIP-v1.4's sntp code.
`sntp_set_update_delay` is not in lwIP-v2, and this delay is constant and by default set to one hour ([sources](http://git.savannah.nongnu.org/cgit/lwip.git/tree/src/include/lwip/apps/sntp_opts.h?h=STABLE-2_1_2_RELEASE#n165))
It is not part of Arduino API so we (I in fact) did not feel the obligation of porting this call.

Now everything can be discussed.

bperrybap, 2019-04-01:
I knew it was an hour interval , but I'd like to shorten that for some environments and for testing.

So it sounds like this is really a lwip sntp application API issue.
I'm assuming that the resolution of issue #5879 might resolve this, and potentially for everyone assuming the sntp_setserver_update_delay() support gets pushed and accepted upstream to be part of the official API.

In the mean time, I'm not understanding why I can't a create an extern declaration for sntp_set_update_delay() and call it like I can do for clock_gettime() which also is missing a declaration.
Is the included code down in ~/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/lwip/src/core/sntp.c not what is being used for Arduino builds?
Where do sntp_ functions like sntp_stop() etc come from as I don't see them in sntp-lwip2.c

As a fallback I guess I could run my own timeouts and call sntp_stop() and then reinitalize everything.

Part of what I'm preparing for is some testing to see how things behave during power and network failures.
Things like power failures where after power restoration, the embedded device comes up long before the WiFi router and the cable modem providing IP connectivity behind it.

d-a-v, 2019-09-30:
fixed by #6373

closed 2019-09-30 by d-a-v