esp8266/Arduino #5940: [REQUEST] WiFiMulti with BSSID...

kugelkopf123, 2019-04-02:
It would be great if WifiMulti would also works with BSSID's. If the ESP8266 can not connect to one AP via BSSID, it will use the next BSSID and so on.
To use the BSSID on ESP speeds up the WiFi connection time extremely.
Thanks!

devyte, 2019-04-02:
Are you willing to implement this yourself?
Relevant code:
[Adding an AP entry](https://github.com/esp8266/Arduino/blob/ca79f2ce39724527052319717a731b3501d5f0b2/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp#L186)
[Loop that iterates over the wifi scan results looking for the AP to connect to](https://github.com/esp8266/Arduino/blob/ca79f2ce39724527052319717a731b3501d5f0b2/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp#L95)

kugelkopf123, 2019-04-05:
> Are you willing to implement this yourself?
> Relevant code:
> [Adding an AP [...]

I wish i could say yes, but i have no idea how to do this.