esp8266/Arduino #5942: WebServerSecure hangs for 61+ seconds on GET, WebServer doesn't

owendelong, 2019-04-02:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [NA] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12
- Core Version:			2.5.0
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			AdaFruit ESP8266 Huzzah Breakout
- Flash Mode:			qio
- Flash Size: 		       4MB/1MB
- lwip Variant:                v2 IPv6 lower memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:            460800

### Problem Description

I have a sketch that attempts to provide a web server via HTTPs using the ESP8266WebServerSecure library and BearSSL.

The code is mostly working as expected, but when I do a connection using SSL, I get very long delays (61+ seconds) at each step...
Connect: Get connected message from OpenSSL, then wait 61+ seconds, and receive the OpenSSL Negotiation response.
Issue HTTP GET command, then wait 61+ seconds to receive the response.

I'm using the <server>.on("/<path>", <callbackFunction>); method and I've defined an onNotFound, as well as 5 URIs that are supported.

I have each of the callback functions reporting when it is called and the delay occurs before the functions are called.

Below is output from Serial Monitor. I'm printing the mills() value each time I finish the loop() function.

I turned on the timestamp feature in the Monitor as a convenient way of marking when the delay ended, nothing more. Note the elapsed time between Loop...189402 and the next iteration of loop() at 251410. Also, note that the timestamp difference between entry to the callback function (02:51:38.051) and the next loop iteration is 00:00:01.252. That's 1.252 seconds in my code and 60.488 seconds doing god knows what inside the system.

`
Loop...184210
Loop...185210
Loop...189402
02:51:38.051 -> Servicing getTemp
02:51:39.303 -> Loop...251410
02:51:40.301 -> Loop...252411
02:51:41.297 -> Loop...253412
`

My sketch (below) is still under early development, so lots of incomplete pieces and dangling things.

One final note... If I replace ESP8266WebServerSecure with ESP8266WebServer, the problem goes away. Also of note, this is using IPv6 only (host only has an AAAA record), even though it's dual stack on a dual-stack network.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Wire.h>
#include <Adafruit_MCP9808.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServerSecure.h>
#include <ESP8266WebServer.h>
#include <PolledTimeout.h>
#include <AddrList.h>
#include <lwip/dns.h>
[... 187 lines cut ...]
    "application/json",
    temp
  );
  digitalWrite(PROCESS_LED, HIGH);
}

void serviceLimits() {
  Serial.println("Servicing Limits");
}

void serviceSubscribe() {
  Serial.println("Servicing Subscribe");
}

void serviceUnSubscribe() {
  Serial.println("Servicing UnSubscribe");
}

```

### Debug Messages

```
See above
```

owendelong, 2019-04-02:
Additional debugging information:

`
03:44:09.250 -> Initializing...
03:44:09.359 ->     pins...done.
03:44:09.396 ->     WiFi
03:44:09.396 ->         trying DeLong-BGN-03......CONNECTED!
03:44:10.843 ->     Wifi Connected to DeLong-BGN-03 with address(es):
03:44:10.914 ->         IF='st' Family=INET LinkLocal=0 addr=192.159.10.215
03:44:10.952 ->         IF='st' Family=INET6 LinkLocal=1 addr=fe80::827d:3aff:fe5d:5d78
03:44:11.025 ->         IF='st' Family=INET6 LinkLocal=0 addr=2620::930:0:827d:3aff:fe5d:5d78
03:44:11.137 ->     Server...done.
03:44:11.249 ->     Sensor(s)...done.
03:44:11.286 -> Loop...1977
03:44:12.259 -> Loop...2980
03:44:13.259 -> Loop...3981
03:44:14.283 -> Loop...4981
03:44:15.272 -> Loop...5982
03:44:16.257 -> Loop...6982
03:44:17.252 -> Loop...7982
03:44:18.243 -> Loop...8983
03:44:19.272 -> Loop...9983
// Here's where I connected with the openssl s_client... This one is interestingly faster than usual
// at only 4 seconds.
03:44:23.453 -> Loop...14178
03:44:24.451 -> Loop...15178
03:44:25.484 -> Loop...16178
// Here's where I sent the GET command (03:44:25.484)
03:45:26.941 -> Servicing UnSubscribe
// Notice the full minute+ elapsed between the GET and the Servicing message
03:45:27.927 -> Loop...78656
03:45:28.942 -> Loop...79657
03:45:29.967 -> Loop...80658
`

devyte, 2019-04-02:
Obvious questions:
1. what happens if you don't use ipv6 and stick to ipv4?
2. What happens at 160MHz cpu freq?

And for the love of silicon, remove that loop delay(1000), or at least reduce it to something sane, at most 10ms or so. If you need to do something every x seconds, please use a periodic polledTimeout.
Also, and for the same love: don't use assignment constructor! Use normal constructor for the tempsensor. Assignment constructor should only ever be used if you're absolutely sure it is implemented correctly, or if you're absolutely sure that the object internals are trivial.

owendelong, 2019-04-03:
Took some doing to enable IPv4 on the network in question, but after doing so, the results are identical for IPv4. (This is intended to be an IPv6 only network).

The behavior is identical even with the lwipv2 non-IPv6 higher bandwidth variant.

The behavior is identical regardless of the delay() being 10 or 1000.

The behavior is identical with a 160Mhz or an 80Mhz clock speed.

As for the temperature sensor assignment constructor:

1. I am sure it's implemented correctly.
2. The object internals are incredibly trivial.
3. It's the example the library author provides, so take it up with Lady Ada if you care that much. That part of the software is working fine, so I'm not going to worry about it.

owendelong, 2019-04-03:
This is on an IPv6 only network, so test number 1 took some effort. The results are identical.

I don't have a 160Mhz board, or if the ESP8266 breakout I'm using can be switched to 160Mhz, I'm unaware of how to do so.

I actually don't need to do anything every x seconds, I put the long delay in to reduce the verbosity of the printing to a sane rate. I've used an alternative mechanism for that now and reduced to 10.

As for the assignment constructor, that's the example supplied by the authors of the object and the object internals are extremely trivial.

The change to delay(), however, doesn't effect the behavior in question. During the freezes I'm talking about, the loop stops executing and the delay statement isn't re-entered until the freeze thaws.

The behavior has become more consistent. It's about a 4 second delay from the completion of the three-way handshake to the display of the SSL information. The loop then begins behaving normally until I send the GET request. At that point, I get a 60+ second delay before the appropriate registered callback function is called.

It also appears that the web server in question is somewhat single-threaded. If I start a new connection while the previous one is hung, the new one waits for the previous  one to complete before being processed at all beyond the three-way handshake.

devyte, 2019-04-03:
> I don't have a 160Mhz board
All ESP8266 boards can be set to clock at 160MHz. Power use goes up, of course. In the Arduino IDE menu there's an option for cpu speed somewhere.

> The change to delay()
I mentioned it because it can mess you up. You have a server set up. During that 1000 delay, if you happen to make a connection and start requesting, the incoming buffers can't be passed from the wifi stack to the app layer (your server), and so mem can blow up internally,  or overflow, or subtly corrupt, etc.

> assignment constructor
It is bad practice that comes from other languages. If the object has any kind of internal init/deinit, including dynamic alloc, hw init/release, etc, you could end up with random behavior if the class doesn't implement the operator. I mention it to raise awareness that, in general, it should not be used, especially in Arduino world where classes are often implemented without all methods, double-especially for Adafruit code. Best case: it works fine. Worst case: it works sometimes depending on the alignment of the stars and planets and mood swings of the closest plant.
There is no sane case where using that makes sense. Direct construction is always best, in this case: 
`Adafruit_MCP9808 tempsensor;`

> somewhat single-threaded.
Not somewhat. Fully. There is no OS here, or threads. There are two contexts that act like cofunctions, that's it. The arduino CONT context is the setup and loop. It gets paused to allow the SYS context to unwind, where the wifi radio stuff executes. After a while execution goes back to CONT.

> about a 4 second delay
I'm not the expert in the bssl code, that would be @earlephilhower , but this sounds about right. It should be reduced with 160MHz.

The 60s thing sounds like some timeout to me, assuming that 160MHz doesn't affect it. Because IPv4 also doesn't change behavior, it sounds like it's not in lwip.

owendelong, 2019-04-03:
> On Apr 2, 2019, at 20:08 , Develo <notifications@github.com> wrote:
> 
> I don't have a 160Mhz board
> All ESP8266 boards can be set to clock at 160MHz. Power use goes up, of course. In the Arduino IDE menu there's an option for cpu speed somewhere.
> 
> The change to delay()
> I mentioned it because it can mess you up. You have a server set up. During that 1000 delay, if you happen to make a connection and start requesting, the incoming buffers can't be passed from the wifi stack to the app layer (your server), and so mem can blow up internally, or overflow, or subtly corrupt, etc.
> 
> assignment constructor
> It is bad practice that comes from other languages. If the object has any kind of internal init/deinit, including dynamic alloc, hw init/release, etc, you could end up with random behavior if the class doesn't implement the operator. I mention it to raise awareness that, in general, it should not be used, especially in Arduino world where classes are often implemented without all methods, double-especially for Adafruit code. Best case: it works fine. Worst case: it works sometimes depending on the alignment of the stars and planets and mood swings of the closest plant.
> There is no sane case where using that makes sense. Direct construction is always best, in this case:
> Adafruit_MCP9808 tempsensor;
> 
I don’t mind changing that, but I’ve looked at the source code for the library (or class if you must) and it’s not doing anything special in the constructor.

LIterally (the actual source):
/*!
 *    @brief  Instantiates a new MCP9808 class
 */
Adafruit_MCP9808::Adafruit_MCP9808() {}

The call to tempSensor.begin() does all the initialization (what little there is, basically spinning up I2C and making sure it can see an MCP9808 on the I2C bus.

> somewhat single-threaded.
> Not somewhat. Fully. There is no OS here, or threads. There are two contexts that act like cofunctions, that's it. The arduino CONT context is the setup and loop. It gets [...]
Got it… Well, that makes sense and it’s fine for what I’m doing with it.

> about a 4 second delay
> I'm not the expert in the bssl code, that would be @earlephilhower <https://github.com/earlephilhower> , but this sounds about right. It should be reduced with 160MHz.
>
> The [...]
I _NEVER_ thought this was in lwip.

I’m pretty sure it’s in the SSL routines, since that’s the change that makes the difference…

No SSL, no delays. SSL, 4 second delay between three-way handshake completion and delivery of the initial SSL data (the stuff about the certificate and negotiation that openSSL prints when you connect as a client). Then, a 60+ second delay after sending the GET to the HTTP server before it transfers to my callback function.

If I turn off the SSL and just use ESP8266WebServer instead of ESP8266WebServerSecure, then everything is about the speed I’d expect.

The 160 Mhz clock speed does not appear to make any difference to either delay, though a small gain might have occurred on the initial delay and my measurement of elapsed time for that isn’t particularly precise.

So… I’m pretty sure the problem is in the SSL implementation, not the lwip or any of the shims on top of lwip.

Unfortunately, the way stuff is put together in the Arduino environment where ESP8266WebServerSecure looks like the full web server implementation in a secure flavor vs. being an SSL shim added to the plain-text web server, even though because classes, it’s actually most likely exactly that and there are some places where that shows through (functions for handling the web server that reference BearSSL, for example).

I am, however, very very very glad to see IPv6 working on my ESP8266.

Owen

devyte, 2019-04-05:
> I _NEVER_ thought this was in lwip.
IPv6 is still experimental. If the delay had disappeared when you tested IPv4, it would have been a strong hint that the issue was in handling IPv6, which meant lwip or lwip-glue. Good thing that is not the case.

> where ESP8266WebServerSecure looks like the full web server implementation in a secure flavor
That's not how [our implementation](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/src/ESP8266WebServerSecureBearSSL.cpp) works. The [secure webserver](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/src/ESP8266WebServerSecureBearSSL.h) inherits from the non-secure one, and just adds on top of it for the secure aspects.

d-a-v, 2019-04-05:
Have you tried without `delay()` ?

```
esp8266::polledTimeout::periodicMs talkNow(1000);

void loop() {
  if (talkNow) {
      Serial.print("Loop...");
      Serial.println(millis());
  }
  /* Wait for connection or next scheduled asynchronous notification */
  /* Handle connection, if any */
  webServer.handleClient();
  /* Handle subscribers, if any */
  //delay(1000);  // no delay in the main loop! handle functions need to be called often
}
```

owendelong, 2019-04-08:
> > I _NEVER_ thought this was in lwip.
> 
> IPv6 is still experimental. If the delay had disappeared when you tested IPv4, it would have been a strong hint that the issue was in handling IPv6, which meant lwip or lwip-glue. Good thing that is not the case.

Sure, but since it was already known that the display disappeared with ESP8266WebServer, even while running IPv6, I kind of thought that pointed strongly away from lwip and towards ESP8266WebServerSecure.

> That's not how [our implementation](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/src/ESP8266WebServerSecureBearSSL.cpp) works. The [secure [...]

This doesn't surprise me, but it's obfuscated from the developer's view of the situation because inheritance happens behind the scenes. (hence my careful use of the term "looks like" rather than "is").

Nonetheless, given no significant difference in timing with delay(10) vs. delay(1000), I doubt removing delay() from loop() will change anything. I'll try it when I get home (I'm on business travel this week), but I'll be surprised if it has any effect.

earlephilhower, 2019-04-11:
@owendelong I'm still going over the longish thread here, but a few points:
. Don't `delay()` in your main loops.  It's not going to accomplish what you want and can cause connection attempts to be missed.
. Run at 160MHz if possible.  SSL is CPU intensive and the 8266 has very poor encryption performance.
. Make sure you've got the latest GIT head.  There were issues in the pre3.0 SDK version where the SDK itself would be *very* slow

I'll try the sketch (minus the delays) and see what I can see...

earlephilhower, 2019-04-11:
I'm not seeing any long pauses with Firefox pulling "/getTemp" (I faked the temp sensor as always returning 100.1).  This is git head, 160MHz, IPV4.

There is a 2-4 second delay in making the SSL connection. This is expected and normal given the SSL overhead and CPU requirements.  

````
11:22:34.742 -> Loop...44068
11:22:35.305 -> WS:ac
11:22:35.305 -> :rn 517
11:22:35.736 -> Loop...45069
11:22:35.736 -> :ref 1
11:22:35.736 -> :rd 5, 517, 0
11:22:35.736 -> :rdi 517, 5
11:22:35.736 -> :rd 512, 517, 5
11:22:35.736 -> :rdi 512, 512
11:22:35.736 -> :c0 512, 517
11:22:35.769 -> :wr 517 0
11:22:35.769 -> :wrc 517 517 0
11:22:37.260 -> :wr 517 0
11:22:37.260 -> :wrc 517 517 0
11:22:37.260 -> :wr 229 0
11:22:37.260 -> :wrc 38 229 0
11:22:37.492 -> :wr 191 38
11:22:37.989 -> :wr 191 38
11:22:38.486 -> :wr 191 38
11:22:38.983 -> :wr 191 38
11:22:39.016 -> :ack 1072
11:22:39.016 -> :wr 191 38
11:22:39.016 -> :wrc 191 191 0
11:22:39.016 -> :ack 191
11:22:39.016 -> :rn 93
11:22:39.016 -> :rch 93, 380
11:22:39.016 -> :rd 5, 473, 0
11:22:39.016 -> :rdi 93, 5
11:22:39.016 -> :rd 37, 473, 5
11:22:39.016 -> :rdi 88, 37
11:22:39.082 -> :rd 5, 473, 42
11:22:39.082 -> :rdi 51, 5
11:22:39.082 -> :rd 1, 473, 47
11:22:39.082 -> :rdi 46, 1
11:22:39.082 -> :rd 5, 473, 48
11:22:39.082 -> :rdi 45, 5
11:22:39.082 -> :rd 40, 473, 53
11:22:39.082 -> :rdi 40, 40
11:22:39.082 -> :c 40, 93, 473
11:22:39.115 -> :wr 6 0
11:22:39.115 -> :wrc 6 6 0
11:22:39.115 -> :wr 45 0
11:22:39.115 -> :wrc 45 45 0
11:22:39.115 -> WS:av
11:22:39.115 -> :ref 2
11:22:39.115 -> :ur 2
11:22:39.115 -> :ref 2
11:22:39.115 -> :ur 2
11:22:39.115 -> :rd 5, 380, 0
11:22:39.115 -> :rdi 380, 5
11:22:39.115 -> :rd 375, 380, 5
11:22:39.115 -> :rdi 375, 375
11:22:39.115 -> :c0 375, 380
11:22:39.115 -> New secure client
11:22:39.115 -> :ref 2
11:22:39.115 -> :ur 2
11:22:39.115 -> request: GET /getTemp HTTP/1.1
11:22:39.115 -> method: GET url: /getTemp search: 
11:22:39.115 -> headerName: Host
11:22:39.115 -> headerValue: 192.168.1.138
11:22:39.115 -> headerName: User-Agent
11:22:39.149 -> headerValue: Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:65.0) Gecko/20100101 Firefox/65.0
11:22:39.149 -> headerName: Accept
11:22:39.149 -> headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8
11:22:39.149 -> headerName: Accept-Language
[... 147 lines cut ...]
11:22:54.160 -> Loop...63493
11:22:55.154 -> Loop...64493
11:22:56.148 -> Loop...65494
11:22:57.142 -> Loop...66494
11:22:58.136 -> Loop...67495
11:22:59.163 -> Loop...68495
11:23:00.157 -> Loop...69495
11:23:01.151 -> Loop...70496
11:23:02.146 -> Loop...71496
11:23:03.140 -> Loop...72497
11:23:04.167 -> Loop...73497
11:23:05.161 -> Loop...74497
11:23:06.155 -> Loop...75498
11:23:07.149 -> Loop...76498
11:23:08.143 -> Loop...77499
11:23:09.171 -> Loop...78500
11:23:10.165 -> Loop...79500
11:23:11.159 -> Loop...80501
11:23:12.153 -> Loop...81501
11:23:13.147 -> Loop...82501
11:23:14.174 -> Loop...83502
11:23:15.168 -> Loop...84502
11:23:16.162 -> Loop...85503
11:23:17.155 -> Loop...86503
````

earlephilhower, 2019-04-11:
My guess is this is the pre3.0 SDK issue.  Move to GIT head, set 160MHz, and re-try please.

earlephilhower, 2019-12-09:
This was a can't repo and could very well have been related to the pre3 SDK issues seen at the same time.  No feedback in months, so closing.

closed 2019-12-09 by earlephilhower