esp8266/Arduino #5943: I2C broken 2.5.0?  I2C scanner fails...

plocher, 2019-04-02:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [~] I have tested that the issue is present in current public release 2.5.0.
- [X] I have searched the issue tracker for a similar issue.
- [N/A ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				[Wemos D1 R2]
- Core Version:			[2.5.0 via Board Manager]
- Development Env:		[Arduino IDE 1.8.9]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Wemos D1 R2]
- Flash Mode:			[qio|dio|other] UNKNOWN
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth] UNKNOWN
- Reset Method:		[ck|nodemcu] UNKNOWN
- Flash Frequency:	[40Mhz] UNKNOWN
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600]

### Problem Description

I2C changes between 2.3.0 and 2.5.0 cause existing, working I2C code to fail.  Several different I2C tools fail, so I for reproduction and debugging I picked the simple Arduino I2C scanner example from [Arduino Playground I2C Scanner](https://playground.arduino.cc/Main/I2cScanner/) 

On my WemosD1R2, I use 
  Wire.begin(D2, D3); 
otherwise, the sketch is stock as shown on the playground site.

(The I2C slave is a PCA 9685 LED/Servo controller if it matters...)

On 2.3.0:

```
Scanning...
I2C device found at address 0x40  !
I2C device found at address 0x70  !
done
```

on 2.5.0
```
Scanning...
No I2C devices found
```

Simplistic scope work shows no I2C activity on D2 & D3 in the 2.5.0 case.

jandrassy, 2019-04-02:
try version 2.4.2 please

plocher, 2019-04-03:
Now that I have time to test in depth:
Works on 2.3.0, 2.4.0, 2.4.1 and 2.4.2
2.5.0-beta1 fails compiling with a permission error
2.5.0-beta2 works
2.5.0-beta3 works
and, finally, looping back the long way to 2.5.0, ...  it works. 

All along the way, no changes were made to the board type (tho the text changed from Wemos to Lolin(Wemos)...

I tried to force a fail by intentionally choosing a wrong board type (Wemos D1 R*1*), which compiled, and failed to run (as expected); subsequently choosing the correct D1 R*2* (and rebuilding/re-flashing) at first seemed to "fail", but worked as expected after a forced hardware (press reset button) reset, something that emphatically did not work this morning...

I can't explain the fail, but the "reinstall and try again after clearing your browser cache" hammer seems to have fixed whatever issues I may have had.

I'm closing this issue as not reproducible, sorry for any angst and heartache :-)

closed 2019-04-03 by plocher