esp8266/Arduino #5947: ESP HTTP Client file download in sdcard

HTerminal, 2019-04-04:
### Basic Infos

- [- ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [- ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ -] I have tested that the issue is present in the current master branch (aka latest git).
- [- ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[other]
- Flash Mode:			[other]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Hello,
**Example/ESP8266HTTPStream**
I am trying to make a file downloader using ESP12E and Sdcard module.
I want to make ESP as an HTTP client and when given a file address (HTTP not HTTPs) It downloads the file from the server and put it into sd card.
So far by now, I have achieved this functionality by SPIFFS. But when I try the same with the SDcard it crashes.

**Points i have checked**
- Sdcard detects fine.
- Http connection successfully connected.

**Problem i am facing in**
I am unable to point the SDcard file to **http.writeToStream(&f);** function.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <SD.h>
#include <SPI.h>>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#define USE_SERIAL Serial
HTTPClient http;

void setup() {
USE_SERIAL.begin(115200);
  // USE_SERIAL.setDebugOutput(true);

  USE_SERIAL.println();
  USE_SERIAL.println();
  USE_SERIAL.println();

  for (uint8_t t = 4; t > 0; t--) {
    USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
    USE_SERIAL.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("TS-Wireless1", "Ts@#10@labs");
  Serial.print(F("Initializing SD card..."));
  if (!SD.begin(D2)) { // see if the card is present and can be initialised. Wemos SD-Card CS uses D8
    Serial.println(F("Card failed or not present, no SD Card data logging possible..."));
    
  }
  else
  {
    Serial.println(F("Card initialised... file access enabled..."));
    
    
  }
}

void loop() {
if ((WiFiMulti.run() == WL_CONNECTED)) {
    
    String url = "http://192.168.1.114/123.mp3";
    String file_name = "/123.mp3";
    SD.remove(file_name);  
    Serial.println(url);
    File f = SD.open(file_name, "w");
    Serial.println(f);
    if (f) {
      Serial.println("Downloading file");
      http.begin(url);
      int httpCode = http.GET();
      if (httpCode > 0) {
        if (httpCode == HTTP_CODE_OK) {
          http.writeToStream(&f); //This is the part gives error i don't know why.
          
        }
      } else {
        Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
      }
      Serial.println("Done");
      Serial.println(f.size());
      f.close();
    }
    http.end();
  }
delay(10000); //Delay of 10 seconds
}

```

### Debug Messages

Sorry but i was unable to install the stack decoder becuase it gave error 

ERROR: xtensa-lx106-elf-gdb.exe not found!

HTerminal, 2019-04-05:
Found What i needed. It was easy i readBytes from stream pointer then feed into the Sdcard file.

closed 2019-04-05 by HTerminal

Nathan-ma, 2019-07-16:
so, how did you fixed? pointing it to the memory address?

ram-prasanth-9394, 2020-09-18:
@HTerminal Hi, The source code of yours was very helpful to and could you plz tell how do you readBytes from stream pointer;
here i've tried but i couldn't
can you plz help me,

regards,
Ram.

HTerminal, 2020-09-23:
> so, how did you fixed? pointing it to the memory address?

@ram-prasanth-9394 

I changed this part
`http.writeToStream(&f);`

with this
First check the available file size.

`size_t size = stream->available();`
Then if the size is not zero then .

```

          if (size) {
            // read up to 128 byte
            int c = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));

            // write it to Serial
            // USE_SERIAL.write(buff, c);
            f.write(buff, c);
            if (len > 0) {
              len -= c;
            }
```

Then write the file with the buffer.

ram-prasanth-9394, 2020-09-24:
@HTerminal can you plz share working sample code. 
coz im trying this with ESP32 .

Thanks in Advance 
Regards, Ram.

ram-prasanth-9394, 2020-09-25:
can you plz share working sample code.
coz im trying this with ESP32 .

Thanks in Advance
Regards, Ram.

HareshPrajapati, 2020-09-25:
HTTPClient Http;
    Serial.print("[HTTP] begin...\n");
    // configure server and url
    Http.begin(fileURL);
    // http.begin("media.gettyimages.com", 80, "/photos/red-heart-shape-balloon-flying-above-buildings-in-city-picture-id608954473");
    Serial.print("[HTTP] GET...\n");
    // start connection and send HTTP header
    int httpCode = Http.GET();
    if (httpCode > 0)
    {
        // HTTP header has been send and Server response header has been handled
        Serial.printf("[HTTP] GET... code: %d\n", httpCode);
        // file found at server
        if (httpCode == HTTP_CODE_OK)
        {
            uploadFile = SD.open(fileName, FILE_WRITE);
            // get lenght of document (is -1 when Server sends no Content-Length header)
            int len = Http.getSize();
            Serial.printf("Payload size [%d] bytes.\r\n", len);
            // create buffer for read
            uint8_t buff[2048] = {0};
            // get tcp stream
            WiFiClient *stream = Http.getStreamPtr();
            // read all data from server
            while (Http.connected() && (len > 0 || len == -1))
            {
                // get available data size
                size_t size = stream->available();
                if (size)
                { // read up to 128 byte
                    int c = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));
                    // write it to Serial
                    Serial.printf("%d bytes read[c].\r\n", c);
                    Serial.printf("%d bytes available for read \r\n", size);
                    // open file in append mode.
                    uploadFile.write(buff, c);
                    // close file.
                    if (len > 0)
                    {
                        len -= c;
                    }
                }
                delayMicroseconds(1);
            }
            Serial.println();
            Serial.print("[HTTP] connection closed or file end.\n");
            result = true;
        }
    }
    else
    {
        Serial.printf("[HTTP] GET... failed, error: %s\n", Http.errorToString(httpCode).c_str());
    }
    Http.end();
    uploadFile.close();

this will work for you. enjoy

HareshPrajapati, 2020-09-25:
` HTTPClient Http;
    Serial.print("[HTTP] begin...\n");
    // configure server and url
    Http.begin(fileURL);
    // http.begin("media.gettyimages.com", 80, "/photos/red-heart-shape-balloon-flying-above-buildings-in-city-picture-id608954473");
    Serial.print("[HTTP] GET...\n");
    // start connection and send HTTP header
    int httpCode = Http.GET();
    if (httpCode > 0)
    {
        // HTTP header has been send and Server response header has been handled
        Serial.printf("[HTTP] GET... code: %d\n", httpCode);
        // file found at server
        if (httpCode == HTTP_CODE_OK)
        {
            uploadFile = SD.open(fileName, FILE_WRITE);
            // get lenght of document (is -1 when Server sends no Content-Length header)
            int len = Http.getSize();
            Serial.printf("Payload size [%d] bytes.\r\n", len);
            // create buffer for read
            uint8_t buff[2048] = {0};
            // get tcp stream
            WiFiClient *stream = Http.getStreamPtr();
            // read all data from server
            while (Http.connected() && (len > 0 || len == -1))
            {
                // get available data size
                size_t size = stream->available();
                if (size)
                { // read up to 128 byte
                    int c = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));
                    // write it to Serial
                    Serial.printf("%d bytes read[c].\r\n", c);
                    Serial.printf("%d bytes available for read \r\n", size);
                    // open file in append mode.
                    uploadFile.write(buff, c);
                    // close file.
                    if (len > 0)
                    {
                        len -= c;
                    }
                }
                delayMicroseconds(1);
            }
            Serial.println();
            Serial.print("[HTTP] connection closed or file end.\n");
            result = true;
        }
    }
    else
    {
        Serial.printf("[HTTP] GET... failed, error: %s\n", Http.errorToString(httpCode).c_str());
    }
    Http.end();
    uploadFile.close();`

this will work for downloading file , enjoy.

HareshPrajapati, 2020-09-25:
HTTPClient Http;
    Serial.print("[HTTP] begin...\n");
    // configure server and url
    Http.begin(fileURL);
    Serial.print("[HTTP] GET...\n");
    // start connection and send HTTP header
    int httpCode = Http.GET();
    if (httpCode > 0)
    {
        // HTTP header has been send and Server response header has been handled
        Serial.printf("[HTTP] GET... code: %d\n", httpCode);
        // file found at server
        if (httpCode == HTTP_CODE_OK)
        {
            uploadFile = SD.open(fileName, FILE_WRITE);
            // get lenght of document (is -1 when Server sends no Content-Length header)
            int len = Http.getSize();
            Serial.printf("Payload size [%d] bytes.\r\n", len);
            // create buffer for read
            uint8_t buff[2048] = {0};
            // get tcp stream
            WiFiClient *stream = Http.getStreamPtr();
            // read all data from server
            while (Http.connected() && (len > 0 || len == -1))
            {
                // get available data size
                size_t size = stream->available();
                if (size)
                { // read up to 128 byte
                    int c = stream->readBytes(buff, ((size > sizeof(buff)) ? sizeof(buff) : size));
                    // write it to Serial
                    Serial.printf("%d bytes read[c].\r\n", c);
                    Serial.printf("%d bytes available for read \r\n", size);
                    // open file in append mode.
                    uploadFile.write(buff, c);
                    // close file.
                    if (len > 0)
                    {
                        len -= c;
                    }
                }
                delayMicroseconds(1);
            }
            Serial.println();
            Serial.print("[HTTP] connection closed or file end.\n");
            result = true;
        }
    }
    else
    {
        Serial.printf("[HTTP] GET... failed, error: %s\n", Http.errorToString(httpCode).c_str());
    }
    Http.end();
    uploadFile.close();

try this. it will work