esp8266/Arduino #5951: Uploading Error to ESP8266

ffaacceelee, 2019-04-05:
### Basic Infos

- [y] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [y ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [y ] I have tested that the issue is present in current master branch (aka latest git).
- [ y] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

When I uploading programs, an error occured,
python /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/upload.py --chip esp8266 --port /dev/cu.wchusbserial1a1230 --baud 115200 --trace version --end --chip esp8266 --port /dev/cu.wchusbserial1a1230 --baud 115200 --trace write_flash 0x0 /var/folders/hf/1pgb0bfj1bjblvw7035mlllc0000gn/T/arduino_build_14437/sketch_apr05a.ino.bin --end 
Error in command line, need pyserial path as 1st arg and esptool path as 2nd.

And I found this error is from Upload.py, because import esptool failed. But when get.py is running , esptool and pyserial is not downloaded. Because there are no path in file 

package_esp8266com_index.template.json. So, please tell me how can I fix it , or where could I download the pyserial then put the file on pyserial fold.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

ffaacceelee, 2019-04-11:
thankx for  your  answer.  yes, I download the latest git. Put the git into fold esp8266com>esp8266, and then run the get.py, it download the tool automatically. When I compiled code, it is worked, but  when I upload the program, the error occured. The same thing is happened in MAC OS, But it is worked in Windows. I will uninstall it and try install once again.

earlephilhower, 2019-04-11:
GET.py doesn't grab those utilities, it's part of the GIT tree.  You need to:
````
git submodule init
git submodule update
````

That will pull pyserial and esxptool.py into the tree so you can do uploads.

ffaacceelee, 2019-04-15:
Thank you very much, the problom is solved,

closed 2019-04-15 by ffaacceelee