esp8266/Arduino #5952: BSSID incompatibility with arduino API

aster94, 2019-04-05:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				all
- Core Version:			git 06/04/2019 

### Settings in IDE

not relevant

### Problem Description

The [BSSID](https://www.arduino.cc/en/Reference/WiFiBSSID) core api is incompatible with the arduino one

[Wifi nina code](https://github.com/arduino-libraries/WiFiNINA/blob/master/src/WiFi.cpp#L240) which I copy it here to see it easier:
```cpp
uint8_t* WiFiClass::BSSID(uint8_t* bssid)
{
	uint8_t* _bssid = WiFiDrv::getCurrentBSSID();
	memcpy(bssid, _bssid, WL_MAC_ADDR_LENGTH);
    return bssid;
}
```

esp8266 code:
```cpp
uint8_t* ESP8266WiFiSTAClass::BSSID(void) {
    static struct station_config conf;
    wifi_station_get_config(&conf);
    return reinterpret_cast<uint8_t*>(conf.bssid);
}
```

same for the wifi scan:
arduino [code](https://github.com/arduino-libraries/WiFiNINA/blob/master/src/WiFi.cpp#L289):
```cpp
uint8_t* WiFiClass::BSSID(uint8_t networkItem, uint8_t* bssid)
{
	return WiFiDrv::getBSSIDNetowrks(networkItem, bssid);
}
```

esp8266 code:
```cpp
uint8_t * ESP8266WiFiScanClass::BSSID(uint8_t i) {
    struct bss_info* it = reinterpret_cast<struct bss_info*>(_getScanInfoByIndex(i));
    if(!it) {
        return 0;
    }
    return it->bssid;
}
```

At the end the BSSID and macAddress functions are similar, both gave back a mac address, and for this reason them should be used in a similar way, note that your macAddress function is perfect:
```cpp
uint8_t* ESP8266WiFiSTAClass::macAddress(uint8_t* mac) {
    wifi_get_macaddr(STATION_IF, mac);
    return mac;
}
```

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>

char ssid[] = "Aster-gopro3";
char pass[] = "catania94";
int status = 0;

void setup()
{
  Serial.begin(115200);
}

void loop ()
{
  if (status != WL_CONNECTED) {
    WiFi.begin(ssid, pass);
    Serial.println(".");
  }
  else {
    // print the MAC address of the router you're attached to:
    byte mac[6];
    WiFi.macAddress(mac);
    printMacAddress(mac);

    byte bssid[6];
    WiFi.BSSID(bssid);
    printMacAddress(bssid);
    delay(2000);
  }
}

void printMacAddress(byte mac[]) {
  for (int i = 5; i >= 0; i--) {
    if (mac[i] < 16) {
      Serial.print("0");
    }
    Serial.print(mac[i], HEX);
    if (i > 0) {
      Serial.print(":");
    }
  }
  Serial.println();
}
```

devyte, 2019-04-09:
You seem to be confusing the MAC address of the ESP's station interface with the MAC address of the router the station interface is connected to, aka Arduino's BSSID.
From Arduino's page you linked above:
> Gets the MAC address of the routher you are connected to

Our ESP8266WiFi::BSSID(void) gets the BSSID of the ESP's station interface.
Our ESP8266WiFi::BSSID(int i) gets the BSSID of the ith networkItem of a scan result.
Our  ESP8266WiFi::macAddress get/set the BSD of the ESP's station interface.

None of them gets the BSSID of the router to which the station interface is connected. There is currently no way to get the router's BSSID. This is a limitation of the SDK. One thing you could do is scan, then iterate over the scan results to match the SSID you specified with the SSIDs found, then when there is a match get the BSSID. However, in the case where there are several routers with the same SSID and each with a different BSSID, you can't assure picking the correct scan item.

The one exception is when you specify which BSSID the station interface should connect to in the begin() method, and even then I think you can't really retrieve it the way you seem to want.

Closing.

aster94, 2019-04-09:
Hi @devyte,

I wasn't aware that the sdk can't get the bssid of the router's BSSID, so just forget about that part of the issue and focus on the first half:

The BSSID core api is incompatible with the arduino one

Wifi nina code:
```cpp
uint8_t* WiFiClass::BSSID(uint8_t* bssid)
{
	uint8_t* _bssid = WiFiDrv::getCurrentBSSID();
	memcpy(bssid, _bssid, WL_MAC_ADDR_LENGTH);
    return bssid;
}
```
esp8266 code:
```cpp
uint8_t* ESP8266WiFiSTAClass::BSSID(void) {
    static struct station_config conf;
    wifi_station_get_config(&conf);
    return reinterpret_cast<uint8_t*>(conf.bssid);
}
```

In the purpose of compatibility, would it be possible to change the function from `BSSID(void) ` to `BSSID(uint8_t* bssid);`?

Otherwise everyone who use a sketch from an Arduino example, as linked in the reference, would get an error, and not everyone is able to scan source code to find the problem 😅

Would you mind to open again the issue? I wasn't confusing the esp and the router Mac, the last one was an example

devyte, 2019-04-11:
> Is this explained somewhere?

readthedocs

> Just for curiosity: could the freertos sdk do that?

Probably not, the internal code is very similar. I can't say for sure, though, i don't use the freertos sdk. 

Notice that our wifi class is named ESP8266WiFi. This is on purpose, precisely because it can't be fully compatible.

aster94, 2019-04-11:
https://arduino-esp8266.readthedocs.io/en/2.5.0/esp8266wifi/station-class.html#bssid

Well, english is not my first language but this could be misleading, from your words i know that
> None of them gets the BSSID of the router to which the station interface is connected
But in the docs is not very clear

And by the way imho it would be better just to make the function returning an empty pointer then returning something i wasn t looking for

devyte, 2019-04-11:
[This](https://github.com/esp8266/Arduino/blob/882d5ba4226f3120a23e58fba10d7c585e2b7481/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L633) is the function that returns the mac of the station interface. Notice that it is a thin layer over the SDK functionality.

About the docs, you're right, it's not clear. Reopening to track a doc update.

devyte, 2019-04-11:
Actually, After reading Espressif's NONOS API ref document 2c, I'm wrong: the returned mac is not the one of the station interface, it's the bssid the user specified the station should connect to. 
Thing is, it does not mean the same thing as the mac of the router the station is in fact connected to. Its value is just a copy of the value configured by the user, and its content is valid only when bssid_set is 1.

aster94, 2019-04-11:
> Thing is, it does not mean the same thing as the mac of the router the station is in fact connected to. Its value is just a copy of the value configured by the user, and its content is valid only when bssid_set is 1.

you mean the 4th parameter here, right? https://github.com/esp8266/Arduino/blob/882d5ba4226f3120a23e58fba10d7c585e2b7481/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L126

closed 2019-04-12 by devyte

d-a-v, 2019-07-28:
WiFi.BSSID() is indeed confusing.

> since the BSSID() should return
`A byte array containing the MAC address of the router the WiFi shield is currently connected to`,
make this function return a empty array if bssid_set is 0 and the correct MAC if it is set to 1

@aster94 Have you tried something on your side that would be worth being integrated to the core ?

aster94, 2019-08-02:
hello @d-a-v actually not, but would something like this work?

```cpp
uint8_t* ESP8266WiFiSTAClass::BSSID(void) {
    static struct station_config conf;
    wifi_station_get_config(&conf);
if (conf.bssid_set == 1)
{
return 0;
}
else
{
    return reinterpret_cast<uint8_t*>(conf.bssid);
}
}
```

sorry for bad formattation but i wrote it here

PR #5977 (2019-04-12): Update station-class.rst
Clarify ESP8266WiFiSTAClass::BSSID()
Fixes #5952