esp8266/Arduino #5955: OTA update fails every other attempt after reset (> r2.4.2)

aboulfad, 2019-04-06:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
-  [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12]
- Core Version:			[2.5.0-b3, 2.5.0, 2.4.2]
- Development Env:		[Arduino IDE, cmd line]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA]
- Upload Speed:   OTA

### Problem Description
Hello, Arduino OTA fails every other attempt after rest, after much searching and debugging I believe this is a side effect of issue #3481. Arduino OTA uses UDP to initiate the transfer and espota.py expects a reply to the UDP Flash command packet within a timeout. If the esp8266 does not reply with an UDP "OK" payload the script will timeout with the famous "No Answer".

Upon doing various tests, wireshark proves the above behaviour. The first UDP packet issue does not seem to exist in r2.4.2, where OTA update works all the time. The OTA UDP "OK" reply is always sent after an esp8266 reset.

I am not familiar with esp8266 src code, but i've looked at the diffs for 'ArduinoOTA.cpp' of 2.4.2 and 2.5.0, and although i know NULL about C++, it does not seem the issue is introduced from all those updates.

I also quickly diffed lwip v2 of [2.4.2](https://github.com/d-a-v/esp82xx-nonos-linklayer/tree/25d5e8186f710a230221021cba97727dbfdfd953) vs [2.5.0](https://github.com/d-a-v/esp82xx-nonos-linklayer/tree/163bb82dd97e985df76c07d1c3f7325212cbdd5e), there is a bunch of patches that also likely did not affect the UDP first packet issue.

I am willing to continue to dig further, if someone could be kind to point me in some direction! A temporary workaround could be to implement a retry on the sendto socket call https://github.com/esp8266/Arduino/blob/77451d6ec9c4c8ce014704214aa98252a37f3d76/tools/espota.py#L95 in `espota.py` script.

### Steps to reproduce
- Use either 2.5.0 or later installed from IDE board manager.
- Load BasicOTA sketch via serial, then reset
- Attempt OTA update , it will fail on first try, observe that there is NO UDP OK packet sent from esp8266
- Attempt another OTA update, it will work as the UDP OK packet is sent.

[image]
[image]

d-a-v, 2019-04-06:
Me, I would [replace this](https://github.com/esp8266/Arduino/blob/77451d6ec9c4c8ce014704214aa98252a37f3d76/libraries/ArduinoOTA/ArduinoOTA.cpp#L276L278) by
```
  for (int i = 0; i < 10; i++)
  {
    _udp_ota->append("OK", 2);
    _udp_ota->send(ota_ip, _ota_udp_port);
    delay(10);
  }
```
and replace python's `if (data != "OK"):` by `if ("OK" not in data):`.

Better, I would replace this UDP message by a TCP one.
Because UDP has nothing to do here. A TCP message, would be always sent, received and acknowledged reliably.

aboulfad, 2019-04-06:
@d-a-v , merci 😄 your workaround should work as we wait for a conclusion on issue #3481 . I thought a temporary workaround to retry on revc fail  in espota.py only would work but noticed that the ArduinoOTA code initiates the TCP transfer irregardless if the UDP OK is sent (and it can’t know as it’s UDP), whereafter espota times out on UDP recv.

I agree whole heartedly that the OTA update init sequence (Flash command/OK) is best served using TCP, but I can’t comment why UDP was chosen.

aboulfad, 2019-04-07:
Salut @d-a-v , is the enhancement tag for implementing a TCP solution for OTA session initiation instead of UDP? If yes, and if this is accepted, I would like to « attempt » implementing it. I know it would take you guys literally hours to do. I have much spare time and would be nice to make this tiny contribution. I think the scope is mostly limited to `ArduinoOTA.cpp ` and `espota.py`?

What do you think?

d-a-v, 2019-04-07:
Yes go ahead!

(Keep backward compatibility in mind though)

aboulfad, 2019-04-07:
That was quick 😀

By backward compatibility, do you mean a solution that initiates the OTA handshake based on a common transport between client and esp?

A scenario I can think of, client has an old version of esports.py which does not support TCP - so switch to UDP, How valid is this?

Afraid I don’t much insight or history about all this so I may be way off... my first attempt will be a simple TCP transport implemented in both .py and .cpp.

PS: heavy focus on attempt as very new to rap code base and C++/python

aboulfad, 2019-04-07:
@igrr , do you have any insight why UDP was chosen as the transport for the OTA Flash command? As you are one of the authors for OTA, do you have any additional thoughts regarding our proposal to use TCP for that initial OTA setup?

As this is my first public contrib, I just want to consult and include all relevant folks. Thank you.

igrr, 2019-04-07:
UDP was chosen due to resources used, mostly. Keeping a TCP PCB (~socket) listening needed more memory than an open UDP PCB. Also the total number of TCP sockets was limited to 4 or so, in earlier versions of the LwIP (which came pre-compiled with the SDK).

Although one hand I agree with the "eh, UDP is not reliable, let's slap on TCP here" approach, it might be worth actually looking into the UDP loss issue, especially if it happens predictably ("on every other reset").

aboulfad, 2019-04-07:
Thank you for the very valuable insight and quick reply. 
One point to add, and I know very little, after the U_FLASH command exchange which uses UDP, a TCP session is used for the firmware transfer. 
Couldn’t this same TCP socket be also used for the initial commands too? This way there is no additional resource usage. 

The UDP first packet loss is reproduced in every other OTA attempt, predictably, which I believe is related to issue #3481. I can also attempt to look into this but some guidance would be helpful.

d-a-v, 2019-04-07:
> it might be worth actually looking into the UDP loss issue, especially if it happens predictably ("on every other reset").

I agree with that. I got back to #3481 and found that without some delays, UDP packets are not event sent out.

> Couldn’t this same TCP socket be also used for the initial commands too? This way there is no additional resource usage.

I agree with that too.

aboulfad, 2019-04-08:
@d-a-v , Small update, lwip v1.4 (high BW)  does NOT exhibit the issue compared to lwip v2 (high BW) in **same release 2.5.0**. OTA transfer (and hence UDP exchange) works everytime. 

The lwip v1.4 -> v2 evolution is convoluted to my noob eyes, I don't think I can narrow it further as the lwip codebase is large and don't have much history.

aboulfad, 2019-04-08:
@d-a-v , need some help svp, trying to use lwip debugging by setting  `ULWIPDEBUG`, using your instructions from https://github.com/esp8266/Arduino/issues/3884#issuecomment-347537526, running the sketch outputs nothing related to those defines. (using git checkout 2.5.0)

Other defines that I set, e.g. `DEBUG_ESP_MDNS_RESPONDER` works fine and show up on serial port.

And I believe if I set both defines `UDEBUG &  ULWIPDEBUG`, compilations breaks ...
<details>
  <summary>lwip2 make install</summary>
  
 ```make -f makefiles/Makefile.glue target=arduino BUILD=build-536-feat-v6 TCP_MSS=536 LWIP_FEATURES=1 LWIP_IPV6=1 V=0
C  build-536-feat-v6/glue/doprint.o
C  build-536-feat-v6/glue/uprint.o
C  build-536-feat-v6/glue-lwip/lwip-git.o
glue-lwip/lwip-git.c: In function 'new_display_netif':
glue-lwip/lwip-git.c:144:37: error: 'ip_addr_t' has no member named 'addr'
  display_ip32(" ip=", netif->ip_addr.addr);
                                     ^
glue-lwip/lwip-git.c:145:39: error: 'ip_addr_t' has no member named 'addr'
  display_ip32(" mask=", netif->netmask.addr);
                                       ^
glue-lwip/lwip-git.c:146:32: error: 'ip_addr_t' has no member named 'addr'
  display_ip32(" gw=", netif->gw.addr);
                                ^
glue-lwip/lwip-git.c: At top level:
glue-lwip/lwip-git.c:467:9: note: #pragma message: 

-------- TCP_MSS = 536 --------
-------- LWIP_FEATURES = 1 --------
-------- LWIP_IPV6 = 1 --------

 #pragma message "\n\n" VAR_NAME_VALUE(TCP_MSS) VAR_NAME_VALUE(LWIP_FEATURES) VAR_NAME_VALUE(LWIP_IPV6)
         ^
make[3]: *** [build-536-feat-v6/glue-lwip/lwip-git.o] Error 1
make[2]: *** [liblwip6-536-feat.a] Error 2
make -f makefiles/Makefile.glue-esp BUILD=build-1460-feat-v6 V=0
C  build-1460-feat-v6/glue-esp/lwip-esp.o
AR liblwip6-1460-feat.a
make -f makefiles/Makefile.glue target=arduino BUILD=build-1460-feat-v6 TCP_MSS=1460 LWIP_FEATURES=1 LWIP_IPV6=1 V=0
C  build-1460-feat-v6/glue/doprint.o
C  build-1460-feat-v6/glue/uprint.o
C  build-1460-feat-v6/glue-lwip/lwip-git.o
glue-lwip/lwip-git.c: In function 'new_display_netif':
glue-lwip/lwip-git.c:144:37: error: 'ip_addr_t' has no member named 'addr'
  display_ip32(" ip=", netif->ip_addr.addr);
                                     ^
glue-lwip/lwip-git.c:145:39: error: 'ip_addr_t' has no member named 'addr'
  display_ip32(" mask=", netif->netmask.addr);
                                       ^
glue-lwip/lwip-git.c:146:32: error: 'ip_addr_t' has no member named 'addr'
  display_ip32(" gw=", netif->gw.addr);
                                ^
glue-lwip/lwip-git.c: At top level:
glue-lwip/lwip-git.c:467:9: note: #pragma message: 

-------- TCP_MSS = 1460 --------
-------- LWIP_FEATURES = 1 --------
-------- LWIP_IPV6 = 1 --------

 #pragma message "\n\n" VAR_NAME_VALUE(TCP_MSS) VAR_NAME_VALUE(LWIP_FEATURES) VAR_NAME_VALUE(LWIP_IPV6)
         ^
make[3]: *** [build-1460-feat-v6/glue-lwip/lwip-git.o] Error 1
make[2]: *** [liblwip6-1460-feat.a] Error 2
make[1]: *** [install] Error 2
make: *** [install] Error 2```
</details>

d-a-v, 2019-04-09:
This bug is already fix in master (and I just fixed the previous fix), in `tools/sdk/lwip2/`:
```
cd builder
git fetch
git checkout master
```

aboulfad, 2019-04-09:
Sorry, unsure to which bug you say is fixed, no lwip debugging output or setting both defines that breaks compilation? Or the UDP packet loss?

In any case I’ll checkout builder master and go from there. Thank you.

d-a-v, 2019-04-09:
You might also have a look to https://github.com/esp8266/Arduino/issues/3481#issuecomment-480643361

aboulfad, 2019-04-09:
Thanks I saw it when you commented there. For this OTA usecase, it’s just one single UDP packet not being sent (the first one post boot). I’ve done about 100’s of uploads, and I can’t put my finger on it, but probably there’s some interaction with the first ARP. I’ve noticed in some case esp does not respond to ARP request. And one person in that issue also mentioned ARP...

It would really help me if I can activate lwip2 debugging, does your latest fix has anything to do with `ULWIPDEBUG` ?

d-a-v, 2019-04-09:
> does your latest fix has anything to do with ULWIPDEBUG ?

No, it's supposed to work. You need to enable debug on serial in IDE Tools menu.

aboulfad, 2019-04-09:
> > does your latest fix has anything to do with ULWIPDEBUG ?
> 
> No, it's supposed to work. You need to enable debug on serial in IDE Tools menu.

Yes, I did that, and also added (in case) ‘Serial.setdebugouput(true)’, nothing is displayed aside from whatever I set in debug level (OTA, wifi, core,...). Sadly I spent like 4 hours trying to figure it out before I asked for help!

d-a-v, 2019-04-09:
Ah!
That was working with lwIP-2.0.x but when I updated to lwIP-2.1.2 I reworked the whole lwipopts.h
and I didn't change the debug default options.
You need to change all (or some) `LWIP_DBG_OFF` to `LWIP_DBG_ON` in:
https://github.com/d-a-v/esp82xx-nonos-linklayer/blob/25d5e8186f710a230221021cba97727dbfdfd953/glue-lwip/arduino/lwipopts.h#L3259

aboulfad, 2019-04-09:
ok, just an update on the logging, finally I can see :-) The debug logging for lwip is very sophisticated, by just setting `ETHARP_DEBUG` , i got tons of traces in the IDE console and more than ARP in the serial console when I ran your script from https://github.com/esp8266/Arduino/issues/3481#issuecomment-480643361, netdump is super useful too !

I am lost between all these high level defines: `LWIP_DBG_LEVEL_ALL`, `LWIP_DBG_TYPES_ON` and the individual defines. I'll check '`debug.h`' later. If I get an exception 28 or any during debugging, i'll implement that line change you asked. So from now on, I cloned master.

aboulfad, 2019-04-09:
ok I implemented your one liner above, recompiled, uploaded... run a bit, then I reset the board, I got 3 panics in a row... in between resets. Now, is this related to the debug defines ???
Panic 1,2,3: 
Panic `$ESP8266/libraries/ESP8266WiFi/src/include/ClientContext.h:604 err_t ClientContext::_connected(tcp_pcb*, err: Assertion '_connect_pending' failed.`

aboulfad, 2019-04-11:
Ok solution and problem details in my comment https://github.com/esp8266/Arduino/issues/3481#issuecomment-482293996. In the end, we will not change the OTA transport to TCP given ARP_QUEUEING=1 solves the OTA UDP packet loss.

PR #5978 (2019-04-12): reenable ARP queuing in lwip2 by courtesy of @aboulfad
refs:
- https://github.com/d-a-v/esp82xx-nonos-linklayer/pull/33
- https://github.com/esp8266/Arduino/issues/3481#issuecomment-482293996

fix #5955 
fix #3481 (first packet lost in a row of 4 - need to increase lwIP's `ARP_QUEUE_LEN` which defaults to 3) 

Thanks @aboulfad !

(it was removed by error on 9 oct 2018 https://github.com/esp8266/Arduino/commit/a1e59e9c014c907c008e96c9a3f1b7fc731adedb#diff-8a46852651388faec204ec21c86b8098L586)
closed 2019-04-11 by aboulfad