esp8266/Arduino #5958: Checkbox limitation WiFiManager

Craig1516, 2019-04-07:
### Basic Infos

- [SortOf] This issue complies with the [issue POLICY doc]  Code Snipit provided
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12F
- Core Version:			[2.5.0]
- Development Env:		[Arduino IDE 1.8.9]
- Operating System:		[Windows 7 pro]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|
- Flash Mode:			[Dout]
- Flash Size: 		        [4MB/1MB]
- lwip Variant:                  [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:		[80Mhz|]
- Upload Using:		[SERIAL]
- Upload Speed:              [115200] 

### Problem Description

In AP mode ESP hosts an LAN input page for user input.  Some of the checkboxes are displayed properly, but not all.  Using WiFiManager (Taylor) I'm trying to build checkboxes for user WIFI input.  The program tries to render 4 sets of 3 checkboxes and some other text input.  The text input fields work perfectly.  The first 11 checkboxes work, but the 12th does not (image below).  To remedy this I have inserted 3 additional fake checkboxes at the bottom and have had partial success (except for #12).  I did this fake stuffing as I found that when the program is modified to only show 9 checkboxes then the last 2 or 3 are similarly corrupted.

### Sketch Segments

#include <FS.h>
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#include <ESP8266WebServer.h>
#include <DNSServer.h>
#include <WiFiManager.h>          //https://github.com/kentaylor/WiFiManager
...
void setup() {
...
}

void loop() {
...    WiFiManagerPara

meter p_thingspeakApiKey("thingspeakApiKey", "Thingspeak API Key", thingspeakApiKey, 17);
    WiFiManagerParameter p_IFTTTApiKey("IFTTTApiKey", "IFTTT API Key", IFTTTApiKey, 44);
    WiFiManagerParameter p_Upper_Temp("Upper_Temp", "Upper_Temp(deg_F)", Upper_Temp, 4);
    WiFiManagerParameter p_Voltage_Limit("Voltage_Limit", "Voltage_Limit (milliV)", Voltage_Limit, 5);
    WiFiManagerParameter p_Frequency("Frequency", "Frequency (minutes)", Frequency, 4);

    WiFiManagerParameter p_lineBreak_text("<p>Choose one of the following three:</p>");
    WiFiManagerParameter p_lineBreak_notext("<p></p>");
    WiFiManagerParameter p_FzAlarmEmail("FzAlarmEmail", "FzAlarmEmail", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);
    WiFiManagerParameter p_FzAlarmPhone("FzAlarmPhone", "FzAlarmPhone", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);
    WiFiManagerParameter p_FzAlarmText("FzAlarmText", "FzAlarmText", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);
    WiFiManagerParameter p_BatAlarmEmail("BatAlarmEmail", "BatAlarmEmail", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);
    WiFiManagerParameter p_BatAlarmPhone("BatAlarmPhone", "BatAlarmPhone", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);
    WiFiManagerParameter p_BatAlarmText("BatAlarmText", "BatAlarmText", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);
    WiFiManagerParameter p_VoltAlarmEmail("VoltAlarmEmail", "VoltAlarmEmail", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);
    WiFiManagerParameter p_VoltAlarmPhone("VoltAlarmPhone", "VoltAlarmPhone", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);
    WiFiManagerParameter p_VoltAlarmText("VoltAlarmText", "VoltAlarmText", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);
    WiFiManagerParameter p_AliveEmail("AliveEmail", "AliveEmail", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);
    WiFiManagerParameter p_AlivePhone("AlivePhone", "AlivePhone", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);
    WiFiManagerParameter p_AliveText("AliveText", "AliveText", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);
    WiFiManagerParameter p_FakeAlarmEmail("FakeText", "FakeText", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);
    WiFiManagerParameter p_FakeAlarmPhone("FakeText", "FakeText", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);
    WiFiManagerParameter p_FakeAlarmText("FakeText", "FakeText", "T", 2, "type=\"checkbox\" ", WFM_LABEL_AFTER);

...
    // Initialize WiFIManager
    WiFiManager wifiManager;    
...
Manager.addParameter(&p_BatAlarmEmail);
    wifiManager.addParameter(&p_BatAlarmPhone);
    wifiManager.addParameter(&p_BatAlarmText);
  wifiManager.addParameter(&p_lineBreak_text);             //linebreak
    wifiManager.addParameter(&p_VoltAlarmEmail);
    wifiManager.addParameter(&p_VoltAlarmPhone);
    wifiManager.addParameter(&p_VoltAlarmText);
  wifiManager.addParameter(&p_lineBreak_text);             //linebreak
    wifiManager.addParameter(&p_AliveEmail);
    wifiManager.addParameter(&p_AlivePhone);
    wifiManager.addParameter(&p_AliveText);
  wifiManager.addParameter(&p_lineBreak_text);             //linebreak
    wifiManager.addParameter(&p_FakeAlarmEmail);
    wifiManager.addParameter(&p_FakeAlarmPhone);
    wifiManager.addParameter(&p_FakeAlarmText);
  wifiManager.addParameter(&p_lineBreak_notext);             //linebreak

    wifiManager.addParameter (&p_thingspeakApiKey);
  wifiManager.addParameter(&p_lineBreak_notext);             //linebreak
    wifiManager.addParameter (&p_IFTTTApiKey);
  wifiManager.addParameter(&p_lineBreak_notext);             //linebreak
    wifiManager.addParameter (&p_Upper_Temp);
  wifiManager.addParameter(&p_lineBreak_notext);             //linebreak
    wifiManager.addParameter (&p_Voltage_Limit);
  wifiManager.addParameter(&p_lineBreak_notext);             //linebreak
    wifiManager.addParameter (&p_Frequency);
...
}

### NO Debug Messages - compiles fine.

### Partial Browser Image

[image]

Jeroen88, 2019-04-08:
In https://github.com/kentaylor/WiFiManager/blob/master/WiFiManager.h WIFI_MANAGER_MAX_PARAMS is set to 10.

Craig1516, 2019-04-08:
@Jeroen88  Thanks.  Do I have the prerogative to modify this value?  Will it cause harm elsewhere?   Funny that 11 of my checkboxes work, and teh 12th does not.  I'm presuming that the MAX_PARAMS is zero based so that 10 really means 11.   I'll play with that value tomorrow.  Many thanks for pointing this out!!!
Best, 
Craig

Jeroen88, 2019-04-08:
I do not use this library, I just looked it up in the source code. So I do not know of any impact elsewhere but I expect no other impact. 
You can edit your local copy of WiFiManager.h, but every time you update this library you have to edit it again. 
Please close the issue, because this is not an issue for the ESP8266 core.

Craig1516, 2019-04-08:
@jeroen88.  Closing.  I did edit MAX_PARAMS from 10 to 20 since I have 20 total parameters.  Thanks again for that.  The result sorta helped: instead of the corruption of the 1 valid input checkbox and 3 fake ones, I'm now getting all my valid checkboxes, but the 3 fake stuffing parameters are still corrupted.  Still don't know.

closed 2019-04-08 by Craig1516

Jeroen88, 2019-04-08:
You're welcome!

Just a few wild guesses: 
- const char *id, const char *placeholder (and const char *custom) are not copied by WiFiManagerParameter, only const char *defaultValue is copied into private WiFiManagerParameter::_value. If the WiFiManagerParameters that you declare at the beginning of loop() go out of scope (basicly outside loop()) they are invalid
- because you declare all WiFiManagerParameters in a function they are allocated on the stack. Might the stack size be insufficient?
What if you declare them outside loop()?