esp8266/Arduino #5961: trying to build lwip2 from  tools/sdk/lwip2

cloeffler, 2019-04-08:
I need to build lwip2 using the instructions located in tools/sdk/lwip2 using make install.

But the makefiles are looking for ../../../xtensa-lx106-elf/bin/extensa-lx106-elf-gcc and cant find them and failing to build. My question is why in the make files is it looking for things that dont exist? /tools/xtensa-lx106-elf is not in the repository. I can get all of those files that its looking for from esp-open-sdk and place them where the make files are looking but then get other errors. 

any help would be appreciated.

aboulfad, 2019-04-09:
It's very straightforward if you follow these [instructions](https://github.com/esp8266/Arduino/blob/master/doc/installing.rst#using-git-version). Its literally few commands. You shouldn't have any trouble if you follow it. I've compiled lwip2 dozen of time already using above steps.

devyte, 2019-04-09:
@cloeffler are you using latest git core? I'm not sure rebuilding lwip from source is meant to be used with a  vanilla core release installed via board manager. 
The previous comment seems to be aimed at precisely that: install latest git first, then try to rebuild lwip from source.
@d-a-v ?

cloeffler, 2019-04-09:
hey guys, I actually figured it out by editing some makefiles, and then just building lwip2 by itself.
Develo, yes i was using the latest git core.

[https://docs.google.com/uc?export=download&id=15xQQ2IBAggvOz0VO9wEkXjfxdAIyW-dL&revid=0B3knPoivDEFYUlQ0cDRVci8vUytFV2UyMWlKSjl0ekhURnQwPQ]

cloeffler, 2019-04-09:
Fadi, thanks for the info, i was not aware of that! I will try that also.

On Tue, Apr 9, 2019 at 12:11 PM Chris Loeffler <cloeffler@bartech.com<mailto:cloeffler@bartech.com>> wrote:
hey guys, I actually figured it out by editing some makefiles, and then just building lwip2 by itself.
Develo, yes i was using the latest git core.

[https://docs.google.com/uc?export=download&id=15xQQ2IBAggvOz0VO9wEkXjfxdAIyW-dL&revid=0B3knPoivDEFYUlQ0cDRVci8vUytFV2UyMWlKSjl0ekhURnQwPQ]

[https://docs.google.com/uc?export=download&id=15xQQ2IBAggvOz0VO9wEkXjfxdAIyW-dL&revid=0B3knPoivDEFYUlQ0cDRVci8vUytFV2UyMWlKSjl0ekhURnQwPQ]

earlephilhower, 2019-07-25:
Looks like things are working for the original user now, and I can confirm (via the esp-quick-toolchain) that the LWIP2 makefiles do complete fine, assuming the tools are installed in a GIT version.

However, if you try and build from inside a boards-manager installed repo, I bet the hardcoded paths (../../tools/blah/blah) will *not* exist since Arduino installs the exes in a separate spot from the actual core sources.

@d-a-v, do you think it's worth updating the LWIP2 README.md to note this?

devyte, 2019-11-20:
Per previous comment, the original issue here is addressed.
IMHO, building lwip from source is a very advanced operation, and it's ok to assume that anyone attempting it will likely already be on git.
Closing.

closed 2019-11-20 by devyte