esp8266/Arduino #5973: Some WiFi passwords characters does not work

Nexus1212, 2019-04-10:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			[ESP-8266 F]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Wemos D1 AND Nodemcu]
- Flash Mode:			[normal]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I want to test the BasicOta. But it seems, my password does not work. 
I tested it on to differnt ESP's.

This password does not work also:
#define STASSID "Home"
#define STAPSK  "12345678+*\:(~"

I can not connect with my Wifi with the password above. If I use "12345678" as a password, the connection works.
Mybe the "+" could be the problem...

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

#ifndef STASSID
#define STASSID "Home"
#define STAPSK  "12345678+*\:(~"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
  // ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  // ArduinoOTA.setHostname("myesp8266");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH) {
      type = "sketch";
    } else { // U_SPIFFS
      type = "filesystem";
    }

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) {
      Serial.println("Auth Failed");
    } else if (error == OTA_BEGIN_ERROR) {
      Serial.println("Begin Failed");
    } else if (error == OTA_CONNECT_ERROR) {
      Serial.println("Connect Failed");
    } else if (error == OTA_RECEIVE_ERROR) {
      Serial.println("Receive Failed");
    } else if (error == OTA_END_ERROR) {
      Serial.println("End Failed");
    }
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
}

Nexus1212, 2019-04-10:
Serial Monitor:
21:46:40.643 -> Booting
21:46:59.247 -> Connection Failed! Rebooting...
21:47:04.341 -> 
21:47:04.341 ->  ets Jan  8 2013,rst cause:2, boot mode:(1,6)
21:47:04.341 -> 
21:47:11.410 -> 
21:47:11.410 ->  ets Jan  8 2013,rst cause:4, boot mode:(1,6)
21:47:11.410 -> 
21:47:11.410 -> wdt reset

devyte, 2019-04-11:
> Core Version:	[latest git hash or date]

Are you really using latest git? Please make sure. There was a fix for String added after 2.5.0.

Nexus1212, 2019-04-12:
Hello,

 

I can not compile the sketch with current git hub version:

 

 

I used the version 2.5.0 and some older!

Could you try it?

 

Mit freundlichen Grüßen

 

Christoph Brunner

Nexus1212, 2019-04-14:
Can anybody help me?

Nexus1212, 2019-04-18:
I solfed the problem!

closed 2019-04-18 by Nexus1212