esp8266/Arduino #5974: Core 2.6.0 SDK2.2.2 does not boot with changes of 2019-04-10

TD-er, 2019-04-10:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[f950d53d821ed1f65eebdbefd632956a99a29562]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory low flash]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL] (both)
- Upload Speed:   [115200] (serial upload only)

### Problem Description

Node built using latest merges does not boot anymore.
It keeps outputting garbage in the serial console.

Build flags:
"PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x",
PlatformIO platform: https://github.com/platformio/platform-espressif8266.git#feature/stage  (thus latest master branch of this repo)

d-a-v, 2019-04-10:
It could be that PR and this specific commit missing from pio build process (elf -> bin conversion)
https://github.com/esp8266/Arduino/pull/5922/commits/13d7a4d55efee85084eea09b72a2e5a3671141f6

TD-er, 2019-04-10:
That updated version of the `elf2bin.py` file is present in my framework directory
`C:\Users\gijs\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools`

Can it be anything else?
Or is this more PlatformIO specific @ivankravets ?

d-a-v, 2019-04-10:
Could you try with #5922 only reverted from current master ?
```
$ cd esp8266-arduino-root
$ git diff fa0db2e3ab6c8cb9640e003c3adab81b49131994..0f4c9f710318ab9199714b9adcdd9701d78d264e | patch -p1 -R
```

TD-er, 2019-04-11:
Took a while before I could test it, but this is indeed a patch to make it compile again.

d-a-v, 2019-04-11:
@ivankravets Can you have a look to PIO's esp8266 elf->bin process and this specific change https://github.com/esp8266/Arduino/commit/13d7a4d55efee85084eea09b72a2e5a3671141f6 ?

earlephilhower, 2019-04-11:
@TD-er, I just opened an issue in PIO.  I think it's a minor change in a single file to make it work, would you be able to try my suggested fix and comment on it in the PIO issue above?  Simply adding the .text1 segment after .text, assuming the PIO bootloader has some intelligence, may fix everything.

Jason2866, 2019-04-11:
@earlephilhower  The suggested change in PIO from you does work. 
Use it in my fork to build Tasmota which wasnt booting as ESPeasy

TD-er, 2019-04-14:
@earlephilhower I did make a test build with that change you suggested.
The node then boots and seems to work just fine.

ivankravets, 2019-04-14:
Thanks for your PRs. The changes were applied to `feature/stage` branch of espresssif 8266 dev/platform. Please re-start a CI build.

TD-er, 2019-04-15:
I've been running tests today and the latest builds work just fine, so I consider it fixed.

closed 2019-04-15 by TD-er