esp8266/Arduino #5975: How to do a HTTPS post on ESP8266

s153341, 2019-04-11:
I am trying to do a HTTPS post on ESP8266. It works in Postman where it has the following:
POST https_//api.dgs.com/ccs-int/v1/aaplication/json
Headers:
Content-Type: application/json
Authorization: xxxxxxx
Environment: digitalexploration

But I keep getting: 'HTTP/1.1 401 Unauthorized'.

Can someone please help me with how to build up the headers correctly in the HTTPS POST? And maybe also give a link to some documentation? I cant seem to find any...

My code is:

  ```
      
    const char* host = "api.dgs.com";
    const int httpsPort = 443;
    String url = "/ccs-int/v1/applicationdatalogging/1.0";
    String msg = "";
    
    WiFiClientSecure client;
    
    Serial.print("connecting to : '");
    Serial.print(host);
    Serial.println("'");
    Serial.printf("Using fingerprint '%s'\n", fingerprint);
    client.setFingerprint(fingerprint);
    
    client.connect(host, httpsPort);
    
    Serial.print("requesting URL: '");
    Serial.print(url);
    Serial.println("'");
    
    client.print(String("POST ") + url + " HTTP/1.1\r\n" +
    	"Host: " + host + "\r\n" +
    	"Connection: close\r\n" +
    	"Content-Type: application/json" +
    	"Authorization: Bearer " + authorization_code + "\r\n" +
    	"Content-Length: " + msg.length() + "\r\n" +
    	"\r\n" +
    	msg + "\r\n");
    
    Serial.println("request sent");
    unsigned long timeout = millis();
    while (client.available() == 0) {
    	if (millis() - timeout > 5000) {
    		Serial.println(">>> Client Timeout !");
    		client.stop();
    		return;
    	}
    }
```

SarenT, 2019-04-17:
I use following code (simplified):
```
#include <ESP8266HTTPClient.h>
HTTPClient http;
...
wifiClientSecure.loadCACert(caCertFile);
...
wifiClientSecure.connect(host, 443);
...
http.begin(client, host + url);
http.addHeader("Content-Type", "application/x-www-form-urlencoded"); // e.g.
int httpCode = http.POST(msg); // e.g. a=4&b=6&c=something

```

jandrassy, 2019-04-17:
if you get an 401 error, you get a response from the server. SSL HTTP connection worked, only `authorization_code`  has wrong value

SarenT, 2019-04-18:
I have to say, I am very confused with examples I see. Here are a few confusing answers:

1. ESP8266 can use certificates, too weak for that can only do fingerprint check (I managed to do client and server authentication with both self signed and trusted authority signed certificate chains).
2. BearSSL and documentation doesn't seem to work. e.g. Examples often use BearSSLX509List but I see X509List instead.
3. How does it even work with good old loadPrivateKey, loadCertificate and loadCACert? Are they broken?
4. I was able to make MQTT two way authentication but it was with axTLS. Now, I can't get it work with HTTPS (server authentication works though).

For me the main problem right now is to understand, how to correctly use current libraries.

devyte, 2019-04-26:
This is a "how do I..." type question. Closing.

closed 2019-04-26 by devyte

jabernalv, 2019-11-17:
devyte, yes is a quesrtion "how do I...". This site is for THAT questions. 
Why did you close?

devyte, 2019-11-17:
@jabernalv This is an issue tracker, meant for tracking ussues in the core hosted in the repo. It is NOT for "how do I..." type questions or general help requests. Please see issue POLICY doc and ISSUE_TEMPLATE instructions for more details.

jabernalv, 2019-12-23:
@devyte Because of people like you this world is boring. You look like a lawyer, programmers are willing to help not follow absurd protocols.

devyte, 2019-12-23:
I am an Engineer, and I volunteer my free time to develop and maintain the core in this repository. In agreement with the repo owner when I was invited to be maintainer, I established a repo POLICY, and it has served well: the number of open issues has dropped from 1000+ when I arrived to 270 today, and is still on a downwards trend.
Maybe I make things a bit boring, but numbers prove the policy is being effective, and getting the repo under control is my primary interest. 
As someone who shows 0 contributions in the last year, please beware of criticism against those who actually contribute to open source. Here in this repo we choose to develop, not to hold your hand. You're right that other programmers are willing to help, but such basic help isn't meant for this tracker.