esp8266/Arduino #5979: http.GET() returns code of -11

bill-orange, 2019-04-12:
Using ESP8266HTTPClient  to collect data from OpenWeatherMap with an `http.GET(); ` I get a return code of -11.  It happens once very 15 or 20 calls.  I can not find any documentation of this code.  The normal return code on successful attempts is 200.

Does anybody know what a -11 code means?

bill-orange, 2019-04-12:
answered my own question.  Its a timeout.

closed 2019-04-12 by bill-orange

santigo171, 2022-09-05:
How did you solved it?

santigo171, 2022-09-05:
How did you solve it?

bill-orange, 2022-10-11:
II made multiple attempts.  If it failed repeatedly, I had the option to restarted the ESP8266.

Serial.println("  ");
   OWMError = getOWMWeatherCondition();
   Serial.print("OWM Error Return Code: "); Serial.println(OWMError);
   if (OWMError != 200) {
     WiFiFix();
     OWMError = getOWMWeatherCondition();
     Serial.print("OWM Error Return Code: "); Serial.println(OWMError);
     if (OWMError != 200) {
       OWMError = getOWMWeatherCondition();
       if ( testJSON() != 200 ) {
         Serial.println("                                            Bad OWM Test Return Code ........ Skipping");
         //esp_restart();

From: santigo171 
Sent: Monday, September 5, 2022 4:33 PM
To: esp8266/Arduino 
Cc: bill-orange ; State change 
Subject: Re: [esp8266/Arduino] http.GET() returns code of -11 (#5979)

How did you solve it?