esp8266/Arduino #5980: Enable to connect to Windows host in local network using Windows host name

milanbx, 2019-04-12:
### Basic Infos
- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [not sure] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [N/A] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				ESP8266 / ESP-12E
- Core Version:			not sure, flashed with NodeMCU ESP8266Flasher.exe (commit d929c51 on 8 Aug 2017) using INTERNAL://NODEMCU
- SDK:					SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:1.4.0rc2/BearSSL:6778687
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			tested with 2 breakouts of NodeMCU v1.0 + Wemos D1 mini R2
- Flash Mode:		not sure
- Flash Size: 		4MB
- lwip Variant:   	tested with all v2 types (Lower Memory/Higher Bandwidth)
- Reset Method:		not sure, all breakouts have RST btn
- Flash Frequency:	40Mhz
- CPU Frequency:	80Mhz
- Upload Using:		SERIAL
- Upload Speed:   	tested all speeds from 115200 up

### Problem Description
First of all, I have search the internet for hours including [#4949](https://github.com/esp8266/Arduino/issues/4949) not findin a solution!
The sketch connects to a wifi and wants to access a Windows host in LAN using Windows hostname (e.g. `DESKTOP-123` as given in Windows 10 This computer's properties).
The sketch uses 2 methods one after each other (an in loop):
1) ping the host using `ESP8266Ping` library
2) direct call of `WiFi.hostByName()` from `ESP8266WiFi` library

(The ping uses a call to `hostByName` anyway.)
The problem is, that ping from ESP always fails with `DNS lookup error: -5!`
Ping within the same network from another PC host works fine.
Ping from ESP to a host in internet (e.g. `www.google.com`) works fine too.
I have tested this in 8 different LANs with different APs.

I understand, that DNS lookup may not resolve Windows host names (NetBios over TCP is not working here, as it would on a PC host).
Any idea, how to solve this?
Thanks for any help.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266Ping.h>

ESP8266WiFiMulti WiFiMulti;
String remote_host = "DESKTOP-BGHIGLC";

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  WiFiMulti.addAP("<ssid>", "<pwd>");
}

void loop() {
  if ((WiFiMulti.run() == WL_CONNECTED)) { // wait for WiFi connection
    WiFi.printDiag(Serial);
    Serial.print(">>> PING: ");
    Serial.println(remote_host);
    if (Ping.ping(remote_host.c_str())) {
      Serial.println(">>> PING: Success!!");
    } else {
      Serial.println(">>> PING: Error");
    }
    delay(1000);
    printIPAddressOfHost(remote_host.c_str());  // call to resolve host name to IP
  }
  delay(3000);
}

void printIPAddressOfHost(const char* host) {
  IPAddress resolvedIP;
  Serial.print(">>> Resolve DNS: ");
  Serial.print(host);
  Serial.print(" IP: ");
  if (!WiFi.hostByName(host, resolvedIP)) {
    Serial.println(">>> DNS lookup failed.");
  } else {
    Serial.println(resolvedIP);
  }
}
```

### Debug Messages
`*********` replace real SSID and password.
```
SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:1.4.0rc2/BearSSL:6778687
----------------------------------------
[SETUP] WAIT 4...
[SETUP] WAIT 3...
[SETUP] WAIT 2...
[SETUP] WAIT 1...
bcn 0
del if1
usl
mode : sta(80:7d:3a:74:f3:e3)
add if0
[WIFI][APlistAdd] add SSID: *********
[WIFI] delete old wifi config...
[WIFI] start scan
wifi evt: 8
scandone
[WIFI] scan done
[WIFI] 7 networks found
       0: [1][CC:5D:4E:ED:15:F8] CAT_CAFE (-63) *
 --->  1: [6][30:45:96:91:E1:F2] ********* (-47) *
       2: [8][AC:22:05:62:D6:6C] UPC68FBA54 (-86) *
       3: [11][EC:43:F6:AD:0D:7E] CAT_BYRO (-84) *
       4: [11][02:15:99:9E:F2:B3] DIRECT-cBML-2160 (-77) *
       5: [13][60:31:97:7B:BE:5A] CAT_CAFE (-82) *
       6: [13][00:13:F7:95:28:82] 225-162-35 (-85) *
[WIFI] Connecting BSSID: 30:45:96:91:E1:F2 SSID: ********* Channel: 6 (-47)
wifi evt: 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 
state: 5 -> 2 (4a0)
rm 0
wifi evt: 1
STA disconnect: 4
reconnect
state: 2 -> 0 (0)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with *********, channel 6
dhcp client start...
wifi evt: 0
[WIFI] Connecting Failed (6).
ip:192.168.43.25,mask:255.255.255.0,gw:192.168.43.1
wifi evt: 3
......................................
Mode: STA
PHY mode: N
Channel: 6
AP id: 0
Status: 5
Auto connect: 1
SSID (6): *********
Passphrase (10): *********
BSSID set: 1
>>>IP: 192.168.43.25
>>>Subnet mask: 255.255.255.0
>>>Gataway IP: 192.168.43.1
>>>Default hostname: ESP_74F3E3
>>> PING: DESKTOP-BGHIGLC
[hostByName] request IP for: DESKTOP-BGHIGLC
[hostByName] Host: DESKTOP-BGHIGLC lookup error: -5!
>>> PING: Error
>>> DNS #1, #2 IP: 192.168.43.1, (IP unset)
>>> Resolve DNS: DESKTOP-BGHIGLC IP: [hostByName] request IP for: DESKTOP-BGHIGLC
[hostByName] Host: DESKTOP-BGHIGLC lookup error: -5!
>>> DNS lookup failed.
......................................
Mode: STA
PHY mode: N
Channel: 6
AP id: 0
Status: 5
Auto connect: 1
SSID (6): *********
Passphrase (10): *********
BSSID set: 1
>>>IP: 192.168.43.25
>>>Subnet mask: 255.255.255.0
>>>Gataway IP: 192.168.43.1
>>>Default hostname: ESP_74F3E3
>>> PING: www.google.com
[hostByName] request IP for: www.google.com
[hostByName] Host: www.google.com IP: 172.217.23.228
>>> PING: Success!!
ping 5, timeout 0, total payload 160 bytes, 5035 ms
>>> DNS #1, #2 IP: 192.168.43.1, (IP unset)
>>> Resolve DNS: www.google.com IP: [hostByName] request IP for: www.google.com
[hostByName] Host: www.google.com IP: 172.217.23.228
172.217.23.228
```

5chufti, 2019-04-12:
If your windows box does not register its name with the DNS on your router it won't work.

In general this is not a problem of the esp8266 arduino core but of your knowledge about networking, so please close this issue and look for help in some more appropriate forum like https://www.esp8266.com

milanbx, 2019-04-12:
OK. Sorry.

closed 2019-04-12 by milanbx