esp8266/Arduino #5990: OTA stopped working completely with 2.5.0

paai, 2019-04-16:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

d-a-v, 2019-04-16:
> OTA stopped working completely with 2.5.0

> I have tested that the issue is present in current master branch (aka latest git).

Those two are unclear. There were fixes in git since 2.5.0

What did you really try, release-2.5.0 or latest git ?

paai, 2019-04-16:
Sorry I got lost in the directions on how to submit issues. They are not immediatly clear.

paai, 2019-04-16:
As I said, sometimes a few bullets appear before the 'not responding'. Other sketches incorporate a small webserver, and I can access the webserver no problem, so Wifi works. Just OTA does not work.

paai, 2019-04-16:
@David: this was with the release-2.5.0. I have not yet found out how to install the latest git.

d-a-v, 2019-04-16:
The first point in the main readme on the frontpage of this repository has links do documentation

devyte, 2019-04-17:
This is supposed to be fixed already in latest git, the cause was a missed configuration in the lwip compile time options.
@paai please test with latest git and provide feedback.

paai, 2019-04-18:
sorry, away from computer the next few days.
paai

paai, 2019-04-20:
OK, I am back. I found the documentation on how to install the 'normal' git (how else could I have installed this library to begin with), but nowhere how to install the 'latest' git as you suggest. Went to directory ~/arduino-1.8.9/hardware/esp8266com/esp8266 and issued the 'git status' and 'git pull' commands. Some information was displayed on getting espSoftwareSerial, but OTA still not working.
Sorry, I certainly am not a newbee, but this git-stuff is as yet very strange.

Thanks.

kurtovic, 2019-04-20:
I was in a similar situation recently experiencing persistent issues with OTA updates on release-2.5.0, but testing the latest git solved the problem immediately.

@paai 
https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version

paai, 2019-04-21:
@kurtovic, yes, that is what I did twice, yesterday and today. I am working on the assumption that installing  'the latest git' taking the steps as described in that link, automatically installs the very last version, that is to say that there is not somewhere a 'developers version' as opposed to a stable version.
Can I check the exact version that I installed on my PC, to see if there are minor version numbers that may differ?

For what it is worth: when I try to update the sketch On The Air and looking at the serial output with debugging CORE+WIFI+HTTP-UPDATE+UPDATER+OTA... enabled and the loop in the sketch just printing the word "idioot", I get this:
...
idioot
:urch 40, 40
:urd 2, 40, 0
:urd 1, 40, 2
:urd 1, 40, 3
:urd 2, 40, 4
:urd 2, 40, 6
:urd 2, 40, 8
:urd 2, 40, 10
idioot
...

Paai

tbdltee, 2019-04-22:
I also have the same problem and try every combination without success. So I decide to fall back to 2.4.2. Here are my combination test:
**Platform**
Hardware:	[esp8266]
Core Version:	[see below]
Development Env:	[Arduino IDE 1.8.9]
Operating System:	[Windows 10]
**Settings in IDE**
Module: [ESP-01s]
Flash Mode:	[dio]
Flash Size: [1MB/no SPIFFS]
lwip Variant: [v2 Lower Memory]
Reset Method:	[ck]
Flash Frequency: [40Mhz]
CPU Frequency:	[80Mhz]
Upload Using:	[OTA]
Upload Speed: [115200]

case 1:  SDK 3.0.0/Core 2.5.0.  No OTA port shown on ArduinoIDE. HTTP OTA failed.
case 2:  SDK 3.0.0/Core 2.5.0 git/choose SDK3.0.0-pre.   OTA port found but ESP return "No Answer"
case 3:  SDK 2.2.1/Core 2.5.0.  No OTA port shown on ArduinoIDE.
case 4:  SDK 2.2.1/Core 2.5.0 git/choose SDK2.2.1 (legacy).   OTA port found but ESP return "No Answer"

I saw the git version was updated 2 days ago. I will try git version again. Since my test was 2 weeks old.
/tee

paai, 2019-04-22:
@Teerapong, could you tell me how to fall back to that version?
Paai

tbdltee, 2019-04-22:
If you want to fall back to 2.4.2, in the arduino IDE goto Tools -> Boards -> Boards Manager
in the search box, type esp8266. At the lower left drop down list, choose 2.4.2

but when I installled 2.5.0 from git, I just delete the whole esp8266 git folder and restart the IDE

> @Teerapong, could you tell me how to fall back to that version?
> Paai

paai, 2019-04-22:
@Teerapong: YESSS! Finally someone told me how to how to change versions! 
I was looking all over the git documentation on how to do that from the command line, but I should have been looking in the IDE. That comes with being an old fashioned CLI nerd.

Paai

devyte, 2019-04-22:
The git install and installed versions from the board manager are mutually exclusive. If you do one, then you have to uninstall the other.
The git install allows to sync to a particular release version by syncing to the release tag, or to the hash of the release. It is therefore more general and flexible, and you should never have to go back to the board manager, ever again.

tbdltee, 2019-04-22:
Of course. I unstalled from board manager before install git version and vice versa
br/tee

> The git install and installed versions from the board manager are mutually exclusive. If you do one, then you have to uninstall the other.
> The git install allows to sync to a particular release [...]

paai, 2019-04-22:
@devyte: "syncing to the release tag"... What is the release tag and which CLI command should I give?
In any case, I have not much luck yet, even though I now tried the versions that are displayed in the board manager. Going back to 2.4.2 did not help. Sometimes five or six upload-dots are displayed before the No Answer error message appears. Tried different baudrates, different ports. There are no firewalls in this network. Everything.

The esp8266 board is a Robotdyn D1R2. I tried three of them, so the chance of all three being faulty is small. If I upload a sketch that contains a httpserver or -client, they work flawlessly, so the Wifi connection should be good. Tried the LOLIN(WEMOS)D1R2 and the WEMOS D1R1. 

The sketch is the BasicOTA example with only a Serial.println("idioot"); and delay(500); added.

Totally at wits end...

// #include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

#ifndef STASSID
#define STASSID "yyyyy"
#define STAPSK  "xxxxx"
#endif

const char* IP_host = "192.168.178.43";
IPAddress ip(192, 168, 178, 43);
IPAddress gateway(192, 168, 178, 1); // set gateway to match your network
IPAddress subnet(255, 255, 255, 0); // set subnet mask to match your network

const char* ssid = STASSID;  
const char* password = STAPSK;

void setup() { 
  Serial.begin(115200);
  Serial.println("\nBooting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  yield();
  WiFi.config(ip, gateway, subnet, gateway); Serial.println("3");
  
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
  // ArduinoOTA.setPort(8080);

  // Hostname defaults to esp8266-[ChipID]
   ArduinoOTA.setHostname("idioot");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA.onStart([]() {
    // Serial.end();
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH) {
      type = "sketch";
    } else { // U_SPIFFS
      type = "filesystem";
    }

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) {
      Serial.println("Auth Failed");
    } else if (error == OTA_BEGIN_ERROR) {
      Serial.println("Begin Failed");
    } else if (error == OTA_CONNECT_ERROR) {
      Serial.println("Connect Failed");
    } else if (error == OTA_RECEIVE_ERROR) {
      Serial.println("Receive Failed");
    } else if (error == OTA_END_ERROR) {
      Serial.println("End Failed");
    }
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  // yield();
  ArduinoOTA.handle(); 
  Serial.println("idioot");
  delay(500);
}

tbdltee, 2019-04-23:
@paii what SDK versnio are you using?  You can see by using Serial.println (ESP.getSdkVersion());
SDK 3.0.0 seem to have problem for both Core 2.4.2 and 2.5.0 (from board manager). I also need to downgrade to SDK 2.2.1

I have not try SDK 3.0.0 with Core 2.5.0 lastest git version yet.

earlephilhower, 2019-04-23:
Please use GIT head.  I just tried OTA using the BasicOTA.ino sketch (for the first time ever) and it just worked, and crazy fast, too.  I don't know what version broke it, but I know it runs under the latest head.

As said before by other, **you can not use the Arduino IDE installer to get this version.**

The directions are straightfoward, link below:
https://github.com/esp8266/Arduino#using-git-version-basic-instructions

devyte, 2019-04-23:
There were issues with 2.5.0, but they have since been fixed. 
Closing as can't reproduce.

closed 2019-04-23 by devyte

tbdltee, 2019-04-23:
**@devyte** I can confirmed. This issue is resolved in the lastest git.
br/Tee

paai, 2019-04-23:
Been working on this all over Eastern. I did set up a new wifi router three meters from my desk with the ESP8266 and PC and now I can sometimes upload sketches using OTA. Changing baud rates does not change things. I live out in the country side, so only four or five wifi stations are visible, three of which are mine.

As I said, the boards are Robotdyn  D1 R2, that always have performed satisfactorily. I tried a WEMOS Mini too, with the same results.

Conclusion: for some reason the OTA upload performs badly and mostly times out before the sketch is sent. Perhaps the developers could look into this?

Paai

paai, 2021-05-16:
Left it for two years because suddenly it worked. A few months ago it broke down again.
Uninstalled all traces of arduino and libraries  from my computer. Installed the last Arduino IDE 1.8.15 followed immediatly by the git version as described here. No other libraries or anything. No firewall on this PC. 

Still the same behavour, i.e. OTA update stops around 25-50%.

d-a-v, 2021-05-17:
Related: #4054 and #6369
This issue has "2.5.0" in the title and is closed.
Please provide more insights in a new issue or one of the two ones cited above.
Was your router software updated ? your computer's OS ?
A tcpdump trace around where the transfer stops would also help.