esp8266/Arduino #5999: Can Delay(..) be interrupted?!

fabiuz7, 2019-04-19:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			v2.5.0
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   v1.4
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL

### Problem Description

This is the sketch which shows delay(..) doesn't work as expected (i.e. delay is blocking until esp_schedule() is called). Maybe it is a desidered and righteous behaviours, but for me wasn't expected.

In general: given that this repo is not using a traditional OS, is there any documentation for these functions?

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

``` cpp
/**
 * Test to check if a delay(..) could be interrupted
 * by an esp_schedule/esp_yield.
 */
#include <Ticker.h>
#include <coredecls.h>

Ticker ticker;

int a = 0;
int b = 0;
void foo(){
  a++;
  
  // Reschedule the tasks, this means that delay(..) will be broken
  esp_schedule();
  // Give back the control to the system, no rescheduling
  // is performed i.e. delay will work as desidered
  //esp_yield();

  // Eventually this instruction is executed
  b++;
}

void setup() {
  Serial.begin(115200);
  Serial.println("Start");

  ticker.attach(0.5, foo);
}

void loop() {
  Serial.println(b);
  delay(5000);
}

```

igrr, 2019-04-19:
Historically it was assumed that esp_schedule and esp_yield are internal functions, and are not intended to be used from sketch code. The code (inside this library) which uses esp_schedule/esp_yield is written in a way that sequences these functions properly, so under normal circumstances a delay should not be interrupted by an unrelated esp_yield. However, no explicit checking is done.
If these functions are to be exposed for general use, some 'tokens' need to be implemented to guarantee correct sequence.

Although I am pretty sure that a solution to your original problem exists, which doesn't involve the use of esp_schedule/yield.

fabiuz7, 2019-04-19:
It was more a curiosity, raised by the fact that this weird behaviour happens with WebServer. But probably was an improper usage of WebServer + Ticker library.

In particular this is the code: 
```cpp
Ticker ticker;
void apModeActions(){
    server.handleClient();
}

void setup(){
  ticker.attach(0.25, apModeActions);
}

void loop(){
  delay(10000);
  Serial.println("a");
}
```

I can note that loop() it not always blocked for 10 seconds

closed 2019-04-19 by devyte

devyte, 2019-04-19:
The Ticker callback executes in the SYS context. Our synchronous webserver is meant only for execution in the CONT context.
Why do you want to handle only every 0.25? The handleClient call doesn't do anything unless there is actually a new connection. If you really need to handle less often, which I think is a bad idea, then i suggest using a periodic polledTimeout.

fabiuz7, 2019-04-19:
OK thanks, the difference between SYS and CONT context wasn't very clear (I didn't know the existence of these different contexts). I've decided to use Ticker because i don't want to "pollute" the main loop. But, given that ticker is non intended for this purpose, I will switch back to traditional loop(..) approach.
0.25 was decides because the performace is good enough.

fabiuz7, 2019-04-19:
As final question: aren't SPI / Wire libraries recommend to use withing Ticker context?

From the [official docs](https://arduino-esp8266.readthedocs.io/en/latest/libraries.html#ticker) there is reference to network, serial and file.

devyte, 2019-04-19:
Correct. You can still use Ticker, but use the scheduled methods to run the callback in CONT as if you were calling it from the loop.

fabiuz7, 2019-04-19:
Sorry, stilla new question, to confirm my understanding:

When you say _but use the scheduled methods to run the callback in CONT_ you mean the method Ticker::attach_ms_scheduled(..); ?

hreintke, 2019-04-20:
@fabiuz7 
You can do that without using Ticker.

```
void apModeActions(){
     server.handleClient();
    schedule_function(apModeActions);
}
```
And call `apModeAcions()` once within setup