esp8266/Arduino #6001: GET request exception 29

dmdt, 2019-04-19:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				Nodemcu v1.0 (esp-12e)
- Core Version:			v2.4.2 
- Development Env:		Arduino IDE/PlatformIO
- Operating System:		Windows

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			default
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

In function which recieve data from server over HTTPS protocol, in case when certificate is incorrect ESP throws exception. Decoded stack below:
```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 20 results
0x4020588e: WiFiClient::WiFiClient() at ?? line ?
0x4010654d: millis at ?? line ?
0x4020ac7c: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at ?? line ?
0x4010654d: millis at ?? line ?
0x4020ac7c: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at ?? line ?
0x401004e4: malloc at ?? line ?
0x40209950: operator new(unsigned int) at ?? line ?
0x40205c56: BearSSLTraits::create() at ?? line ?
0x40206359: HTTPClient::connect() at ?? line ?
0x40206b8c: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at ?? line ?
0x40207ffc: HardwareSerial::write(unsigned char const*, unsigned int) at ?? line ?
0x402082a5: Print::write(char const*) at ?? line ?
0x40206c3e: HTTPClient::GET() at ?? line ?
0x40202aec: Strip_Control::_getData() at ?? line ?
0x40209b90: esp_yield at ?? line ?
0x4020360b: Strip_Control::start() at ?? line ?
0x40203c7c: loop at ?? line ?
0x40209c1c: loop_wrapper() at core_esp8266_main.cpp line ?
0x40100721: cont_wrapper at ?? line ?
```
Function _getData() calls every 2 seconds. On 2-3d call appears exception. 
Intresting fact, when I disable debug (DEBUG_S, DEBUG_N functions) all works well, and exception do not appears. Also I wrote simple sketch where just 10 requests to server - all requests goes through.

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

#define SSID "ssid"
#define PASS "pass"

void setup() {
    Serial.begin(115200);
    WiFi.begin(SSID,PASS);
    while(WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(500);
    }
    Serial.println("\nWifi connected!");

    uint8_t cert[20] = {42+1,180,2,164,216,43,148,148,229,203,186,227,194,178,70,121,142,36,244,105};

    HTTPClient https;
    https.setReuse(true);
    String url = "url";
    https.begin(url,cert);
    Serial.println(https.GET());
    Serial.println(https.connected());
    for(uint8_t i = 0; i < 10; i++) {
        Serial.print(String(i)+" : ");
        Serial.println(https.GET());
        Serial.println(https.getString());
        delay(1000);
    }
    
}

void loop() {
}
```

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

void Strip_Control::_getData() {
    _response_code = _https.GET();
    DEBUG_N("[DEBUG]Request sent.");
    if (_response_code == 200) {
        String response;
        response = _https.getString();
        DEBUG_N("[DEBUG]Old response: " + _response);
        DEBUG_N("[DEBUG]Recieved response: " + response);
        if (_response == response) {
            return;
        } else {
            _response = response;
            _isNewData = true;
            return;
        }
    }
    _isNewData = false;
    DEBUG_S("[DEBUG]Request error. Response code: ");
    DEBUG_N(_response_code);
    return;
}

```

### Debug Messages

```
[DEBUG]Connecting to API - failed.
[DEBUG]Request sent.
[DEBUG]Request error. Response code: -1

Exception (29):
epc1=0x4020588e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000
```

earlephilhower, 2019-04-19:
We need the full crash dump, including the line showing where it failed, including line numbers to look at it.  Please re-run and decode the entire exception output.

dmdt, 2019-04-19:
```
Exception (29):
epc1=0x402064b6 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3ffffc10 end: 3fffffd0 offset: 01a0

last failed alloc call: 000007D0(176)

```

```
Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores
Decoding 20 results
0x402064b6: Print at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiClient.cpp line 92
:  (inlined by) Stream at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Stream.h line 51
:  (inlined by) Client at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Client.h line 26
:  (inlined by) WiFiClient::WiFiClient() at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiClient.cpp line 51
0x40104ff0: system_get_time at ?? line ?
0x40207254: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.cpp line 1300
0x4010654d: millis at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/core_esp8266_wiring.c line 171
0x40207254: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.cpp line 1300
0x401004e4: malloc at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266\umm_malloc/umm_malloc.c line 1668
0x4020b8c0: operator new(unsigned int) at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/abi.cpp line 71
0x40207f22: BearSSL::WiFiClientSecure::setFingerprint(unsigned char const*) at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 53
:  (inlined by) BearSSLTraits::create() at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 87
0x4020861e: std::unique_ptr   >::get() const at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 53
:  (inlined by) std::unique_ptr   >::release() at c:\users\dmtprs\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits/unique_ptr.h line 251
:  (inlined by) std::unique_ptr   >::operator=(std::unique_ptr   >&&) at c:\users\dmtprs\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits/unique_ptr.h line 192
:  (inlined by) HTTPClient::connect() at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 905
0x40208e48: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 53
0x40209f70: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/HardwareSerial.cpp line 133
0x4020a219: Print::write(char const*) at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/Print.cpp line 138
0x40208ef6: HTTPClient::GET() at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 53
0x40202aec: Strip_Control::_getData() at C:\Repository\Development\Arduino\Light/Light.ino line 457
0x4020bb00: esp_yield at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/core_esp8266_main.cpp line 91
0x40203673: Strip_Control::start() at C:\Repository\Development\Arduino\Light/Light.ino line 523
0x40203ce4: loop at ?? line ?
0x4020bb8c: loop_wrapper at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/core_esp8266_main.cpp line 125
0x40100739: cont_wrapper at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/cont.S line 81

```

earlephilhower, 2019-04-19:
> 0x402064b6: Print at C:\Users\dmtprs\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi\src/WiFiClient.cpp line 92

> last failed alloc call: 000007D0(176)

You're out of memory and running a very old release, it's crashing in a debug message from BearSSL reporting the failure.  I suggest migrating to the latest release or GIT and re-retying as there were memory saving changes in the months since that 2.4.2 release

dmdt, 2019-04-19:
Firstly I wrote code on 2.5.0, but I have a lot of bugs on it. Such as not working OTA over HTTPS [https://github.com/esp8266/Arduino/issues/5818](https://github.com/esp8266/Arduino/issues/5818). Also on 2.5.0 sketch can work after manipulating rows (separate println's on 2 - sketch working, when in one row - not working). And latest problem that knocks me to 2.4.2 is that responses from GET() not modifies, i.e. I get data (1), then I change data on server to (2), watch at recieved data and there is still data (1).
And so why when I disable Serial all works fine? In 100% of cases.

earlephilhower, 2019-04-19:
Not much to say, it's hard out of space and printf() doesn't have enough memory to do its work at that point and boom.

No debug == more free memory, so you might be able to just squeeze it into the heap.  HTTPS w/o MFLN takes a very large amount of RAM, that's the nature of the beast.

dmdt, 2019-04-19:
Just now tried to compile same sketch at 2.5.0 and gets Soft WDT reset at update checking.

As I described in previous issue, when I enables debug with certain level (on 2.5.0) it's starts to updating.

Is there a way out to OTA update?

earlephilhower, 2019-04-19:
Seems like there's nothing here related to the GET other than OOM.

Please track the OTA stuff in the original bug you had there.  Try the GIT version, not a release 2.5.0, as in the OTA bug I remember comments about a specific LWIP bug being fixed after 2.5.0 which unblocks OTA.

closed 2019-04-19 by earlephilhower