esp8266/Arduino #6004: memory leak when call begin() in WiFiUDP under AP_STA mode

ckyou, 2019-04-22:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:  ESP-12
- Core Version:   2.5.0 & last git(0dd6549c3667550c819b7685a1063cb86c0bc4bd)
- Development Env:  Arduino IDE
- Operating System:  Ubuntu

### Settings in IDE

- Module:  Nodemcu 1.0(ESP-12E module)
- Flash Mode: dio
- Flash Size:  4MB/1MB
- lwip Variant: v2 Lower Memory
- Reset Method: nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:	80Mhz
- Upload Using: SERIAL
- Upload Speed: 115200

### Problem Description

Just find that if setup UDP and AP_STA (or AP) mode, will cause memory leak slowly  
p.s. don't know this issue is exist before 2.5.0 or not

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <Ticker.h>
#include <WiFiUdp.h>

WiFiUDP Udp;
Ticker showHeap;

void printHeapSize()
{
  Serial.print("heap size: ");
  Serial.println(ESP.getFreeHeap());
}

void setup()
{
  Serial.begin(115200);
  WiFi.persistent(false);
  WiFi.mode(WIFI_OFF);
  WiFi.mode(WIFI_AP);

  WiFi.begin("homeAP", "00000000");
  WiFi.softAPConfig(IPAddress(192, 168, 8, 1), IPAddress(192, 168, 8, 1),
                    IPAddress(255, 255, 255, 0));
  WiFi.softAP("tryIt", "12345678");

  Udp.begin(1234);

  showHeap.attach(10, printHeapSize);
}

void loop() {}
```

### Debug Messages

```
16:58:09.616 -> heap size: 49592
16:58:19.632 -> heap size: 49592
16:58:29.619 -> heap size: 49592
16:58:39.635 -> heap size: 49592
16:58:49.618 -> heap size: 49592
16:58:59.634 -> heap size: 49592
16:59:09.615 -> heap size: 49592
16:59:19.628 -> heap size: 49288
16:59:29.611 -> heap size: 49544
16:59:39.627 -> heap size: 49544
16:59:49.609 -> heap size: 49544
16:59:59.619 -> heap size: 49544
17:00:09.627 -> heap size: 49496
17:00:19.610 -> heap size: 49448
17:00:29.624 -> heap size: 49448
17:00:39.636 -> heap size: 49400
17:00:49.612 -> heap size: 49400
17:00:59.618 -> heap size: 49400
17:01:09.623 -> heap size: 49096
17:01:19.622 -> heap size: 49400
17:01:29.629 -> heap size: 49400
17:01:39.633 -> heap size: 49352
17:01:49.608 -> heap size: 49352
17:01:59.612 -> heap size: 49304
17:02:09.619 -> heap size: 49304
17:02:19.627 -> heap size: 49304
17:02:29.634 -> heap size: 49304
17:02:39.616 -> heap size: 49304
17:02:49.631 -> heap size: 49304
17:02:59.610 -> heap size: 49304
17:03:09.623 -> heap size: 49304
17:03:19.630 -> heap size: 49256
17:03:29.608 -> heap size: 49208
17:03:39.617 -> heap size: 49208
17:03:49.627 -> heap size: 49208
17:03:59.634 -> heap size: 49112
17:04:09.611 -> heap size: 49112
17:04:19.616 -> heap size: 49064
17:04:29.625 -> heap size: 49064
17:04:39.602 -> heap size: 49064
17:04:49.618 -> heap size: 49064
17:04:59.627 -> heap size: 49064
17:05:09.631 -> heap size: 49064
17:05:19.626 -> heap size: 49064
17:05:29.602 -> heap size: 48968
17:05:39.616 -> heap size: 48968
17:05:49.624 -> heap size: 48920
....
```

d-a-v, 2019-04-22:
Does it still happen even if you effectively read udp data ?

ckyou, 2019-04-22:
> Does it still happen even if you effectively read udp data ?

Have tried this, doesn't help. Or maybe I misunderstand [documentation](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/udp-examples.html)? 
```cpp
char incomingPacket[255]; 
void loop() 
{
  int packetSize = Udp.parsePacket();
  if (packetSize) Udp.read(incomingPacket, 255);
}
```

d-a-v, 2019-04-22:
I can't reproduce. What are your setting when using git ?

```
23:24:23.855 -> SDK:2.2.1(cfd48f3)/Core:2.5.0-93-g0dd6549c=20500093/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:6778687
23:24:23.903 -> mode : sta(ec:fa:bc:0d:1b:d0) + softAP(ee:fa:bc:0d:1b:d0)
23:24:23.903 -> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
23:24:28.406 -> connected with xxxxx, channel 11
23:24:28.406 -> dhcp client start...
23:24:28.506 -> ip:10.0.1.155,mask:255.255.255.0,gw:10.0.1.254
23:24:34.748 -> heap size: 49120
23:24:44.744 -> heap size: 49120
23:24:54.744 -> heap size: 49120
23:25:04.739 -> heap size: 49120
23:25:14.745 -> heap size: 49120
23:25:24.746 -> heap size: 49120
23:25:34.755 -> heap size: 49120
23:25:37.878 -> :urn 255
23:25:37.878 -> :urd 255, 255, 0
23:25:40.603 -> :urn 7
23:25:40.603 -> :urd 7, 7, 0
23:25:42.265 -> :urn 6
23:25:42.265 -> :urd 6, 6, 0
23:25:44.732 -> heap size: 48528
23:25:49.979 -> :urn 255
23:25:49.979 -> :urd 255, 255, 0
23:25:53.068 -> :urn 7
23:25:53.068 -> :urd 7, 7, 0
23:25:54.563 -> :urn 12
23:25:54.563 -> :urd 12, 12, 0
23:25:54.762 -> heap size: 48528
23:25:55.925 -> :urn 9
23:25:55.925 -> :urd 9, 9, 0
23:26:04.761 -> heap size: 48528
23:26:08.382 -> wifi evt: 7
23:26:14.757 -> heap size: 48528
23:26:24.756 -> heap size: 48528
23:26:34.758 -> heap size: 48528
23:26:44.747 -> heap size: 48528
23:26:54.747 -> heap size: 48528
23:27:04.745 -> heap size: 48528
23:27:14.741 -> heap size: 48528
23:27:24.768 -> heap size: 48528
23:27:34.736 -> heap size: 48528
23:27:44.738 -> heap size: 48528
23:27:54.768 -> heap size: 48528
23:28:04.764 -> heap size: 48528
23:28:14.758 -> heap size: 48528
23:28:24.751 -> heap size: 48528
23:28:34.747 -> heap size: 48528
23:28:39.675 -> :urn 255
23:28:39.675 -> :urd 255, 255, 0
23:28:41.071 -> :urn 9
23:28:41.071 -> :urd 9, 9, 0
23:28:44.765 -> heap size: 48528
23:28:54.768 -> heap size: 48528
23:29:04.756 -> heap size: 48528
23:29:14.757 -> heap size: 48528
23:29:24.758 -> heap size: 48528
23:29:34.767 -> heap size: 48528
23:29:44.748 -> heap size: 48528
23:29:53.637 -> :urn 255
23:29:53.637 -> :urd 255, 255, 0
23:29:54.766 -> heap size: 48528
23:30:04.768 -> heap size: 48528
23:30:14.752 -> heap size: 48528
23:30:24.766 -> heap size: 48528
```

```
#include <ESP8266WiFi.h>
#include <Ticker.h>
#include <WiFiUdp.h>

WiFiUDP Udp;
Ticker showHeap;

void printHeapSize()
{
  Serial.print("heap size: ");
  Serial.println(ESP.getFreeHeap());
}

void setup()
{
  Serial.begin(115200);
  WiFi.persistent(false);
  WiFi.mode(WIFI_OFF);
  WiFi.mode(WIFI_AP_STA);
  WiFi.begin(STASSID, STAPSK);   // <-- I changed this
  WiFi.softAPConfig(IPAddress(192, 168, 8, 1), IPAddress(192, 168, 8, 1),
                    IPAddress(255, 255, 255, 0));
  WiFi.softAP("tryIt", "12345678");
  Udp.begin(1234);

  showHeap.attach(10, printHeapSize);
}

char incomingPacket[255];
void loop()
{
  int packetSize = Udp.parsePacket();
  if (packetSize) Udp.read(incomingPacket, 255);
}
```

ckyou, 2019-04-23:
Only modify board configuration in Arduino IDE > Tools.
And it's interesting that using lwip v1.4 doesn't have this behaviour in my environment.

lwip Variant: v2 Lower Memory
```
10:44:42.864 -> SDK:2.2.1(cfd48f3)/Core:2.5.0-93-g0dd6549c=20500093/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:6778687
10:44:42.864 -> bcn 0
10:44:42.864 -> del if1
10:44:42.864 -> scandone
10:44:42.864 -> del if0
10:44:42.864 -> usl
10:44:42.864 -> mode : null
10:44:42.864 -> mode : sta(5c:cf:7f:d0:fb:d2) + softAP(5e:cf:7f:d0:fb:d2)
10:44:42.864 -> add if0
10:44:42.864 -> add if1
10:44:42.864 -> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
10:44:42.897 -> bcn 100
10:44:43.762 -> bcn 0
10:44:43.762 -> del if1
10:44:43.762 -> add if1
10:44:43.762 -> dhcp server start:(ip:192.168.8.1,mask:255.255.255.0,gw:192.168.8.1)
10:44:43.762 -> bcn 100
10:44:46.487 -> scandone
10:44:47.384 -> state: 0 -> 2 (b0)
10:44:47.384 -> state: 2 -> 3 (0)
10:44:47.417 -> state: 3 -> 5 (10)
10:44:47.417 -> add 0
10:44:47.417 -> aid 1
10:44:47.417 -> cnt 
10:44:47.516 -> 
10:44:47.516 -> connected with mp, channel 1
10:44:47.516 -> dhcp client start...
10:44:48.680 -> ip:192.168.2.237,mask:255.255.255.0,gw:192.168.2.254
10:44:53.770 -> heap size: 48952
10:45:03.778 -> heap size: 48888
10:45:13.789 -> heap size: 48696
10:45:23.761 -> heap size: 48696
10:45:33.781 -> heap size: 48696
10:45:43.768 -> heap size: 48632
10:45:53.768 -> heap size: 48568
10:46:03.780 -> heap size: 48504
10:46:13.784 -> heap size: 48504
10:46:23.757 -> heap size: 48504
10:46:33.756 -> heap size: 48440
10:46:43.784 -> heap size: 48440
10:46:53.784 -> heap size: 48376
10:47:03.784 -> heap size: 48312
10:47:13.764 -> heap size: 48248
10:47:23.765 -> heap size: 48184
10:47:33.760 -> heap size: 48184
10:47:43.773 -> heap size: 48184
10:47:53.783 -> heap size: 48184
10:48:03.761 -> heap size: 48120
10:48:13.775 -> heap size: 48120
10:48:23.755 -> heap size: 48120
10:48:33.761 -> heap size: 48120
10:48:43.764 -> heap size: 48120
10:48:53.774 -> heap size: 48056
10:49:03.785 -> heap size: 48056
10:49:13.756 -> heap size: 47992
10:49:23.766 -> heap size: 47992
...
```

lwip Variant: v1.4 Compile from source
```
10:52:17.533 -> SDK:2.2.1(cfd48f3)/Core:2.5.0-93-g0dd6549c=20500093/lwIP:1.4.0rc2/BearSSL:6778687
10:52:17.533 -> bcn 0
10:52:17.533 -> del if1
10:52:17.533 -> scandone
10:52:17.533 -> del if0
10:52:17.533 -> usl
10:52:17.533 -> mode : null
10:52:17.533 -> mode : sta(5c:cf:7f:d0:fb:d2) + softAP(5e:cf:7f:d0:fb:d2)
10:52:17.533 -> add if0
10:52:17.533 -> add if1
10:52:17.533 -> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
10:52:17.533 -> bcn 100
10:52:18.349 -> bcn 0
10:52:18.349 -> del if1
10:52:18.349 -> add if1
10:52:18.349 -> dhcp server start:(ip:192.168.8.1,mask:255.255.255.0,gw:192.168.8.1)
10:52:18.382 -> bcn 100
10:52:21.107 -> scandone
10:52:22.005 -> state: 0 -> 2 (b0)
10:52:22.005 -> state: 2 -> 3 (0)
10:52:22.005 -> state: 3 -> 5 (10)
10:52:22.005 -> add 0
10:52:22.005 -> aid 1
10:52:22.005 -> cnt 
10:52:22.005 -> 
10:52:22.005 -> connected with mp, channel 1
10:52:22.005 -> dhcp client start...
10:52:23.631 -> ip:192.168.2.237,mask:255.255.255.0,gw:192.168.2.254
10:52:28.380 -> heap size: 48752
10:52:38.357 -> heap size: 48752
10:52:48.364 -> heap size: 48752
10:52:58.373 -> heap size: 48752
10:53:08.382 -> heap size: 48752
10:53:18.361 -> heap size: 48752
10:53:28.368 -> heap size: 48752
10:53:38.380 -> heap size: 48752
10:53:48.387 -> heap size: 48752
10:53:58.360 -> heap size: 48752
10:54:08.368 -> heap size: 48752
10:54:18.380 -> heap size: 48752
10:54:28.357 -> heap size: 48752
10:54:38.359 -> heap size: 48752
10:54:48.370 -> heap size: 48752
10:54:58.375 -> heap size: 48752
10:55:08.381 -> heap size: 48752
10:55:18.357 -> heap size: 48752
10:55:28.369 -> heap size: 48752
10:55:38.377 -> heap size: 48752
10:55:48.368 -> heap size: 48752
10:55:58.381 -> heap size: 48752
10:56:08.362 -> heap size: 48752
10:56:18.379 -> heap size: 48752
10:56:28.363 -> heap size: 48752
10:56:38.380 -> heap size: 48752
10:56:48.365 -> heap size: 48752
10:56:58.386 -> heap size: 48752
10:57:08.373 -> heap size: 48752
10:57:18.388 -> heap size: 48752
10:57:28.376 -> heap size: 48752
...
```

devyte, 2019-04-29:
What is the status here?

brandon007pillay, 2019-06-17:
Changing the Lwip Variant worked for me: 
[image]

devyte, 2020-02-02:
@ckyou @d-a-v what is the status here? is this still reproducible with latest git?

devyte, 2020-06-15:
Closing due to can't reproduce and lack of feedback.

closed 2020-06-15 by devyte