esp8266/Arduino #6005: Enhancement - Allow linking in smaller set of ciphers on BearSSL to save code space

earlephilhower, 2019-04-22:
This would be, unfortunately, yet another Arduino menu option where you would be able to define specific cipher groups (say, only RSA or only EC or only ChaCha20) would be available for SSL negotiation.

Make sure there are no references to the unused codes in question, and the linker will leave out large swaths of BearSSL code.

Useful for flash constrained people and folks trying to do OTA in 1MB devices.

earlephilhower, 2019-04-22:
Did some quick testing and we can drop ~45KB of code by only supporting the subset that AXTLS used to support which are still generally supported by servers in the wild.
````
    BR_TLS_RSA_WITH_AES_128_CBC_SHA256,
    BR_TLS_RSA_WITH_AES_256_CBC_SHA256,
    BR_TLS_RSA_WITH_AES_128_CBC_SHA,
    BR_TLS_RSA_WITH_AES_256_CBC_SHA,
````

sislakd, 2019-05-09:
It would be good to have there option to use only BR_TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256 which is forward secure, symmetric encryption/decryption is faster than AES on ESP8266 and is supported by servers long time.

At some moment I was heap & stack constrained. Thus I've performed detailed investigation and found some places which can help you save heap and stack space:

- drop X509 context after successful server verification (a.k.a null _x509_minimal, _x509_insecure and _x509_knownkey); (saves heap)
- reduce BSSL stack in StackThunk for selected cipher suites, e.g. BR_TLS_ECDHE_RSA_WITH_CHACHA20_POLY1305_SHA256 works well with stack 4300. For ciphers used by AxTLS (those 4) it requires even less stack; (saves heap)
- move constants from ECDHE KEX part of BSSL from RAM to FLASH and update few places to use pgmspace code to access constants; speed effect is only within KEX part and slow down is minimal even using default CPU & FLASH freq; (saves stack)

When I've tested some unstable SSL implementation at remote end, I found that WiFiClientSecureBearSSL is reporting that it is connected even BSSL is already in BR_SSL_CLOSED state (in my case it was caused by invalid MAC after some time sent from the remote peer to ESP8266). Change of condition in WiFiClientSecure::connected() to include test of BSSL state help to fail quickly and application can reconnect without implementing own response timeouts. Specifically, I've added (br_ssl_engine_current_state(_eng) != BR_SSL_CLOSED) into condition.

earlephilhower, 2019-05-09:
Those are great ideas!  Did you want to try a PR on them?

W/BSSL stack size, it'd really be best to key that off of the compiled-in ciphers and not runtime determined, of course, as you don't want to allocate/reallocate 5K chunks w/only 40KB free heap and no MMU.  But It can definitely be a function of the `#define`.  Now that there's an infrastructure and menus it's relatively straightforward.

The original BSSL was ported before the non 32b PROGMEM handler was included in the SDK, so I couldn't easily swap it out w/o massively changing (slowing) the EC code.  Today, that's a great idea to move them to flash.

The last bit looks like a bug on my part.  ::connected() with SSL is an odd character...

earlephilhower, 2019-05-09:
Reopening since there are some more good ideas here for 2.6.0...

sislakd, 2019-05-10:
Attaching PR for the first item.

earlephilhower, 2019-05-15:
Reopening since there are some more things we can do here (EC key stuff, heap sizing)

earlephilhower, 2019-05-28:
@sislakd , can you check the BearSSL repo and see if you have anything to add for saving heap?  The EC  P/N constants I can't move without massive hacking, but I've got everything else out of RODATA/heap and freed ~.5KB.

PR #6006 (2019-04-25): New menu option to minimize BSSL ROM with only RSA
Adds a menu option and define to limit BearSSL to older RSA connection
options.  This saves ~45K program memory and can speed up connections
since EC, while more secure, is significantly slower on the chip.
The supported ciphers are identical to the ones that axTLS supported.

Fixes #6005
commit d9b0480f (2019-04-25): New menu option to minimize BSSL ROM with only RSA (#6006)

* New menu option to minimize BSSL ROM with only RSA

Adds a menu option and define to limit BearSSL to older RSA connection
options.  This saves ~45K program memory and can speed up connections
since EC, while more secure, is significantly slower on the chip.
The supported ciphers are identical to the ones that axTLS supported.

Fixes #6005

* Add default SSLFLAGS(blank) to platform.txt

* Fix unused variable warning

* Add clarifying comment to menu items
commit 75f01dc3 (2019-05-15): Drop X509 after connection, avoid hang on TLS broken (#6065)

* Drop X509 context after successful server verification to save heap space

After completing handshake in BSSL, server is already verified and X509 context is no longer needed. Depending on verification method it save more or less heap space.

* Bugfix: Report not connected if there is no ready data and TLS connection is broken

Added the change for reporting not connected if TLS session is broken and there is no more buffered decrypted data. TLS can be broken if message authentication (MAC) cannot be verified. BearSSL enters BR_SS [... cut]
PR #6157 (2019-05-28): Save 484 bytes of heap for BSSL applications
Move additional constants to flash and use _P/pgm_read routines to
access them.  Minimal runtime impact, but remove variables from RODATA
and gives addition 484 bytes of heap to SSL applications.

Fixes #6005
closed 2019-05-28 by d-a-v