esp8266/Arduino #6009: Corrupted Arrays if using different arrays in [system_rtc_mem_read] & [system_rtc_mem_write]

Rob58329, 2019-04-23:
### Basic Infos
- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [n/a] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform
- Hardware:			ESP8266 D1-mini
- Core Version:			git at at 13Apr19
- Development Env:		Arduino IDE v1.8.9
- Operating System:		Windows10

### Settings in IDE
- Module:  			Wemos D1 mini r2 
- Flash Mode:			n/a
- Flash Size: 		       4MB
- lwip Variant:                 v2 Lower Memory
- Reset Method:		n/a
- Flash Frequency:	        n/a
- CPU Frequency:		[80Mhz|]
- Upload Using:		[SERIAL]
- Upload Speed:              [921600]

### Problem Description

Below is a test-sketch with both Describes and Demonstrates the issue:

```
// Demonstration of bug (the arrays get corrupted) when using...
// ...different arrays for [system_rtc_mem_read] and [system_rtc_mem_write] calls

extern "C" {
  #include "user_interface.h"
}

void setup() {
  Serial.begin(74880);
  byte rtcStore[2];

#if true // change to "false" to show working example
// if you use a different array in [system_rtc_mem_read] & [system_rtc_mem_write] calls...
// ...this example returns junk in BOTH arrays!
  Serial.println("Bug Example 1 (both arrays corrupted):");
  system_rtc_mem_read(64, rtcStore, 2);
  Serial.print("Start=");Serial.println(rtcStore[0]);
  rtcStore[0]++;
  Serial.print("New save=");Serial.println(rtcStore[0]);
  system_rtc_mem_write(64, rtcStore, 2);
  byte rtcRead[2];
  rtcRead[0]=0;
  rtcStore[0]=0;
  system_rtc_mem_read(64, rtcRead, 2); 
  Serial.print("Re-Read (rtcRead) value=");Serial.println(rtcRead[0]);
  Serial.print("Corrupted (rtcStore) value=");Serial.println(rtcStore[0]);

#elif false 
// if you use a different array in [system_rtc_mem_read] & [system_rtc_mem_write] calls...
// ...this example just does not save the [system_rtc_mem_write]
  Serial.println("Bug Example 2 (no save):");
  byte rtcRead[2];
  system_rtc_mem_read(64, rtcRead, 2);
  Serial.print("Start=");Serial.println(rtcRead[0]);
  rtcStore[0]=rtcRead[0]+1;
  Serial.print("New save=");Serial.println(rtcStore[0]);
  system_rtc_mem_write(64, rtcStore, 2);
  rtcRead[0]=0;
  rtcStore[0]=0;
  system_rtc_mem_read(64, rtcRead, 2); 
  Serial.print("Re-Read (rtcRead) value=");Serial.println(rtcRead[0]);
  Serial.print("Corrupted (rtcStore) value=");Serial.println(rtcStore[0]);

#else
// BUT using the SAME array in [system_rtc_mem_read] & [system_rtc_mem_write]" calls...
// ...then it works properly and return the correct value!
  Serial.println("Working Example:");
  system_rtc_mem_read(64, rtcStore, 2);
  Serial.print("Start=");Serial.println(rtcStore[0]);
  rtcStore[0]++;
  Serial.print("New save=");Serial.println(rtcStore[0]);
  system_rtc_mem_write(64, rtcStore, 2);
  rtcStore[0]=0;
  system_rtc_mem_read(64, rtcStore, 2);
  Serial.print("Re-Read (rtcStore) value=");Serial.println(rtcStore[0]);
#endif
}

void loop() {
}
```

### Debug Messages (example of the output from the above test-sketch):

(Eg1) Bug Example 1 (both arrays corrupted):
Start=239
New save=240
Re-Read (rtcRead) value=24
Corrupted (rtcStore) value=68

(Eg2) Bug Example 2 (no save):
Start=24
New save=25
Re-Read (rtcRead) value=24
Corrupted (rtcStore) value=68

(eg3) Working Example:
Start=23
New save=24
Re-Read (rtcStore) value=24

devyte, 2019-04-23:
The calls in question are Espressif's SDK API and not our core calls. Any issues with them should be directed at the NONOS sdk repo, and not here.
Having said that, and from looking at the [SDK api reference doc 2c](https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf), I suggest looking into the following:
-Compiler optimization. Try declaring your rtsStore volatile and/or try declaring it as a global.
-Alignment. The addresses passed as arg must be 4-byte aligned. Print out the address of rtcStore and check that it actually is 4-byte aligned.
-You're not checking the return value of the calls. How do you know the calls even succeeded?

In addition, I seem to remember that the first part of the user rtc space has special use related to the bootloader. It is supposed to be free if you don't use the updater class, e.g.: don't use OTA or similar, but I suggest trying with an rtc block higher up in the rtc space anyways to compare. 
Reference [here](https://github.com/esp8266/Arduino/blob/9515f466846f665fa68a7169465ee89aaa97b67e/cores/esp8266/Esp.cpp#L136).

Closing for now.

closed 2019-04-23 by devyte

Rob58329, 2019-04-25:
@devyte  -  Many thanks for your above "having said that" suggestions.

The key one was:
"Alignment. The addresses passed as arg must be 4-byte aligned. Print out the address of rtcStore and check that it actually is 4-byte aligned."

Checking using "Serial.println((long) &rtcStore,DEC);" etc confirmed that the pointers were not both 4-byte aligned. 

If anyone else is interested, the fix was to change the above "byte rtcStore[2];" to "byte rtcStore[4];", and this 4-byte aligned both the pointers and the above sketch worked.  (Or putting an extra 2-byte local variable-definition between rtcStore[2] and rtcRead[2] definitions also worked.) (Printing out the "return value of the calls" also showed which calls were failing at which point in the sketch.)

Interestingly, the fact that non-alignment corrupted both [2]-byte variables, suggests to me that the calls may actually perhaps be writing 4 bytes at a time, so always making these  "byte rtcStore[x];" etc variable-sizes a multiple of 4 bytes might be a sensible idea - but this is just my suspicion.

liebman, 2019-04-25:
I've not used this but maybe use: 'alignas' https://en.cppreference.com/w/cpp/language/alignas
alignas(4) byte rtcStore[2];
alignas(4) byte rtcRead[2];

Rob58329, 2019-04-25:
@liebman 
Thanks! Yes, "alignas(4) byte rtcStore[2];" does work as well!
* Interestingly, it occasionally seems to add a 5 byte space (for example) instead of a 1 byte space to do this though (although this might be something to do with the compilation-optimisation, as the total heap usage for the whole subroutine only increased by 3 bytes for this case).

* Also from the couple of subroutines I've checked, they have both started the first local variable at a 4-byte aligned address.  So it might be that if all subroutines do this (using ArduinoIDE for ESP8266), creating any required 4-byte(or multiple of 4 byte) variables at the top of the  subroutine may be enough to ensure they are 4-byte aligned?  (But as I say, I've only tried this on a couple of subroutines so far!...)  If not, then the "alignas(4)" command will be very usefull!!