esp8266/Arduino #6016: [Vulnerability] found - Question

Matheus-Garbelini, 2019-04-26:
Hello,

I've found a DoS Vulnerability and was wondering what is the correct procedure to follow (as I'm not sure if this would fit as a new CVE).
Would I need to report everything in detail here or directly to Espressif site?(https://www.espressif.com/en/company/contact-extra/technical-inquiries-software)

The issue in question appears to affect all esp8266 using NONOS SDK up to the latest version (3.0.0). This includes the current SDK version used in this repository. I've also created this issue in ESP8266_NONOS_SDK repository: [https://github.com/espressif/ESP8266_NONOS_SDK/issues/237](url)

If anyone has more information, let me know if this is not the correct place to handle this types of issues. Thanks.

devyte, 2019-04-26:
Is suspect your issue will be related to lwip or lwip integration.
Do you see the issue with our core or with a direct sdk build?
If with our core, please open a new issue here and follow the instructions in the issue template, especially an MCVE and instructions how to reproduce the DoS. Please also reference this issue.
If with a direct SDK build, then the correct place to report it is in Espressif's NONOS github repo.
As an FYI, Espressif's lwip is v1.4+, while we use v2.x, so how your app is implemented is important.

There have been some measures put in place in our core (lwip glue, really) to survive DoS attacks, but maybe you found something that was overlooked and can be mitigated. However, there is only so much that can be done on the tiny ESP.

I am closing this one due to lack of info.

closed 2019-04-26 by devyte

Matheus-Garbelini, 2019-04-27:
@devyte Thanks for the info.
What I found is actually related to their mac implemention, so I can trigger this issue not even touching IP stack, just plain 802.11. 
I'm going to follow your instructions then and reference the SDKs tested.

Thanks.

devyte, 2019-04-27:
In that case, their NONOS github repo would be the right place. Their freertos repo probably as well.

Matheus-Garbelini, 2019-05-01:
@devyte the issue has been reported and Espressif already fixed it. 
Just found another vulnerability for both esp8266 and esp32. Going for the bounty this time. :smile:

Matheus-Garbelini, 2019-08-19:
@earlephilhower @devyte @me-no-dev.  Just a quick update on the vulnerability matter.
It has been assigned the following CVEs: [CVE-2019-12586](https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2019-12586), [CVE-2019-12588](http://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2019-12588), [CVE-2019-12587](http://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2019-12587)
Espressif has [already fixed](214129bf542c832006bd26ad0ff21a3da7465ce6) in the latest nonos sdk with the libwpa. 

No public announcement has been made yet, but I strongly advise for at least libnet80211.a be updated in arduino SDK. CVE-2019-12586 and CVE-2019-12588 affect both ESP8266/ESP32 SDKs.

Thanks.