esp8266/Arduino #6019: Initializing SD card...initialization failed! with GIT  but ok with 2.5

luc-github, 2019-04-27:
SD support looks broken in GIT version **when it was Ok in 2.5**

### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:			        ESP-12
- Core Version:			latest git 04/27/2019
- Development Env:		Arduino IDE
- Operating System:		Windows 10

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			DOUT
- Flash Size: 		       4MB
- lwip Variant:                 v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	        40Mhz
- CPU Frequency:		160MHz
- Upload Using:		SERIAL
- Upload Speed:            115200

### Problem Description

SD card init failed with git version
using sample : https://github.com/esp8266/Arduino/blob/master/libraries/SD/examples/listfiles/listfiles.ino

`Initializing SD card...initialization failed!`

 **on same hardware with core 2.5 same code works**

Also tried the new library : 
https://github.com/earlephilhower/ESP8266SdFat/blob/92ab4b59eb920bf3134347dbe07e7c059eb64d46/examples/ReadWrite/ReadWrite.ino modifing 
`#define SD_CS_PIN 4` to not change the wiring 
same error:
`Initializing SD card...initialization failed!`

earlephilhower, 2019-04-28:
I've just run the ReadFiles.ino example on 2 cards, changing the one line to `D0` from `4` as that's the way I'm wired up:
````
    if (!SD.begin(D0)) {
````

On the 128MB Canon SDC-128MH card I get the expected output (Debug enabled):
````
SDK:2.2.1(cfd48f3)/Core:2.5.0-100-gcdb54957=20500100/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
Initializing SD card...initialization done.
Writing to test.txt...done.
test.txt:
testing 1, 2, 3.
````

And the new Samsung EVO Select U1 32GB I get the right output, too:
````
SDK:2.2.1(cfd48f3)/Core:2.5.0-100-gcdb54957=20500100/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-7-g82abda3/BearSSL:a143020
Initializing SD card...initialization done.
Writing to test.txt...done.
test.txt:
testing 1, 2, 3.
````

Assuming it is a card related issue, what can you tell us about the card?  Is it a very old MMC one?  What size, brand, type, any other info?

earlephilhower, 2019-04-28:
Also, calling @TD-er.  I think I remember you going to the GIT head for your ESPEasy.  Does it support SD cards, and if so have you or any of your users tried it or had issues like this?  Thx!

luc-github, 2019-04-28:
My card is Micro SD Kingston 4Gb  microSDHC – Classe 4 -  
I did not expect it was a SD card issue as it work on 3D printers and on version 2.5
It works also on ESP32 with SD library

I have also tried an ADATA 16GBMicro SD class 10 with same result

earlephilhower, 2019-04-28:
> Also tried the new library ReadWrite.ino modifing
> #define SD_CS_PIN 4 to not change the wiring
> same error:
> Initializing SD card...initialization failed!

The `SD_CS_PIN` define isn't used anywhere in the library code...
````
earle@server:~/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266SdFat$ grep -r SD_CS_PINT
earle@server:~/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266SdFat$ 
````

Nor is it used in the SD library:
````
earle@server:~/Arduino/hardware/esp8266com/esp8266/libraries/SD$ grep -r SD_CS_PINT
earle@server:~/Arduino/hardware/esp8266com/esp8266/libraries/SD$ 
````

Can you edit the readfiles.ino example and change the same line I showed `SD.begin(4)` -> `SD.begin(D0)` to your pin (making sure to use appropriate `D4` or whatever CS pin macro, or the raw integer if you're very sure of the mapping between 8266 internal GPI numbering and your board)?

luc-github, 2019-04-28:
I am not sure to understand what you mean SD_CS_PIN is used in file
```
#include <SPI.h>
//#include <SD.h>
#include "SdFat.h"

using namespace sdfat;

SdFat SD;

#define SD_CS_PIN 4
File myFile;

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Serial.print("Initializing SD card...");

  if (!SD.begin(SD_CS_PIN)) {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");
```

earlephilhower, 2019-04-28:
Ahha, I think I see the difference:

I am using the `SD` library example `ReadFiles.ino`, which is the same as was always included in the ESP8266 ARduino tree.

You are using the ESP8266SdFat example `ReadFiles.ino` which is different and much updated by the SdFat maintainer.  That library was just added by me since we threw out the unmaintained, cut-n-pasted version from Arduino.cc

So, there is something in the SdFat examples which is incorrect, but the SD ones are fine.  LEt me see.  It's probably a simple define/namespace/config issue.

luc-github, 2019-04-28:
and about SD library :
```
Serial.print("Initializing SD card...");

  if (!SD.begin(4)) {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");
```
sample is not good anymore ?

earlephilhower, 2019-04-28:
Think we cross-posted.  You're not using the SD examples, but the SdFat ones...

luc-github, 2019-04-28:
read my first post I tried both

earlephilhower, 2019-04-28:
So, what's going on is the ESP8266SdFat examples have an updated, incompatible with the older SD SD.h where they make an ESP8266-incompatible File (defined in a private header) and do other badness.  It's fine for other systems where they're the only filesystem, just not good for us.  And not needed since our SD.h is compatible with all our other Filesystems (SPIFFS, LittleFS, etc.).

Basically, I think the only answer is to ignore the SdFat examples and use the 8266-compatible SD ones.  I can blow the examples away from the repo to avoid this issue.

earlephilhower, 2019-04-28:
> read my first post I tried both

For the SD one, did you modify the SD.begin() line or set the `#define`?  The `#define` would be unused.  I'm unable to repro any failure, so we'll need some others to chime in to get any other idea.

luc-github, 2019-04-28:
the sample code from SD :
 https://github.com/esp8266/Arduino/blob/master/libraries/SD/examples/listfiles/listfiles.ino

I did not change anything - same code in 2.5 works and same code failed in GIT

luc-github, 2019-04-28:
I changed code like you:
```
 if (!SD.begin(D0)) {
...
```
and I get 
```
listfiles:38:17: error: 'D0' was not declared in this scope

   if (!SD.begin(D0)) {

                 ^
```
is tht normal ?

earlephilhower, 2019-04-28:
That's definitely not normal.

Do you maybe have the GIT and the IDE version installed at the same time?  Need to uninstall one to use the other successfully.  The lack of a `D0` define means even basic `Arduino.h` isn't working for you.  Assuming that there is a "D0" on your board, of course. :)

luc-github, 2019-04-28:
Yes I carefully uninstall 2.5 , removed url from preferences, deleted directories:
[image]

I use ESP8266 generic board as mentioned in bug description:
```
Settings in IDE
Module: Generic ESP8266 Module
Flash Mode:	DOUT
Flash Size: 4MB
lwip Variant: v2 Lower Memory
Reset Method:	nodemcu
Flash Frequency:	40Mhz
CPU Frequency:	160MHz
Upload Using:	SERIAL
Upload Speed: 115200
```

where is the D0 defined ? I do not see it in pins_arduino.h
[image]

luc-github, 2019-04-28:
Ok seems D0 is defined for NodeMCU but not for generic

TD-er, 2019-04-28:
As a short reply to me being mentioned.

I know there is support in ESPeasy for SD card (must enable and build yourself), but not many use it.
At least I've not seen reported issues regarding SD card support being broken recently.

earlephilhower, 2019-04-28:
Thanks, @TD-er . 

@luc-github it could be a different default speed for the SPI bus which your reader/wiring may not support.

Try changing the begin line to set the SPI speed as well:
````
if (!SD.begin(16, SD_SCK_MHZ(1)) { 
...
````

Obviously using the CS pin you've wired up in place of 16.

luc-github, 2019-04-28:
Actually I tried before SPI_QUARTER_SPEED and SPI_HALF_SPEED without success, I tried these values because they worked with 2016 version of SDFat beta and because SPI_HALF_SPEED  was default value in 2.5.

but SD_SCK_MHZ(1) seems to work , new SD need lower SPI SPEED ?

earlephilhower, 2019-04-28:
That's probably a question better left to the SDFAT maintainer, @greiman.  I just brought his massively updated and improved (LFNs, etc.) version into the 8266 Arduino repo, but did not actually change any of the guts (other than to place it in a namespace).

I think it's more likely the older settings were not doing what they were supposed to and you were really going slower/faster than expected on the board.  Or, it could be some change to SPI (which I haven't been following, sorry) that's now interpreting things differently.

earlephilhower, 2019-04-28:
And, FWIW, I'm not seeing any problem w/o the 1MHZ specification (i.e. I just use  "SD.begin(XX)", no SPI settings param) when I run the SD examples.  So it's very odd, maybe something timing related on the SPI bus changed....

luc-github, 2019-04-29:
Thank you, well looking at speeds in SD 2.5 they are defined here : https://github.com/esp8266/Arduino/blob/951aeffa765605f30ae144a1075fac9c93fa199e/libraries/SD/src/utility/Sd2Card.h#L29-L41
```
#ifdef ESP8266
#include "SPI.h"
uint32_t const SPI_FULL_SPEED = 8000000;
uint32_t const SPI_HALF_SPEED = 4000000;
uint32_t const SPI_QUARTER_SPEED = 2000000;
#else
/** Set SCK to max rate of F_CPU/2. See Sd2Card::setSckRate(). */
uint8_t const SPI_FULL_SPEED = 0;
/** Set SCK rate to F_CPU/4. See Sd2Card::setSckRate(). */
uint8_t const SPI_HALF_SPEED = 1;
/** Set SCK rate to F_CPU/8. Sd2Card::setSckRate(). */
uint8_t const SPI_QUARTER_SPEED = 2;
#endif
```
in new SD :
```
new SD
// SPI divisor constants
/** Set SCK to max rate of F_CPU/2. */
#define SPI_FULL_SPEED SD_SCK_MHZ(50)
/** Set SCK rate to F_CPU/3 for Due */
#define SPI_DIV3_SPEED SD_SCK_HZ(F_CPU/3)
/** Set SCK rate to F_CPU/4. */
#define SPI_HALF_SPEED SD_SCK_HZ(F_CPU/4)
/** Set SCK rate to F_CPU/6 for Due */
#define SPI_DIV6_SPEED SD_SCK_HZ(F_CPU/6)
/** Set SCK rate to F_CPU/8. */
#define SPI_QUARTER_SPEED SD_SCK_HZ(F_CPU/8)
/** Set SCK rate to F_CPU/16. */
#define SPI_EIGHTH_SPEED SD_SCK_HZ(F_CPU/16)
/** Set SCK rate to F_CPU/32. */
#define SPI_SIXTEENTH_SPEED SD_SCK_HZ(F_CPU/32)
```
which give  following values:
```
SPI_FULL_SPEED 50000000
SPI_DIV3_SPEED 26666666
SPI_HALF_SPEED 20000000
SPI_DIV6_SPEED 13333333
SPI_QUARTER_SPEED 10000000
SPI_EIGHTH_SPEED 5000000
SPI_SIXTEENTH_SPEED 2500000
SD_SCK_MHZ(1) 1000000
```
Using SPI_SIXTEENTH_SPEED in new SD instead of SPI_HALF_SPEED in 2.5 SD make it works on my side
I do not know if SPI changed but using same range of  value give consistent behavior
 
I can imagine  your setup is using fast hardware, so going faster is not an issue, but on my side it is obviously a problem, and so even the new SD is far better than old one - somewhere compatibility is broken and a note concerning speeds would be welcome IMHO.

On my side I was using old SDFat beta 2016 until now, so I can do speed adjustment in same time I will integrate the new SD instead of old SDFat beta, so impact is not high, but for people that do not change anything (like I did using sample code) they may get surprised if same code failed now.

Thank you again for your great work - new lib will be far better than old one.

earlephilhower, 2019-04-29:
@luc-github I'm reopening this as it's a simple fix to be backwards compatible.

I think @igrr fixed the ancient Arduino libs, and before you were running at 2.5MHz fot SPI_SPEED_FULL...

https://github.com/esp8266/Arduino/blob/951aeffa765605f30ae144a1075fac9c93fa199e/libraries/SD/src/utility/Sd2Card.cpp#L576

earlephilhower, 2019-04-29:
Thanks again for the debug, @luc-github .

PR #6023 (2019-04-29): Use original ESP8266-specific speeds for SPI speeds
Replace the default SPI_FULL/HALF/QUARTER_SPEEDs with ESP8266
appropriate ones taken from the original SD library that @igrr modified.

Fixes #6019

@luc-github , can you try your unmodified code out on this PR and see if it works with no changes from 2.5.0?  My setup always seemed to work, so I'm unable to reproduce the failure you had, but it looks pretty straightforward.
commit feb3988b (2019-04-29): Use original ESP8266-specific speeds for SPI speeds (#6023)

Replace the default SPI_FULL/HALF/QUARTER_SPEEDs with ESP8266
appropriate ones taken from the original SD library that @igrr modified.

Fixes #6019
closed 2019-04-29 by earlephilhower