esp8266/Arduino #6024: [Static IP] Code that previously worked no longer works

rtek1000, 2019-04-29:
Hi,

Without setting a fixed address, the ESP8266 can get the IP address via DHCP, so the hello webserver example works normally. But if I add the static IP configuration, then the browser can not find the server (The serial monitor shows the static IP address correctly)

The code worked correctly until about November 2018 (it was the last time I tested it).

Were there any radical changes in the library?

Arduino IDE 1.8.9
ESP8266 2.5.0 (library installed using the Board Manager)
Windows 10 x64
NodeMCU Lolin V3

I will downgrade the library version to try to find a version that works.

Ref.: https://arduino-esp8266.readthedocs.io/en/2.5.0/esp8266wifi/station-class.html#config

Basic code:
```
WiFi.begin(ssid, password);
IPAddress ip(192,168,1,74);   
IPAddress gateway(192,168,1,1);   
IPAddress subnet(255,255,255,0);   
IPAddress dns(8,8,8,8); 
WiFi.config(ip, gateway, subnet, dns);
```

Thank you.

rtek1000, 2019-04-29:
Yep, It worked with version 2.4.0 (but what a giant package this is!)

rtek1000, 2019-04-29:
I uninstalled version 2.4.0 and installed version 2.5.0 and then IDE showed this error:

> cc1plus.exe: out of memory allocating 65536 bytes

If someone else receives this error: I've seen this error before, and to resolve it just reboot the computer.

d-a-v, 2019-04-29:
It will work if you move `WiFi.config()` before `WiFi.begin()`.
This is #5839 duplicate

closed 2019-04-29 by d-a-v

rtek1000, 2019-04-29:
Yep, It worked with version 2.5.0 now.

It looks like the problem was waiting for the connection before setting the IP address.

Thank you.

[image]

PR #6026 (2019-05-01): fix switching to static address with lwip2
For some reason, ip address is not propagated in a visible way for lwip2
when switching to static address (wifi.config()) *after* wifi.begin().

This patch calls lwip-v1.4's netif_set_addr() with the new ip address to set
all things up, just like it is done and right when wifi.begin() is called
after wifi.config().

Also tested when IPv6 is enabled.

fixes #5839
fixes #6024
commit 17500226 (2019-05-01): fix switching to static address with lwip2 (#6026)

fix switching to static address with lwip2

For some reason, ip address is not propagated in a visible way for lwip2
when switching to static address (wifi.config()) *after* wifi.begin().

This patch calls lwip-v1.4's netif_set_addr() with the new ip address to set
all things up, just like it is done and right when wifi.begin() is called
after wifi.config().

Also tested when IPv6 is enabled.

fix documentation: It is more natural to set an IP address before starting WiFi
(.. and not after dhcp has started)
fixes #5839
fixes #6024